TrunkRecorder/trunk-recorder #131: trunk-recorder not decoding control channel?

kg6uyz, 2017-07-14:
I have setup trunk recorder but am having issues, it doesnt seem to be decoding my control channel or something of that sort, i am not exactly sure but when there is activity on the system trunk-recorder doesnt seem to do anything, and there are errors that keep coming in the log.  I have pasted the console output below, any help would be greatly appreciated.  I did see **[R82XX] PLL not locked!**  in the console output, i am using a RTL-SDR dongle i bought on ebay, i have tested the dongle with SDRConsole on windows and it works fine in the vhf frequency range i am wanting to decode.

Jeff.

---------------------------------

kg6uyz@JAC04:~/trunk-build$ ./recorder
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.010.001.001-release

[2017-07-14 09:45:25.498960] (info)   Using Config file: ./config.json

[2017-07-14 09:45:25.499426] (info)   System Type: p25
[2017-07-14 09:45:25.499460] (info)   Control Channels:
[2017-07-14 09:45:25.499520] (info)   1.5203e+08
[2017-07-14 09:45:25.499591] (info)   1.5212e+08
[2017-07-14 09:45:25.499649] (info)   1.5215e+08
[2017-07-14 09:45:25.499697] (info)   1.5218e+08
[2017-07-14 09:45:25.499741] (info)   1.5263e+08
[2017-07-14 09:45:25.499777] (info)   1.5272e+08
[2017-07-14 09:45:25.499807] (info)   1.5281e+08
[2017-07-14 09:45:25.499843] (info)
-------------------------------------
System Number: 1
[2017-07-14 09:45:25.499882] (info)   Short Name: vhf_simulcast
[2017-07-14 09:45:25.499908] (info)   API Key:
[2017-07-14 09:45:25.499938] (info)   Upload Script:
[2017-07-14 09:45:25.499957] (info)   Call Log: true
[2017-07-14 09:45:25.499990] (info)   Audio Archive: true
[2017-07-14 09:45:25.500009] (info)   Loading Talkgroups...
[2017-07-14 09:45:25.500057] (error)   Error Opening TG File:

[2017-07-14 09:45:25.500078] (info)   Talkgroups File:
[2017-07-14 09:45:25.500131] (info)   Record Unkown Talkgroups: true
[2017-07-14 09:45:25.500164] (info)
[2017-07-14 09:45:25.500220] (info)   Capture Directory: /home/kg6uyz/recordings
[2017-07-14 09:45:25.500258] (info)   Upload Server: encode-upload.sh
[2017-07-14 09:45:25.500279] (info)   Default Mode: digital
[2017-07-14 09:45:25.500297] (info)   Call Timeout (seconds): 3
[2017-07-14 09:45:25.500331] (info)   Log to File: false
[2017-07-14 09:45:25.500410] (info)   Center: 1.55e+08
[2017-07-14 09:45:25.500439] (info)   Rate: 8e+06
[2017-07-14 09:45:25.500482] (info)   Error: -11000
[2017-07-14 09:45:25.500508] (info)   PPM Error: 0
[2017-07-14 09:45:25.500529] (info)   Gain: 15
[2017-07-14 09:45:25.500562] (info)   IF Gain: 0
[2017-07-14 09:45:25.500592] (info)   BB Gain: 0
[2017-07-14 09:45:25.500614] (info)   LNA Gain: 0
[2017-07-14 09:45:25.500646] (info)   MIX Gain: 0
[2017-07-14 09:45:25.500667] (info)   Squelch: 0
[2017-07-14 09:45:25.500708] (info)   Idle Silence: false
[2017-07-14 09:45:25.500727] (info)   Digital Recorders: 6
[2017-07-14 09:45:25.500760] (info)   Debug Recorders: 0
[2017-07-14 09:45:25.500779] (info)   Analog Recorders: 0
[2017-07-14 09:45:25.500806] (info)   Driver: osmosdr
[2017-07-14 09:45:25.500831] (info)   Modulation: qpsk
[2017-07-14 09:45:25.500864] (info)   Source Device: rtl=00000001
gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[2017-07-14 09:45:26.085032] (info)   SOURCE TYPE OSMOSDR (osmosdr)
[2017-07-14 09:45:26.085117] (info)   Setting sample rate to: 8e+06
Invalid sample rate: 8000000 Hz
[2017-07-14 09:45:26.085222] (info)   Actual sample rate: 1.024e+06
[2017-07-14 09:45:26.085276] (info)   Tunning to 1.54989e+08hz
[2017-07-14 09:45:26.136728] (info)   Gain Stage: LNA supported values: 0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48 49.6
[2017-07-14 09:45:26.136807] (info)   Max HZ: 1.59e+08
[2017-07-14 09:45:26.136861] (info)   Min HZ: 1.51e+08
[2017-07-14 09:45:26.162358] (info)   Gain set to: 14.4
[2017-07-14 09:45:26.180924] (info)      P25 Recorder Initial Rate: 500000 Resampled Rate: 50000 Initial Decimation: 16 Decimation: 10 System Rate: 48000 ARB Rate: 0.96
[2017-07-14 09:45:26.181222] (info)   P25 Recorder Taps - initial: 3031 channel: 1137 ARB: 757 Total: 4925
Project 25 IMBE Encoder/Decoder Fixed-Point implementation
Developed by Pavel Yazev E-mail: pyazev@gmail.com
Version 1.0 (c) Copyright 2009
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; see the file ``LICENSE'' for details.
[2017-07-14 09:45:26.196372] (info)      P25 Recorder Initial Rate: 500000 Resampled Rate: 50000 Initial Decimation: 16 Decimation: 10 System Rate: 48000 ARB Rate: 0.96
[2017-07-14 09:45:26.196656] (info)   P25 Recorder Taps - initial: 3031 channel: 1137 ARB: 757 Total: 4925
[2017-07-14 09:45:26.211607] (info)      P25 Recorder Initial Rate: 500000 Resampled Rate: 50000 Initial Decimation: 16 Decimation: 10 System Rate: 48000 ARB Rate: 0.96
[2017-07-14 09:45:26.211894] (info)   P25 Recorder Taps - initial: 3031 channel: 1137 ARB: 757 Total: 4925
[2017-07-14 09:45:26.226394] (info)      P25 Recorder Initial Rate: 500000 Resampled Rate: 50000 Initial Decimation: 16 Decimation: 10 System Rate: 48000 ARB Rate: 0.96
[2017-07-14 09:45:26.226678] (info)   P25 Recorder Taps - initial: 3031 channel: 1137 ARB: 757 Total: 4925
[2017-07-14 09:45:26.241162] (info)      P25 Recorder Initial Rate: 500000 Resampled Rate: 50000 Initial Decimation: 16 Decimation: 10 System Rate: 48000 ARB Rate: 0.96
[2017-07-14 09:45:26.241447] (info)   P25 Recorder Taps - initial: 3031 channel: 1137 ARB: 757 Total: 4925
[2017-07-14 09:45:26.256021] (info)      P25 Recorder Initial Rate: 500000 Resampled Rate: 50000 Initial Decimation: 16 Decimation: 10 System Rate: 48000 ARB Rate: 0.96
[2017-07-14 09:45:26.256304] (info)   P25 Recorder Taps - initial: 3031 channel: 1137 ARB: 757 Total: 4925
[2017-07-14 09:45:26.262068] (info)   [vhf_simulcast]   Started with Control Channel: 1.5203e+08
[2017-07-14 09:45:26.270223] (info)      P25 Recorder Initial Rate: 500000 Resampled Rate: 50000 Initial Decimation: 16 Decimation: 10 System Rate: 48000 ARB Rate: 0.96
[2017-07-14 09:45:26.270500] (info)   P25 Recorder Taps - initial: 1455 channel: 1137 ARB: 757 Total: 3349
[2017-07-14 09:45:29.013210] (error)   [vhf_simulcast] Retuning to Control Channel: 1.5212e+08
[2017-07-14 09:45:29.013299] (info)      - System Source - Min Freq: 1.51e+08 Max Freq: 1.59e+08
[2017-07-14 09:45:29.013636] (error)   [vhf_simulcast]   Control Channel Message Decode Rate: 0/sec, count:  0
[2017-07-14 09:45:32.020359] (error)   [vhf_simulcast] Retuning to Control Channel: 1.5215e+08
[2017-07-14 09:45:32.020452] (info)      - System Source - Min Freq: 1.51e+08 Max Freq: 1.59e+08
[2017-07-14 09:45:32.020733] (error)   [vhf_simulcast]   Control Channel Message Decode Rate: 0/sec, count:  0
[2017-07-14 09:45:35.017015] (error)   [vhf_simulcast] Retuning to Control Channel: 1.5218e+08
[2017-07-14 09:45:35.017111] (info)      - System Source - Min Freq: 1.51e+08 Max Freq: 1.59e+08
[2017-07-14 09:45:35.017373] (error)   [vhf_simulcast]   Control Channel Message Decode Rate: 0/sec, count:  0
[2017-07-14 09:45:38.015873] (error)   [vhf_simulcast] Retuning to Control Channel: 1.5263e+08
[2017-07-14 09:45:38.015970] (info)      - System Source - Min Freq: 1.51e+08 Max Freq: 1.59e+08
[2017-07-14 09:45:38.016342] (error)   [vhf_simulcast]   Control Channel Message Decode Rate: 0/sec, count:  0
[2017-07-14 09:45:41.013271] (error)   [vhf_simulcast] Retuning to Control Channel: 1.5272e+08
[2017-07-14 09:45:41.013368] (info)      - System Source - Min Freq: 1.51e+08 Max Freq: 1.59e+08
[2017-07-14 09:45:41.013626] (error)   [vhf_simulcast]   Control Channel Message Decode Rate: 0/sec, count:  0

closed 2017-07-22 by kg6uyz

shwetapalande, 2017-08-15:
Did you happen to find a solution for this issue? I am facing a similar problem here.

bctrainers, 2017-08-15:
@shwetapalande In my six RTLSDR setup, and having each of them numbered accordingly, 01-06. I moved the frequency range that contained the CC to the first RTLSDR on the serial numbers. Essentially, the first SDR would be monitoring the CC and its 2Mhz range. It's wonky, but that's what made it work for me.

robotastic, 2017-08-16:
@bctrainers  Ahh... that could be related to this: https://github.com/robotastic/trunk-recorder/wiki/FAQ

Try setting the buflen

kg6uyz, 2017-08-16:
In my case, i realized i specified the wrong center frequency range, the control channels for my works p25 simulcast system is in the 152mhz range all withen 1mhz.  I specified a center of 155mhz.  Took me a while to realize i specified the wrong center, kept overlooking that i guess, once i changed it it was fine.

 However i am running into an new issue where it seems if the recorder is running for more then 1 day my recorded audio files dont contain any audio, they get created, but are 1k in size.  I have been having to restart trunk-recorder about every 24 hours, after its restarted it works fine but after around 24 hours same thing happens.  And it seems to periodically stop decoding the control channel.  I will get my work radio out and verify my system is still operating and my radio sees the control channel, not sure whats going on there, i usually have to stop trunk-recorder, pull the sdr and reconnect and its back to working like normal.