TrunkRecorder/trunk-recorder #134: Audio files 1k in size after ~24hrs of running

kg6uyz, 2017-08-30:
I have been running trunk-recorder for a while now decoding my p25 vhf 7ch simulcast trunking system, i am experiencing an issue where after around 24 hours (and some times it could be sooner) the audio files get no larger then 1k in size.  There will be a whole block of several talk groups that get recorded but audio files get no larger then 1k in size, with no recorded audio contained in the file.  Then every so often a talk group gets recorded successfully.  I have to stop trunk-recorder and start it again, then its fine for another usually 24 hours or so, but then does the same thing.

robotastic, 2017-08-30:
Hmm… that is really weird. What kind of SDR are you using? I haven’t heard of any issues like that. Could it be rotating control channels? If your SDR doesn’t have a TCXO, the amount of error in tuning could be increasing over time, until it is no longer able to tune in the control channel. I would double check your error values for tuning. maybe you are right on the edge and it drifts with changes in the atmosphere.

Treehouseman, 2017-08-30:
Check if it's 44 bytes or 1kb, 44 byes is absolutely a case of bad reception. if it's only after a certain amount of time it'll be heat based and drift, even the txco radios can drift enough given enough temp change, rtlsdr runs hot, even the nice nesdr smart's that have thermal pads and a metal case.

kg6uyz, 2017-08-30:
I am using this SDR dongle:

kg6uyz, 2017-08-30:
Also as soon as i restarted trunk-recorder earlier its recording the audio with no issues.  And i am running analog recorders on another dongle on the same trunk-recorder implimentation,, the analog recorders don't seem to suffer from this issue.

Treehouseman, 2017-08-30:
yeah, it's drift, that's exactly the same issue I had.

closed 2017-09-14 by kg6uyz