TrunkRecorder/trunk-recorder #136: Recording Unit ID of radio's involved in a transmission (P25 Trunking)

kg6uyz, 2017-09-14:
Can trunk-recorder be setup to also extract and log the radio unit ID's and log them to the .json file and be tracked like it tracks the channels the call trunks through?

vabiro, 2017-09-27:
It would be great if was kicked out as part of the stdout and logging too. To be honest, the more info the better. 

I am a crappy C programmer, and know I could adapt something great with this great code, but admit my competing foci. IOW, I can't figure out what I want to do: what I want or what I need.

dreinhold, 2017-09-27:
The radio ID are in the srcList in the json.

`"srcList": [ {"src": 50046, "time": 1506508074, "pos": 0.000000}, {"src": 16299, "time": 1506508086, "pos": 10.899250} ],`

kg6uyz, 2017-09-29:
My srcList doesnt populate any info, and is that radio user ID's as in who is talking in the transmission or ID of the SDR receiver?

dreinhold, 2017-09-29:
I am a little behind so maybe it stopped working. But yes the srcList is the decimal id of the radio that is transmitting.

kg6uyz, 2017-10-03:
Anybody have an idea on why it isnt working or something im missing in my config?

robotastic, 2017-10-03:
Hmm - that is weird. Can you post your config file? It is possible Trunk Recorder is having trouble getting the source id from the control channel messages.

kg6uyz, 2017-10-03:
{
"sources": [{
        "center": 152500000.0,
        "rate": 2048000,
        "error": 0,
        "gain": 0,
        "debugRecorders": 0,
        "digitalRecorders": 7,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=00000002"
	}, {
        "center": 155000000.0,
        "rate": 2048000,
        "error": 0,
        "gain": 0,
        "debugRecorders": 0,
        "digitalRecorders": 6,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=00000004"
	}, {
	"center": 154900000.0,
        "rate": 2048000,
        "squelch": -23,
        "error": 0,
        "gain": 0,
        "analogLevels": 30,
        "modulation": "fsk4",
        "analogRecorders": 8,
        "driver": "osmosdr",
        "device": "rtl=00000004"		
    }],
    "systems": [{
        "control_channels": [152030000, 152120000, 152150000,152180000, 152630000, 152720000, 152810000],
        "type": "p25",
        "recordUnknown": true,
	"callLog": true,
        "talkgroupsFile": "talkgroups_digital.csv",
        "shortName": "NGEN_VHF_SIMULCAST",
	"uploadScript": "vhf_simulcast.sh"
	}, {
	"control_channels": [154085000, 154755000, 155100000,155775000],
        "type": "p25",
        "recordUnknown": true,
	"callLog": true,
        "talkgroupsFile": "talkgroups_digital.csv",
        "shortName": "NGEN_LEWIS_RD",
        "uploadScript": "lewis_rd.sh"
	}, {       
	"control_channels": [154355000, 154785000, 155190000,155535000],
        "type": "p25",
        "recordUnknown": true,
        "talkgroupsFile": "talkgroups_digital.csv",
        "shortName": "NGEN_BRYANT",
        "uploadScript": "bryant.sh"

    }],
    "captureDir": "/var/www/html/trunk/"
}

robotastic, 2017-10-03:
Hmm - It could be that it is a VHF system. There are 2 different types of messages in P25, normal and extended and there is a chance I am not processing the extended 100% correctly. I think some VHF systems use the extended format. Can you try the `status-socket` branch? I may have fixed some problems there.

kg6uyz, 2017-10-03:
I changed to the status-socket branch, and now i am getting the decimal unit id's.  I also noticed in the console output unit registration responces.

kg6uyz, 2017-10-11:
Should i stay on the status-socket or will whatever changes in this socket be brought into the master?

closed 2017-12-10 by kg6uyz