TrunkRecorder/trunk-recorder #137: Best OS without GUI

rosecitytransit, 2017-09-26:
I'm wondering what the best Linux distribution is for just running trunk recorder and maybe a Web server.  I am thinking of something without a desktop since I wouldn't be needing it.

bctrainers, 2017-09-26:
Currently running Ubuntu 16.04.3 LTS (GNU/Linux 4.4.0-96-generic x86_64).

`Linux box04 4.4.0-96-generic #119-Ubuntu SMP Tue Sep 12 14:59:54 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`

No issues here with trunk-recorder. Just need a machine with a decent amount of CPU power.

vabiro, 2017-09-27:
I am having good luck with ArchLinux. It is a PITA to install, but once you do the basic installation is very useful, and you can add anything through its package management system, as easily as debian, but the overhead is very low.

rogerxxxx, 2017-09-28:
> On Tue, Sep 26, 2017 at 12:29:54PM -0700, rosecitytransit wrote:
>   I'm wondering what the best Linux distribution is for just running trunk
>   recorder and maybe a Web server. I am thinking of something without a
>   desktop since I wouldn't be needing it.

I primarily use Gentoo Linux, similar in structure to Debian, but source based 
requiring maintenance for compiling packages.  Gentoo is great for keeping a 
small or slim system, but packages sometimes break requiring finicky 
maintenance.  Gentoo also likes to cut some of the greater features out of it's 
source base distribution, such frowning upon supporting ebuilds for development 
CVS/SVN/GIT software projects, and more so opt for stable versions.  Shrugs...  
focusing on the stable market versus supporting software developers.

Since I did some research recently on trying to find a small easy working 
binary Linux distribution, allowing for creating easy custom packages for 
compiling such exquisite software projects such as these radio utilities, I 
think the following would be my preference:

Alpine Linux
https://www.alpinelinux.org/
Very small & light, binary distro. Uses a newer lighter musl libc and busybox, 
rather than the heavier GNU libc and native tool versions.  Very easy to create 
a custom package, as the package scripts are simple shell scripts, rather than 
requiring some big CPU resource intensive custom extravagant packaging system.

Arch Linux
Just because it is well known.  I would definitly vote for Arch Linux without 
SystemD.  Otherwise, if I had to stick with using the SystemD bloat and 
breakage (including the maintenance headaches of SystemD), I would then avoid 
Arch Linux.

Debian & Fedora, due to being tried and true, except for the headaches of 
SystemD.  You also get a lot of system dependencies unlike Alpine, Arch & 
Gentoo Linux.  I pretty much install mainly Fedora for testing hardware such as 
blue tooth hardware, and or Debian, ensureing the hardware is working as 
expected, as well as any other hardware.  Shrugs.  Hardware testing only 
distributions in my opinion.  Debian & Fedoras packaging systems pretty much 
scripts & C code the last I knew.

Top of my list for running a small server would be Alpine Linux.  There's now 
also a package providing compatible GNU libc support, for providing the 
standard libc for binary only released software such as VueScan.  Alpine 
Linux's website is also really basic and easy to navigate, along with the 
forums.  No confusing eye-candy hiding any ugly interior features, yet.

There are many others, but those many others are just clones of Debian, Ubuntu, 
and/or Fedora; and including SystemD stuff.  It is nice to see package 
compatibility between the cloned distributions.  But for me it's like trench 
warfare, and just jumping into another trench that leads to the same old bloat!

So again in brief, take a look at Alpine Linux, if that's not for you as it may 
require adding a couple of more hours volunteering to write package scripts for 
your packages, then I suggest Arch Linux without SystemD.

Shrugs, if you love SystemD; then ignore what I've said about SystemD here. ;-)

And as far as my success with selecting a small light distribution for my old 
laptop, I just bought an Android based Samsung Galaxy S3 Tab for reading 
digital books and writing notes, or performing anything I used to use my laptop 
for except for programming.  :-/

If I had more money, I'd probably just buy another $2,000 Dell Linux compatible 
laptop and install Gentoo Linux again. ;-)

Good luck!

-- 
Roger
http://rogerx.freeshell.org/

rosecitytransit, 2017-09-28:
After doing some research, I decided to do a [minimal install of Ubuntu](https://help.ubuntu.com/community/Installation/MinimalCD) so I'll close this.

closed 2017-09-28 by rosecitytransit