TrunkRecorder/trunk-recorder #138: Multiple Custom Bandplans on a single system

MaxwellDPS, 2017-09-28:
# Multiple Bandplan Sites?

I'm trying to record site 28's traffic from this system
http://www.radioreference.com/apps/db/?sid=2315

Custom Frequency Tables For System 2315

Base | Spacing | Offset
-- | -- | --
155.41500 | 15.0 | 380
***157.20000*** | ***12.5*** | ***436***
**157.47000** | **15.0** | **454**

The 2nd and 3rd Entry's apply to our freqs

We need to cover

site | Channel | FREQ
-- | -- | -- 
**28** | **449** | **157.3625**
28 | 592 | 159.5400
28 | 608 | 159.7800
28 | 617 | 159.9150
28 | 627 | 160.0650

ALL Freqs ex. 449 work with:

Base | Spacing | Offset
-- | -- | --
157.47000 | 15.0 | 454

Channel 449 Needs this band plan:

Base | Spacing | Offset
-- | -- | --
157.20000 | 12.5 | 436

We have been trying to find a mathematical middle but not sure that will work :(

I can post our config if needed... also what files handle band plans... I do have some C experience 

## NOTE

I am aware that RR says we do not use Channel# 449... but I promise you a lot of our traffic goes over it

Site | Name | County | Freqs |   |   |   |   |   |  
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
028 (1C) | Madison | Lake | 159.540 | 159.780c | 159.915 | 160.065a |   |   |

kb2ear, 2020-07-24:
Any progress on this I have a system that needs at least 2 band plan entries

robotastic, 2023-02-19:
Totally open to a PR on this, but I am not sure how to solve it. I think the best solution maybe creating a second system with that single frequency in it for now.

closed 2023-02-19 by robotastic