TrunkRecorder/trunk-recorder #141: Two Tone Sequential Detection

Dygear, 2017-10-15:
In an effort for the general detection of tones from multiple vendors (General Electric's Type 99 or Motorola's Quik-Call I and Quik-Call II), I propose the following. As these systems use either a single tone, or sequential tones for a defined length of time supporting all of them generically should look something like this.

    "channels": [154115000: [{"TTS": [{"len": 1000, "tone": 1122.5}, {"len": 2000, "tone": 1395}]}]]

* TTS - Two Tone Sequential
  * len - `Len`gth of Tones in Milliseconds 
  * tone - `Tone` to detect in Hz.

All of my comments today, #107, #100 follow the same format for the channel array.

mndmr, 2017-10-15:
Great feature to add. We also broadcast the tones on P25 trunked system. So having the ability to detect the tones on the P25 trunked system is beneficial.

Is the TTS analyzing the recorded wav file after its recorded? 

I would like this array to be similar to how talkgroups are looked up. CSV file containing Name, Tone A, Tone B.  Our system currently had about 15-20 toneout per frequency.

Dygear, 2017-10-15:
I'd prefer that the tone analysis to be done while the audio is being demodulated so it can be placed within the JSON / WebSocket / API information. I didn't think the P25 uses audible tones, my understanding was this was done at the protocol level.

johann8384, 2017-10-26:
P25 doesn't use audible tones, but some of us using P25 transmit audible tones for....reasons...

vabiro, 2017-10-26:
I'm getting the impression that there are systems that - despite "...reasons..." continue to use audible tones over P25 to identify radios. This is probably to accommodate legacy systems, but offer an easy way to work with those systems, rather than being P25 police. After all, we are trying to monitor the system rather than build them.

kg6uyz, 2018-09-07:
> Yeah, that's what I was trying to say, we are still broadcasting the tones
> for various reasons. One of them is that fire department dispatches still
> happen on VHF pagers so we have patches to VHF from talkgroups to the FD
> repeaters for those.

The p25 radio system I maintain utilizes analog channels patched to
talkgroups for cross mode interoperqbility.  We page on the analog side of
that patch for the fire departments, tones dont go out on digital.  Though
we have the capability to have those tones go out the trunked talkgroup
should we need them to.

Motorola has released a new feature that allows digital two tone decode on
a p25 talkgroup.

mikewren, 2020-06-17:
Hi all - I am also very interested in this functionality. Both for kicking off shell scripts based on the presence of specific tones, and also for potentially muting or editing the resultant wav file to omit tones.

Maybe a bounty/reward might help kickstart this feature?  :)

Dygear, 2020-06-17:
Especially now that TTD was bought by IAmResponding.

ZeroChaos-, 2021-08-31:
I'd love to necromance this issue.  If anyone has a place I can add to a feature bounty, I will gladly do that.  Honestly I can't even seem to do two tone decoding in linux at all and it would be a great feature in my mostly analog area

leee, 2021-08-31:
The only bounty/reward that I'm aware of that `trunk-recorder` has participated in is the Broadcastify Calls one.

If you wish to donate to the principal, and find `trunk-recorder` useful or meaningful in some way, you should absolutely sponsor `robotastic` on GitHub. This obviously holds no contract for completion of tasks and projects, but [as said here](https://github.com/sponsors/robotastic):

> Sponsorship will help keep me focused on maintaining Trunk Recorder and also encourage me to build an updated version. Having "Customers" will create a good feeling of obligation.

🙂 

Contributions are always welcome of course! QCII detection (and in general, PL detect) shouldn't require full FFT or audio event detection. An efficient technique for computing single bin DFT would be the Goertzel algo. I think something like this might be better done in post-processing (as in `uploadScript`), but if you really wanted to do this within `trunk-recorder`, `gnuradio` provides `gnuradio/fft/goertzel`.

An "extension" of this problem could also be true audio event detection of console tones, but some of these tones get a lot more complex than just Alert Tone 1 (straight tone). Console tone SED could turn into an interesting ML project, especially to make it small-footprint/low-power.

robotastic, 2021-09-02:
Good question on how to get this solved. I am fully supportive!! ... but don't know much about Tones and how to detect them. The good news, is that with v4.0, the WAV file sink can now capture metadata information about a call transmission being recorded and pass it to be written into the JSON file. This means that could pass the final analog & digital audio through some sort of detector and write that to the JSON. 

What would be really helpful, is if someone could find a good Gnuradio based example of taking in audio with a tone and having it decode the correct info from it. I think I should be able to work that example code into Trunk Recorder pretty easily. I would then need some help with testing and debugging, since I don't have anything nearby.

No donations needed for me to work on this... more a function of finding someone with the skill and background to put together a rough POC.

Dygear, 2021-09-02:
I can probably do that. I’ll dust off my GNURadio install and see what I can bake up.

dboune, 2022-09-11:
I work with a non-profit project called Watch Duty. We are working on a project involving rpi + trunk recorder, and we work with analog systems most often in what we do. We want to incorporate two-tone detection. It looks like we will be able to do this using the WAV sink as described above, but it would be wonderful if that were built into trunk-recorder. This comment is just a show of support for this feature. If possible we will try to contribute *something* in one way or another.

aaknitt, 2022-09-11:
> I work with a non-profit project called Watch Duty. We are working on a project involving rpi + trunk recorder, and we work with analog systems most often in what we do. We want to incorporate [...]

For now, you can use the [ simplestream plugin](https://github.com/robotastic/trunk-recorder/blob/master/docs/CONFIGURE.md#simplestream-plugin) of trunk-recorder to send "live" audio (not a pre-recorded WAV file) from selected talkgroups or conventional channels to pulseaudio, which can then feed the audio into an external two-tone decoding program like TwoToneDetect or FD Tone Notify.  

Documentation on how to send audio from the simplestream plugin to pulseaudio can be found [here](https://github.com/robotastic/trunk-recorder/blob/master/docs/CONFIGURE.md#example---sending-audio-to-pulseaudio).

It's a bit of hair-pulling to get it all working initially, but it's been done successfully.

dboune, 2022-09-12:
@aaknitt Thank you Andy! I'll check that out. I hadn't seen FD Tone Notify so I appreciate the mention.

tdduarte, 2022-10-11:
Can this work with “zello work” the paid ver of zello?

-Tony

robotastic, 2024-12-15:
Closing for now - We do have some support for Tone detection. If someone wants to take on Two Tone detection, please reopen!

closed 2024-12-15 by robotastic