TrunkRecorder/trunk-recorder #142: Motorola Data Communications Decoding (Stat-Alert, MDC-600, MDC-1200)

Dygear, 2017-10-15:
# Background
MDC uses Audio Frequency Shift Keying (AFSK) at a 600 or 1200 baud rate. AFSK digital data is represented by changes in the frequency (pitch) of an audio tone. With this scheme, the "1" is called the mark frequency (`1200 Hz`) and the "0" is called the space frequency (`1800 Hz`). (AFSK is also used in [EAS](https://en.wikipedia.org/wiki/Emergency_Alert_System) to notify stations of the type of emergency, locations affected, and the time of issue without actually hearing the text of the alert. Caller ID on phone lines also uses the AFSK method.)

# Uses
Data bursts can be send pre or post audio transmission.
**Pre-Tone** data bursts sent when the user presses the PTT button and there is a short delay before the user can speak. Useful as it allows the receiver to see the PTT-ID of the currently speaking user. 
**Post-Tone** Data bursts sent out after the user depresses the PTT button. Uses slightly less air time than the pre-tone variant.
* PTT-ID (sometimes called ANI)
  * Are Hexadecimal [0-9 & A-F] four-digit numbers. (Ex `0000`, `FFFF`, `2900`, `A291`, `291C`, etc).
* Emergency Button
  * Sends the PTT-ID 3 times with no audio.
* Status Buttons
  * Status 1 - Status 8 are available. Protocol implementation is unknown to me.
* Selective Inhibit
  * "Stuns" a radio that might be stolen, or the user keeps an open carrier and makes it unable to transmit. Protocol implementation is unknown to me.
* Radio Check
  * If the radio is in range and turned on, after sending the radio check, the destination radio will respond with an ack. Protocol implementation is unknown to me.
* Selective Calling
  * If the radio is in range and turned on, and receives a selective calling message, the radio could display a "CALL" message and / or make an audible BEEP. 

# Implementation
With the P25 decoder, unit ID information is passed out into the json file. Ideally the same information would be present on conventional systems as well. This would make feature parity between the currently supported implementations.

robotastic, 2017-10-16:
Are there any GnuRadio based implementations? If so, I can try to add it in. I don't have this locally, so it would be pretty tough to debug.

blue1sparky, 2017-10-16:
I have some analog handhelds that have MDC-1200 capability. I could gin up some audio record files with a decoder ring if that could be of use.

-Mike Leger

Sent from my mobile device. Responses may be delayed.

Dygear, 2017-10-22:
[Batlabs' MDC-1200 page](http://www.batlabs.com/mdc1200.html) was the closest thing I could find as far as decoding MDC-1200 outside of the US Patent Office. [`US4517669A` Method and apparatus for coding messages communicated between a primary station and remote stations of a data communications system](https://www.google.com/patents/US4517669), [`US4590473A` Data signalling system](https://www.google.com/patents/US4590473),  [`US4517561A` Selective call, paging and priority signalling system](https://www.google.com/patents/US4517561) & [`US4457005A` Digital coherent PSK demodulator and detector](https://www.google.com/patents/US4457005). The good news being that, the patient for these expired in 2004 - 2006.

dBitech, 2017-10-22:
Code exists, and works well...   https://github.com/atmatthewat/mdc-encode-decode

Dygear, 2017-10-22:
@dBitech, thanks mate!

kkotowick, 2018-03-13:
Any chance you've had any luck with this? MDC-1200 capability would be a huge benefit for my use case as well.

Windrac, 2018-11-16:
> Code exists, and works well... https://github.com/atmatthewat/mdc-encode-decode

I see the code exists, but it fails to compile using gcc on Linux (Mint / Ubuntu). 

@dBitech, how did you manage to get the MDC decoder running? Did you have it polling an audio input from a USB sound card or something?

Thanks.

ZeroChaos-, 2019-03-21:
additional location for code that does this, plus fleet sync
https://github.com/russinnes/fsync-mdc1200-decode

closed 2020-05-21 by robotastic

zero48, 2024-10-25:
[https://youtu.be/BMtGD08MCHU?si=vbCYSxD2KrrcM55w](url)

Dygear, 2024-10-30:
> [https://youtu.be/BMtGD08MCHU?si=vbCYSxD2KrrcM55w](https://youtu.be/BMtGD08MCHU?si=vbCYSxD2KrrcM55w)

Looks like that's being done in hardware? We know it can be done. It's just how is it done. How do we integrate that into Trunk-Recorder.

ZeroChaos-, 2024-10-30:
... it's already done in TR, it's been done a while, that's why this ticket is closed.

zero48, 2024-10-30:
Hello,I'm the one who did this encoder . The last version can handle GE-Star and MDC1200. I did the decoder as well . If I may help Anyhow , feel free to contact me...

taclane, 2024-10-30:
Trunk-recorder's mdc/fleetsync/star/TPS implementation dates back to [this commit](https://github.com/robotastic/trunk-recorder/commit/e96c5377d7963537abbfce7c8bde3ad221685665).

Much of the code has since moved into the [decoder directory](https://github.com/robotastic/trunk-recorder/commit/e96c5377d7963537abbfce7c8bde3ad221685665), but any additional contributions or improvements are certainly welcome.

PR #358 (2020-05-21): Clear analog source list on call end, adding support for P25 conventional source ID logging
This PR does pretty much what the title says:

1. The analog source logging wasn't getting cleared on call end, so they kept piling up. This fixes that bug.
2. While there was source logging for P25 trunking systems, there wasn't for P25 conventional systems - this adds that feature.

The changes are a subset of those in #322.

I've tested this on an analog system as well as a P25 conventional one and verified on both the source ID logging works as expected, starting fresh for each new call.

I did have [... cut]