TrunkRecorder/trunk-recorder #143: Solved: Boost/CMake version mismatch upon make

rosecitytransit, 2017-10-16:
So, I have someone else who is wanting to run trunk-recorder and I am currently setting up a server just for it.  I decided to use a fresh minimal install of Ubuntu from https://help.ubuntu.com/community/Installation/MinimalCD and upon trying to `make` trunk-recorder I get the following (outputs 1 and 2):

`[ 35%] Built target imbe_vocoder
[ 69%] Built target gnuradio-op25_repeater
[ 70%] Linking CXX executable recorder
CMakeFiles/recorder.dir/build.make:585: recipe for target 'recorder' failed
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/recorder.dir/all' failed
Makefile:127: recipe for target 'all' failed`

`CMakeFiles/recorder.dir/trunk-recorder/main.cc.o: In function boost::shared_ptr<boost::log::v2_mt_posix::sinks::synchronous_sink<boost::log::v2_mt_posix::sinks::text_file_backend> > boost::log::v2_mt_posix::aux::add_file_log<boost::parameter::aux::arg_list<boost::parameter::aux::tagged_argument<boost::log::v2_mt_posix::keywords::tag::auto_flush, bool const>, boost::parameter::aux::arg_list<boost::parameter::aux::tagged_argument<boost::log::v2_mt_posix::keywords::tag::time_based_rotation, boost::log::v2_mt_posix::sinks::file::rotation_at_time_point const>, boost::parameter::aux::arg_list<boost::parameter::aux::tagged_argument<boost::log::v2_mt_posix::keywords::tag::rotation_size, int const>, boost::parameter::aux::arg_list<boost::parameter::aux::tagged_argument<boost::log::v2_mt_posix::keywords::tag::file_name, char const [27]>, boost::parameter::aux::arg_list<boost::parameter::aux::tagged_argument<boost::log::v2_mt_posix::keywords::tag::format, char const [25]>, boost::parameter::aux::empty_arg_list> > > > > >(boost::parameter::aux::arg_list<boost::parameter::aux::tagged_argument<boost::log::v2_mt_posix::keywords::tag::auto_flush, bool const>, boost::parameter::aux::arg_list<boost::parameter::aux::tagged_argument<boost::log::v2_mt_posix::keywords::tag::time_based_rotation, boost::log::v2_mt_posix::sinks::file::rotation_at_time_point const>, boost::parameter::aux::arg_list<boost::parameter::aux::tagged_argument<boost::log::v2_mt_posix::keywords::tag::rotation_size, int const>, boost::parameter::aux::arg_list<boost::parameter::aux::tagged_argument<boost::log::v2_mt_posix::keywords::tag::file_name, char const [27]>, boost::parameter::aux::arg_list<boost::parameter::aux::tagged_argument<boost::log::v2_mt_posix::keywords::tag::format, char const [25]>, boost::parameter::aux::empty_arg_list> > > > > const&)':
main.cc:(.text._ZN5boost3log11v2_mt_posix3aux12add_file_logINS_9parameter3aux8arg_listINS5_15tagged_argumentINS1_8keywords3tag10auto_flushEKbEENS6_INS7_INS9_19time_based_rotationEKNS1_5sinks4file22rotation_at_time_pointEEENS6_INS7_INS9_13rotation_sizeEKiEENS6_INS7_INS9_9file_nameEA27_KcEENS6_INS7_INS9_6formatEA25_SN_EENS5_14empty_arg_listEEEEEEEEEEEEENS_10shared_ptrINSE_16synchronous_sinkINSE_17text_file_backendEEEEERKT_[_ZN5boost3log11v2_mt_posix3aux12add_file_logINS_9parameter3aux8arg_listINS5_15tagged_argumentINS1_8keywords3tag10auto_flushEKbEENS6_INS7_INS9_19time_based_rotationEKNS1_5sinks4file22rotation_at_time_pointEEENS6_INS7_INS9_13rotation_sizeEKiEENS6_INS7_INS9_9file_nameEA27_KcEENS6_INS7_INS9_6formatEA25_SN_EENS5_14empty_arg_listEEEEEEEEEEEEENS_10shared_ptrINSE_16synchronous_sinkINSE_17text_file_backendEEEEERKT_]+0x52): undefined reference to boost::log::v2_mt_posix::basic_formatter<char> boost::log::v2_mt_posix::parse_formatter<char>(char const*, char const*)'
CMakeFiles/recorder.dir/trunk-recorder/source.cc.o: In function Source::Source(double, double, double, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, Config*)':
source.cc:(.text+0x45b2): undefined reference to osmosdr::source::make(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
source.cc:(.text+0x4889): undefined reference to osmosdr::source::make(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
source.cc:(.text+0x5021): undefined reference to osmosdr::meta_range_t::values() const'
collect2: error: ld returned 1 exit status
make[2]: *** [recorder] Error 1
make[1]: *** [CMakeFiles/recorder.dir/all] Error 2
make: *** [all] Error 2`

After doing some searching, I found that it seems that Ubuntu installs an older version of CMake which is not compatible with the newer version of Boost that it installs.  Note that I'm using the instructions that I've documented here: https://github.com/robotastic/trunk-recorder/wiki/MinimalInstall

A solution that worked for me was to download and manually use the newest version of CMake from https://cmake.org/download/  Another idea would be to modify the PyBOMBS .lwr file to insist that matching versions be installed.

I can close this, and maybe it should be submitted somewhere else since it technically isn't a trunk-recorder issue, but I'd thought I'd put it here in case anyone else comes across it.

System details:
Ubuntu 17.04 "Zesty Zapus" for 64-bit PC (amd64, x86_64)
Linux 4.10.0-35-generic #39-Ubuntu SMP Wed Sep 13 07:46:59 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
cmake version 3.7.2
Boost version: 1.65.1

rosecitytransit, 2017-10-16:
Additionally, CMake generated this but appeared to work:
```
CMake Warning at /usr/share/cmake-3.7/Modules/FindBoost.cmake:761 (message):
  Imported targets not available for Boost version 106501
Call Stack (most recent call first):
  /usr/share/cmake-3.7/Modules/FindBoost.cmake:865 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.7/Modules/FindBoost.cmake:1470 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:145 (find_package)

CMake Warning at /usr/share/cmake-3.7/Modules/FindBoost.cmake:761 (message):
  Imported targets not available for Boost version 106501
Call Stack (most recent call first):
  /usr/share/cmake-3.7/Modules/FindBoost.cmake:865 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.7/Modules/FindBoost.cmake:1470 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:145 (find_package)

CMake Warning at /usr/share/cmake-3.7/Modules/FindBoost.cmake:761 (message):
  Imported targets not available for Boost version 106501
Call Stack (most recent call first):
  /usr/share/cmake-3.7/Modules/FindBoost.cmake:865 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.7/Modules/FindBoost.cmake:1470 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:145 (find_package)

CMake Warning at /usr/share/cmake-3.7/Modules/FindBoost.cmake:761 (message):
  Imported targets not available for Boost version 106501
Call Stack (most recent call first):
  /usr/share/cmake-3.7/Modules/FindBoost.cmake:865 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.7/Modules/FindBoost.cmake:1470 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:145 (find_package)

CMake Warning at /usr/share/cmake-3.7/Modules/FindBoost.cmake:761 (message):
  Imported targets not available for Boost version 106501
Call Stack (most recent call first):
  /usr/share/cmake-3.7/Modules/FindBoost.cmake:865 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.7/Modules/FindBoost.cmake:1470 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:145 (find_package)

CMake Warning at /usr/share/cmake-3.7/Modules/FindBoost.cmake:761 (message):
  Imported targets not available for Boost version 106501
Call Stack (most recent call first):
  /usr/share/cmake-3.7/Modules/FindBoost.cmake:865 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.7/Modules/FindBoost.cmake:1470 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:145 (find_package)
````

robotastic, 2017-10-16:
From the first error log, it looks like it is having trouble finding the osmosdr drivers. It also looks like there is a problem with the log boost drivers: undefined reference to boost::log::v2_mt_posix::basic_formatter<char>

Have you tried working with 16.04? I haven't done much debugging with 17.04.

I can try to do a Docker install with 17.04 and see what happens. I am going to generally stick with the LTS versions.

rosecitytransit, 2017-10-16:
These are some of the references I found when I was searching the errors I saw which noted the incompatibility:

https://stackoverflow.com/questions/42123509/cmake-finds-boost-but-the-imported-targets-not-available-for-boost-version

https://stackoverflow.com/questions/44563475/cmake-3-8-imported-targets-not-available-for-boost-version

I maybe should have used a LTS version instead of going with the newest version, but the computer is a long ways a way from the now (though does have an unused hard drive if it were possible to install from an existing install).

I'm now getting "Unable to setup a System to record, exiting..." when I was trying to record NOAA Weather Radio to do some reception testing (though it does find the tuner).  I'm going to work on it and set up the trunking configuration and can comment again after that.

rosecitytransit, 2017-10-16:
I got the trunking configuration set up and it looks like it is indeed trying to work (I need to fine tune everything).

rosecitytransit, 2017-10-16:
Off-topic but a minor nit: at startup, trunk-recorder gives the WACN and System ID for P25 system in decimal rather than the hex Radio Reference uses.

closed 2017-10-17 by rosecitytransit