TrunkRecorder/trunk-recorder #144: Using muliple HackRF Ones doesn't work.

Dygear, 2017-10-27:
This is my `hackrf_info` output.

```
[dygear@arch ~]$ hackrf_info 
hackrf_info version: 2017.02.1
libhackrf version: 2017.02.1 (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000014d463dc2f2923e1
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00604f5e

Found HackRF
Index: 1
Serial number: 000000000000000014d463dc2f356ce1
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00654765
```

`config.json`
```
{
    "sources": [
        {
            "driver":         "osmosdr",
            "device":   "hackrf=356ce1",
            "center":       500000000.0,
            "rate":          10000000.0,
            "squelch":              -70,
            "gain":                  40,
            "ifGain":                 8,
            "bbGain":                32,
            "error":               3000,
            "digitalRecorders":       4,
            "modulation":        "qpsk"
        }, {
            "driver":         "osmosdr",
            "device":   "hackrf=2923e1",
            "center":       460000000.0,
            "rate":          20000000.0,
            "squelch":              -70,
            "gain":                  40,
            "ifGain":                 8,
            "bbGain":                32,
            "error":               3000,
            "digitalRecorders":       4,
            "modulation":        "qpsk"
        }
    ],
    "systems": [
        {
            "type":             "p25",
            "shortName":        "NCPD",
            "talkgroupsFile":   "NCPD.csv",
            "control_channels": [502400000, 500287500, 501050000, 501125000]
        }, {
            "type":             "conventional",
            "shortName":        "NCFD",
            "talkgroupsFile":   "NCFD.csv",
            "channels":         [460117000, 460174000]
        }
    ]
}
```

This is the output of the program ...

```
[... 160 lines cut ...]
As you can see around the bottom ...

```
[2017-10-26 21:22:11.550586] (info)   Source Device: hackrf=2923e1
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy 

FATAL: Failed to open HackRF device (-1000) Resource busy

Trying to fill up 1 missing channel(s) with null source(s).
This is being done to prevent the application from crashing
due to gnuradio bug #528.

[2017-10-26 21:22:11.551170] (info)   SOURCE TYPE OSMOSDR (osmosdr)
[2017-10-26 21:22:11.551176] (info)   Setting sample rate to: 2e+07
[2017-10-26 21:22:11.551184] (info)   Actual sample rate: 0
[2017-10-26 21:22:11.551189] (info)   Tunning to 4.60003e+08hz
[2017-10-26 21:22:11.551196] (info)   Max HZ: 4.7e+08
[2017-10-26 21:22:11.551202] (info)   Min HZ: 4.5e+08
[2017-10-26 21:22:11.551208] (info)   IF Gain set to: 0
[2017-10-26 21:22:11.551214] (info)   BB Gain set to: 0
[2017-10-26 21:22:11.551220] (info)   Gain set to: 0
```

The second hackrf fails to open.

Treehouseman, 2017-10-27:
Wait, you're getting successful decode with one hackrf running at 10+mhz? I can't get it to play ball with one large radio.

Dygear, 2017-10-27:
Yeah, I haven't had a problem with 1 HackRF One running at 10MS/s. I don't recall an issue with 20MS/s, I just didn't need that much bandwidth so I didn't bother to keep those settings. I currently have one sitting at 10MS/s at all times decoding a P25 signal from my local police department. If you're having issues with your system, try running the `volk_profile` command to see if you can't get some extra pref out of your system. I'm currently trying to decode the P25 system (That's the first one you see in the list.) And the entire Public Safety Band (UHF I) with the second system. But I'm having little luck in that area getting both HackRFs up and running at once on a single system. I'm then taking all of it and scping the json files and m4a files to a playback server so everyone can listen. (And the NCPD one is uploaded to OpenMHz. *hat tip for that btw*).

Treehouseman, 2017-10-30:
I think in the end my issue is the number of recorders needed, your configuration only has a few, I need at least 40.

Dygear, 2019-03-17:
Yeah, you're probably CPU starved at that point.

closed 2019-03-17 by Dygear