TrunkRecorder/trunk-recorder #146: SIGHUP kills application

vabiro, 2017-11-09:
Hi,

Is there a way to prompt recorder to reload the config files?

I have tried sending a SIGHUP (kill -1 23324), which prompts most programmes to reload their configuration, but that just cause it to kill the application.

Is there another way to prompt it without killing and creating a new process? I am running it inside screen, and it would be much easier to do this than logging in, doing a CTRL-C and then restarting.

Thanks
Victor

vabiro, 2017-11-11:
I have found a workaround by using systemd, which has the added benefit that it starts Trunk Recorder at start of the computer.

For future reference, I created a systemd unit file, which allows me to restart the service using ```# systemctl restart recorder```
The content of the file is:
```
[Unit]
Description=Trunk Recorder Service
After=network.target

[Service]
Type=simple
User=root
WorkingDirectory=/root/recorder
ExecStart=/root/recorder/recorder --config=/path to/config.json >/dev/null
ExecStop= /bin/kill -TERM $MAINPID
RestartSec=5
Restart=on-abort

[Install]
WantedBy=multi-user.target
```
The default on restart commands between kill and restart is 100ms, and I found that it was too quick for the SDRs to catch up. They would report that they were unavailable and mess up the functioning of Trunk Recorder, so I added the ```RestartSec=5```

I hope someone finds this helpful.

Cheers
Victor

closed 2017-11-11 by vabiro

kcwebby, 2017-11-12:
I feel dumb, but I've tried this before.  Without googling it, what do you
have to do to add a new service file to systemd?

vabiro, 2017-11-12:
I can never keep it straight between platforms. I was doing this on a machine running Arch Linux, and used the wiki entry at https://wiki.archlinux.org/index.php/systemd#Writing_unit_files

Basically, create a text file with the above text  in ```/etc/systemd/system/``` called ```something.service``` where "something" is the name of the service. then type something like ```# systemctl reload something```  To enable for at startup enter ```# systemctl enable something```

I hope that helps.

rosecitytransit, 2017-11-13:
Having a true ability to re-load the config file would be nice when the SDRs decide to quit recording or otherwise making minor config changes.  Ideally, any open calls would be gracefully ended, including running the encode/upload script.  I find unconverted WAV files when the program gets killed.

robotastic, 2017-11-13:
Nice fix @vabiro ! I think that is the best approach. That is probably the cleanest way to handle it. There is no easy way to change some of configs without stopping all of the SDRs and recordings.

Out of curiosity, what settings would you be interested in being dynamically changed?

vabiro, 2017-11-13:
Luke, 
The most likely file would be the csv files with the TGIDS. The full config would rarely be changed, and I don't mind restarting for that.

Here's what I am doing: to make things a bit easier to manage, I have put all the TGIDs in a MySQL database (>500 in my area) and created a web interface. When I hit the save button the PHP script creates the csv file and does a ```exec('systemctl enable recorder')``` which reloads the recorder and picks-up the new TGIDS, or stops recording the ones that have been removed.

The problem is, as @rosecitytransit pointed out, the exit is unclean and I lose recordings while it is doing the restart. The ideal situation would be some way to keep it doing its thing and just re-read the csv. 

This does the trick, but is less than ideal.

rosecitytransit, 2017-11-13:
For me at the moment, I'd want to reload the error/PPM (though I am getting better SDRs that should take care of that shortly, another idea would be to have a "drift connect" that tries for the best decode rate).  But at least closing cleanly would be nice, either by sending a "close" command to all open recorders/calls or letting them run and ceasing decoding the trunking channel.

rosecitytransit, 2017-11-21:
Looking at main.cc, it looks like trunk-recorder DOES support a clean shutdown on SIGINT (kill -2 <recorder PID>) by exiting the loop of control channel decoding.