TrunkRecorder/trunk-recorder #154: ENHANCEMENT: Ability to mail recordings or run commands per tg

netsound, 2017-12-16:
Looking for the ability to define certain talkgroups that when files are recorded they are emailed off depending on what talkgroup is recorded... IE paging talkgroups.   Or the ability to run certain commands based on certain talkgroups.

rosecitytransit, 2017-12-17:
The uploadScript option will run a command after each file has completed recording with the filename as a parameter, and the talkgroup is the first part of the filename.  So set it in config.json and create a script that parses the filename and then does what you want.

vabiro, 2017-12-22:
@netsound here is how I am getting the TGID out of the filename in my bash script. Once you have that you can do an ```if``` that will perform the action you need.
```
#! /bin/bash
echo "Encoding: $1" 
filename="$1"
basename="${filename%.*}"

fileid=$(basename $filename)
outdir="/root/recorder/audio/txcoded/"
outname="$fileid.wav"
tgid="${fileid%-*}"
```
I hope this gets you started.

closed 2020-08-04 by robotastic