TrunkRecorder/trunk-recorder #155: Freqency and Conventional Chanel Improvements.

amlethojalen, 2017-12-21:
I'm planning on make a few improvements (IMHO) or customisations, and wondering if you want me to send though some pull request for the changes.

1. On my system the frequencies are shown with exponents, change show them as MHz in the log and console.
2. Show the Talk Group Alpha tag in the log and console near the Talk group Id
3. With a conventional channel, don't create the file until there is recording activity.
4. With a conventional channel, calculate the start and stop times based on recording activity.
5. With a conventional channel, output a 'Listening' into the log during start_recorder, and 'Recording' when there is actual activity.
6. Store Recorder stats for each recorder, 'Recordings Made', 'Recording durations', and possibly a few more, and output them during the print_status. I have currently 1 conventional P25 channel recording, and 8 P25 recorders setup on a trunked system, it would be good to know how often the 8 recorders are actually being used, and weather or not i can do with less of them.
7. Hide non recorded TG's from the console and log
8. Create a simple Json status file that includes current active/recording Talk Groups, including recording durations, this file would get updated every time a recorder is started or stopped. This file can then be consumed by other applications to show active TGs.
9. Add ability to split recording files into blocks (eg each recorded file would be a max of 30 seconds in duration), this would lessen the delay when streaming. Some of the conversations im getting on the trunked system can go for 5 minutes in duration.

My plan is that all of the above changes would be implemented independently, and have configuration options to turn each feature on or off, so i dont upset any existing users.

For the conventional channel changes, i'm planning on creating a call_conventional, p25conventional_recorder which are derived from the existing classes.

Happy to discuss the changes in further details if required.

Regards

Amleth

rosecitytransit, 2017-12-21:
The use of exponents does annoy me a little bit.  Regarding 8, check out issue #1

amlethojalen, 2017-12-21:
yep, 8 does look a lot like #1, i could also incorporate your changes in #153 and bundle them both together.

rosecitytransit, 2017-12-21:
That would be great.  I don't have the change separated out at the moment as I was planning on staging it on [rosecitytransit/trunk-recorder:master](https://github.com/rosecitytransit/trunk-recorder/tree/master) once #151 was merged.

Also, if you or someone else is interested, the other changes in [rosecitytransit/trunk-recorder:kcmscanner](https://github.com/rosecitytransit/trunk-recorder/tree/kcmscanner) are what needs to be skipped in order to build with support for just RTL SDRs and not more expensive devices and be able to skip requirements for those.

amlethojalen, 2017-12-21:
Ok i'll have a bit of a look at what you've done. I'm using an Airspy at the moment but i do have some RTL SDR's but i'm not using them for this project as they dont have the bandwidth.

robotastic, 2017-12-21:
These all sound pretty great! I think it would be good to merge most of this in when you are done, they seem like good general enhancements everyone would love.

For #8, I have actually gotten most of that done. Check out the status-socket branch. Can you work off of that branch?

Right now all of the stats are sent to a  websocket. I have a sample server that can dispaly it all. Maybe we could just extend that to also dump to a .json file.

Agreed on the exponent thing. I think it can be easily fixed, using a fixed(10) command or something like.

bctrainers, 2017-12-21:
As for #6, the user known as Treehouseman here on github made something similar, but in a ncurses UI, https://github.com/Treehouseman/trunk-recorder - i assume you're looking for something similar to that?

Treehouseman, 2017-12-21:
Yeah, for 6 I have that functionality in the nurses interface, I need to update my fork sometime soon, I just havent had much motivation for it since I'm not running the Charlotte scanner anymore.

amlethojalen, 2017-12-21:
Yep the ncurses UI is similar to what i was planning to do for 6, except i was just planning to output the values to a file, that can be consumed by nodejs.
6&8 somewhat overlap in functionality, I'll have a look at the status-socket branch and see if i can expand on that.
When are you planning on merging the status-socket branch?

robotastic, 2017-12-22:
I am going to try and get things merged together over the holidays

vabiro, 2017-12-22:
@robotastic 

> Right now all of the stats are sent to a websocket. I have a sample server that can dispaly it all. 

Is there any documentation on using the websocket, or are there comments in a file that would point me in the right direction?

Thanks

amlethojalen, 2017-12-22:
@robotastic Ill wait for you to complete the merge of the status-socket branch until i look at 6&8

dreinhold, 2017-12-22:
All good stuff, for 9) I would like to see something like that but break when the transmission is over {30} seconds and the next radio ID grant happens.

Dygear, 2017-12-25:
I love all of these items! I can't wait to see them. I've downloaded the 3.0.0 version and I'm going to start playing!

amlethojalen, 2018-01-10:
Ok, been busy of the last few days.
1. is completed and merged into master
2. is completed and merged into master
3. is completed awaiting pull request
4. is completed awaiting pull request
5. is completed awaiting pull request
6. this has been implemented using the status sockets and an external application for the moment, still looking for an easy way to write the state to a file, and then reload the state when the application restarts. Also the ability to break this down per day.
7. is completed awaiting pull request
8. this has been implemented using the status sockets, i dont expect to need to write the information to a file any more, but if someone has the need i'm happy to implement it.
9. i investigated this, it didn't really turn out very well, the splitting worked well, but the encoding and uploading were a bit hit and miss, and resulted in several second pauses between the split files (in the middle of a conversation). I'll try and work out a different way to allow the streaming to be quicker.

closed 2020-08-04 by robotastic