TrunkRecorder/trunk-recorder #157: Modifying wav file sample_rate

vabiro, 2017-12-22:
Hi,
I am trying to change the sample rate of the wav files created and thought I had it by changing line 201 in p25_recorder.cc from:
```  wav_sink = gr::blocks::nonstop_wavfile_sink::make(1, 8000, 16, false);```
to
```  wav_sink = gr::blocks::nonstop_wavfile_sink::make(1, 16000, 16, false);```
but the result is a file that sounds sped up.

Any suggestions on what else needs to be changed to get the correct sample_rate?

BTW, what is the native sample rate coming from the P25 system?

vabiro, 2017-12-26:
The reason that I was playing with this is so that I could better results from CMU Sphinx Speech to Text. I wasn't sure whether the problem was the audio file from GNURadio, or something that I had broken when transcoding using ffmpeg, so I also asked over at the Sourceforge forum at http://bit.ly/2BBCeCa 
I thought the discussion may be useful to others.

Cheers
Victor

robotastic, 2017-12-26:
Unfortunately the sample rate is determined by the IMBE codec. I don't think it can do 16000. You might be able to add a conversion block in there, but there wouldn't be any additional improvement in quality since the source is 8000 samples/sec.

rosecitytransit, 2017-12-26:
Listening to the clip attached to the SourceForge discussion, I see they have a robotic dispatcher.  That should at least be better than trying to decipher human dispatchers, which can vary a lot and can be hard for humans to understand.

closed 2018-01-04 by vabiro

FarvaTechnology, 2018-11-06:
Any updates on this?   I am facing the same issue... Sphinx requires 16 rate for processing... And attempting to UP rate from 8K to 16K results in HORRIFIC processing.

How can we change the audio rate inside trunk-recorder?   Anyone???

FarvaTechnology, 2018-11-13:
Any updates on this?  - I explored the WAV sink component of the GNURADIO libraries and it appears that GNU and P25 can do up to 44100.   But this ticket mentioned that it's a limitation of the IMBE codec... But the IMBE specs should support up to 44100 if I read everything correctly.

Can anyone with more wizard than I regarding p25 and IMBE provide some insight on how we can up this from 8K to at least 16K????    I REALLY want to explore the speech to text capabilities... But I need AT LEAST 16K.

Please advise team... Together we can make great things happen! :)

++Todd

robotastic, 2018-11-28:
The IMBE vocoder is configured for 8000khz for P25. That is what all the radio are encoding at, so it is not something that can be changed on the RX side.
You can try converting it in Audacity, but it won't improve the sound quality