TrunkRecorder/trunk-recorder #161: Build Errors

kkotowick, 2018-01-04:
Trying to upgrade to version 3.0, but running into some build errors I can't figure out. Running Ubuntu 16.04.3, all prerequisite packages installed and fully updated. Any tips on how to get this working?

```
myuser@radio-server:~/radios/trunk-recorder$ make
[ 32%] Built target imbe_vocoder
[ 70%] Built target gnuradio-op25_repeater
[ 71%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/main.cc.o
In file included from /usr/include/openssl/bio.h:20:0,
                 from /usr/include/openssl/conf.h:13,
                 from /usr/include/boost/asio/ssl/detail/openssl_types.hpp:19,
                 from /usr/include/boost/asio/ssl/context_base.hpp:19,
                 from /usr/include/boost/asio/ssl/context.hpp:27,
                 from /usr/include/boost/asio/ssl.hpp:19,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/lib/../trunk-recorder/uploaders/uploader.h:12,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/lib/../trunk-recorder/uploaders/call_uploader.h:4,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/lib/../trunk-recorder/call.h:31,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/lib/gr_blocks/nonstop_wavfile_sink.h:26,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/trunk-recorder/recorders/recorder.h:44,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/trunk-recorder/recorders/analog_recorder.h:42,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/trunk-recorder/source.h:11,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/trunk-recorder/main.cc:42:
/usr/include/boost/asio/ssl/detail/impl/openssl_init.ipp: In constructor ‘boost::asio::ssl::detail::openssl_init_base::do_init::do_init()’:
/usr/include/boost/asio/ssl/detail/impl/openssl_init.ipp:43:23: error: expected id-expression before ‘(’ token
     mutexes_.resize(::CRYPTO_num_locks());
                       ^
In file included from /usr/include/boost/asio/ssl/detail/openssl_init.hpp:100:0,
                 from /usr/include/boost/asio/ssl/context.hpp:29,
                 from /usr/include/boost/asio/ssl.hpp:19,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/lib/../trunk-recorder/uploaders/uploader.h:12,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/lib/../trunk-recorder/uploaders/call_uploader.h:4,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/lib/../trunk-recorder/call.h:31,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/lib/gr_blocks/nonstop_wavfile_sink.h:26,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/trunk-recorder/recorders/recorder.h:44,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/trunk-recorder/recorders/analog_recorder.h:42,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/trunk-recorder/source.h:11,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/trunk-recorder/main.cc:42:
/usr/include/boost/asio/ssl/detail/impl/openssl_init.ipp:46:66: error: expected id-expression before ‘;’ token
     ::CRYPTO_set_locking_callback(&do_init::openssl_locking_func);
                                                                  ^
/usr/include/boost/asio/ssl/detail/impl/openssl_init.ipp:47:56: error: expected id-expression before ‘;’ token
     ::CRYPTO_set_id_callback(&do_init::openssl_id_func);
                                                        ^
/usr/include/boost/asio/ssl/detail/impl/openssl_init.ipp: In destructor ‘boost::asio::ssl::detail::openssl_init_base::do_init::~do_init()’:
/usr/include/boost/asio/ssl/detail/impl/openssl_init.ipp:64:32: error: expected id-expression before ‘;’ token
     ::CRYPTO_set_id_callback(0);
                                ^
/usr/include/boost/asio/ssl/detail/impl/openssl_init.ipp:65:37: error: expected id-expression before ‘;’ token
     ::CRYPTO_set_locking_callback(0);
                                     ^
In file included from /usr/include/openssl/engine.h:30:0,
                 from /usr/include/boost/asio/ssl/detail/openssl_types.hpp:22,
                 from /usr/include/boost/asio/ssl/context_base.hpp:19,
                 from /usr/include/boost/asio/ssl/context.hpp:27,
                 from /usr/include/boost/asio/ssl.hpp:19,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/lib/../trunk-recorder/uploaders/uploader.h:12,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/lib/../trunk-recorder/uploaders/call_uploader.h:4,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/lib/../trunk-recorder/call.h:31,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/lib/gr_blocks/nonstop_wavfile_sink.h:26,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/trunk-recorder/recorders/recorder.h:44,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/trunk-recorder/recorders/analog_recorder.h:42,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/trunk-recorder/source.h:11,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/trunk-recorder/main.cc:42:
/usr/include/boost/asio/ssl/detail/impl/openssl_init.ipp:66:7: error: expected id-expression before ‘while’
     ::ERR_free_strings();
       ^
In file included from /usr/include/openssl/x509.h:23:0,
                 from /usr/include/openssl/ssl.h:50,
                 from /usr/include/boost/asio/ssl/detail/openssl_types.hpp:20,
                 from /usr/include/boost/asio/ssl/context_base.hpp:19,
                 from /usr/include/boost/asio/ssl/context.hpp:27,
[... 635 lines cut ...]
                ^
In file included from /usr/include/openssl/engine.h:30:0,
                 from /usr/include/boost/asio/ssl/detail/openssl_types.hpp:22,
                 from /usr/include/boost/asio/ssl/context_base.hpp:19,
                 from /usr/include/boost/asio/ssl/context.hpp:27,
                 from /usr/include/boost/asio/ssl.hpp:19,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/lib/../trunk-recorder/uploaders/uploader.h:12,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/lib/../trunk-recorder/uploaders/call_uploader.h:4,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/lib/../trunk-recorder/call.h:31,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/lib/gr_blocks/nonstop_wavfile_sink.h:26,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/trunk-recorder/recorders/recorder.h:44,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/trunk-recorder/recorders/analog_recorder.h:42,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/trunk-recorder/source.h:11,
                 from /home/myuser/radios/trunk-recorder/trunk-recorder/trunk-recorder/main.cc:42:
/usr/include/boost/asio/ssl/detail/impl/engine.ipp:221:9: error: ‘SSL_R_SHORT_READ’ was not declared in this scope
         ERR_PACK(ERR_LIB_SSL, 0, SSL_R_SHORT_READ),
         ^
CMakeFiles/recorder.dir/build.make:62: recipe for target 'CMakeFiles/recorder.dir/trunk-recorder/main.cc.o' failed
make[2]: *** [CMakeFiles/recorder.dir/trunk-recorder/main.cc.o] Error 1
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/recorder.dir/all' failed
make[1]: *** [CMakeFiles/recorder.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

```

vabiro, 2018-01-04:
Just a guess, but did you run a system update, especially for openssl and it's dependencies.

It seems counterintuitive (i.e I wouldn't think of a radio app requiring it), but a lot of the problems seem related to it and encryption libraries.

rosecitytransit, 2018-01-04:
Is your plan to upload recordings to OpenMHZ?  If not, would it be possible to just remove the upload code?

kkotowick, 2018-01-05:
Figured it out. I had version 1.1.0g of `openssl` and `libssl-dev` installed from a ppa (current stable version on Ubuntu is 1.0.2g). I downgraded them to 1.0.2g and it built properly. Looks like the current version of boost doesn't support `openssl` 1.1.0 yet.

robotastic, 2018-01-05:
Oh weird! I am happy you found the root cause. Boost seems to be the cause of and solution to all C++ problems.

closed 2020-08-04 by robotastic