TrunkRecorder/trunk-recorder #162: Only 44-byte WAV files are being recorded

beaugunderson, 2018-01-04:
I tried recording [KCERS](https://www.radioreference.com/apps/db/?sid=604) with an RTL-SDR and then switched to a HackRF and only get 44-byte WAV files.

Looking through previous issues It seems like this might be an issue with reception, though I can hear calls come through using gqrx.

Here's my current `config.json`:

```json
{
	"sources": [{
		"center": 855000000,
		"rate": 10000000,
		"error": -12000,
		"gain": 0,
		"ifGain": 24,
		"bbGain": 30,
		"debugRecorders": 1,
		"digitalRecorders": 12,
		"modulation": "fsk4",
		"driver": "osmosdr",
		"device": "hackrf=768587"
	}],
	"systems": [{
		"control_channels": [
			853175000,
			851412500,
			853675000,
			854362500
		],
		"bandplan": "800_reband",
		"type": "smartnet",
		"shortName": "KCERS"
	}],
	"frequencyFormat": "mhz"
}
```

I tried with and without `"800_reband"` (there's a note on the linked radioreference page above re: rebanding).

[Here are some debug recordings](https://github.com/robotastic/trunk-recorder/files/1602974/Archive.zip) if that's useful.

rosecitytransit, 2018-01-04:
Are you at least getting good reception on the control channel?  Look for 'Control Channel Decode Rate' messages and/or adjust the controlWarnRate option.

Also, have you tried changing the different gain settings?  I attempted to convert the recordings to WAV files that could be opened in SDR Sharp using [Marcin648/iqToSharp: Converter RTL-SDR iq to SDRSharp WAV format.](https://github.com/Marcin648/iqToSharp) and the gain looked very high, and data might be getting cut off at the top.  (What is the sample rate of the debug files? Am I correct that they're the sliver of spectrum that contains just the actual voice channel?)

bctrainers, 2018-01-04:
This issue exists on the osmosdr driver profile as far as i can see. Generic RTLSDR's (R820T2's) exhibit the same issue you describe. From my findings, any/combo of the following can trigger 44-byte files. And yes, it also causes issues with legitimate recordings (more than 44 bytes of audio data recorded) becoming 44 byte files.

- The frequency that the SDR retuned to ended up being offset on the receiver. 

- The talkgroup in use ended up with two people keying up at or near the same time (within one second of another).

- Slightly on the edge of the frequency range of the SDR reciever, possibly due to signal drifting.

- One of the receivers via the osmosdr driver has severe frequency drifting going on or is at capacity. (frequency off-set due to heat buildup on the oscillator)

- A very notable example here: Let's say one SDR is centered correctly on the frequency chosen by the CC for voice, while the second SDR has frequency drift ongoing. The second SDR activates because the transmission ends up on another frequency group of the RTLSDR's, when this happens, the recording is 44-bytes. On the ncurses variant of trunk-recorder, trunk-recorder is still "recording" and logs it fully - as shown by the json file generated. But once the trunk-recorder hits an osmosdr driver configuration in the array with drifting or some sort of issue, the file 44 bytes and halts pushing in new data. Sometimes, the trunk-recorder application will make a new wav file, however more often than not, it will not. The robotastic original variant does this same behavior. 

It's an issue that has plagued me for quite some time. I've 12 RTL SDR's now (I should have bought a hackRF long ago at this point! :D), the system I monitor requires five RTL SDR's. The five SDR's recording the system still has two SDR's with an inane amount of drifting going on - even with a fan on it 24/7. Applying a PPM or frequency off-set does not resolve the issue, but ends up making things worse.

rosecitytransit, 2018-01-05:
I'm thankful that the two systems I've set up Trunk Recorder for both have their frequencies in small ranges that can be handled by two SDRs each.  I've also switched to the non-drifting SDRs as trying to adjust those second ones is indeed a pain,

robotastic, 2018-01-05:
@bctrainers - interesting! I think there a few issues at play here. Maybe we can try breaking them apart:

1. HackRFs and RTL do not have perfect reception across their entire reception range. There is a large spike in the center and things drop off at the top and bottom of the range. Not too much you an do about this. It could be possible to put some sort of filter on top of the entire range, but I think it would have to be customized for each SDR and it probably changes with heat.

2. Freq drift from heat. This is at the HW level. Better ventilation or more expensive HW that is more stable might solve this. 

3. Two recordings keying up on the same talkgroup, at the same time. This is probably worth diving into a bit. There could be something to fix in Trunk Recorder. I wonder if the same filename ends up being used by 2 recorders at the same time.

3. Switching from one source to another causes problems. This is also probably something we can fix. Here is what is supposed to happen: When the conversation on a talkgroup moves frequencies, a check happens to see if the frequency is covered by the SDR currently being used. If it is, the recorder moves to the new frequency. If the new freq is outside the range covered by the SDR, the current recording is stopped, the current file is closed and uploaded. A new recordings is then started on the SDR that is covering that freq. If that second SDR is drifting or not receiving well, the second recording will probably suck, but the original recording on the good SDR should be fine. Is something different than this happening?

bctrainers, 2018-01-13:
One more thing to note of, and this one is probably a doozey to really fix. 

I currently have my RTLSDR's on trunk-recorder set to monitor two different P25 systems in the county. One tower is a single-site-per-county known as KSICS - [Monitoring the JoCo ICS tower](https://www.radioreference.com/apps/db/?siteId=9022). The other is configured to [KC MARRS JoCo, a six site system currently](https://www.radioreference.com/apps/db/?fccCallsign=WQKX487). 

The six site system _consistently_ is throwing out 44 byte audio files, but has a full json log of the transmitting Radio ID's, etc. Which points to some issue with multi-path/simulcast distortion. This is also causing some recorded wav files to have cross-contamination of two different talkgroups in the same recording. For example, I would hear a random Hospital report come over the recording for about a second or two, then instantly swap to the properly recorded talkgroup ID. 

On the KSICS tower that is configured, this issue does not exist what so ever. And the only time that there is an issue with the KSICS tower is when i am messing with cables/antennas. :-)

JustApu, 2018-07-08:
Any additional suggestions to deal with this "44 byte" problem? I too am getting good info in the JSON files and listening with a real radio all the starts and stops match up well. But I'm getting mostly 44 byte recordings.

I'm trying to monitor [New Jersey's P25 system (specifically Site 7)](https://www.radioreference.com/apps/db/?sid=7021) using four [NooElec NESDR Nano 2+ dongles](https://smile.amazon.com/dp/B01B4L48QU). Simulcast could certainly be part of the problem (scanners often have issues with it) but op25 tracks the system without problems, albeit using only a single dongle and monitoring a single talkgroup at any given time.

[config.json.txt](https://github.com/robotastic/trunk-recorder/files/2173250/config.json.txt)

kg6uyz, 2018-07-08:
@JustApu , I had the same issue in the beginning, i was told it could be frequency drift on the dongle and or poor reception of my simulcast system i was monitoring.  I replaced my [origional RTL-SDR dongles](https://www.amazon.com/JahyShow®-RTL2832U-RTL-SDR-Receiver-Compatible/dp/B01H830YQ6/ref=sr_1_16?s=electronics&ie=UTF8&qid=1531017628&sr=1-16&keywords=r820t2) that do drift and need "error" correction entered in the trunking-recorder config file which i had entered from the begining.  I eventually moved over to more accurate dongles ([this one](https://www.amazon.com/NooElec-NESDR-XTR-Extended-Range-TCXO-Based/dp/B014OH57S4/ref=sr_1_21?s=electronics&ie=UTF8&qid=1531017336&sr=1-21&keywords=nooelec+nesdr) and [this one](https://www.amazon.com/NooElec-NESDR-SMArt-XTR-SDR/dp/B06Y1HKLHY/ref=sr_1_20?s=electronics&ie=UTF8&qid=1531017336&sr=1-20&keywords=nooelec+nesdr)) which had no frequency error, i had better luck with them as they were more accurate but i still had issues, i ended up putting some RF window filters in front of the dongles and a preamp so my dongle thats recording my vhf simulcast system was only seeing 150mhz to 160mhz.  I still had an issue where my recorder would stop decoding the control channel and ultimately i ended up replacing my antenna with something much better and that seemed to fix all my problems and i have been running issue free for the most part for along time now.

JustApu, 2018-07-08:
Thanks, @kg6uyz. Its certainly possible a better antenna could help. The NESDR dongles are the 0.5PPM variety but I am using the antennae that came with them so its not tuned specifically for 700 & 800 MHz reception. I did notice that having the antennae close to each other was interfering and spreading them out a bit helped with control channel reception.

Is there any way to get a signal strength measurement from the various dongles or some other indication on how well each is receiving?

kg6uyz, 2018-07-08:
@JustApu , My antenna issue was using an antenna that was not tuned for the commerical frequency band.  Measuring signal strenth will not really tell you how well your antennas are tuned.  I suggest you get something tuned for the 700/800mhz band.  The telescopic antennas included are not really good to use.  Also if you dont have any error correction setup in trunk-recorder i suggest you connect your dongles to a program like SDR Console, tune into the control channel receive frequency and verify if the spectrum graph lines up with the frequency tuned. It looks like your dongles are like the old ones i used to use (different brand but same chips) which drift.

JustApu, 2018-07-08:
Thanks, @kg6uyz. I'll see what I can do about the antennae. I do have the model dongle on an op25 install (same 700/800 MHz system) with no drift over months of uptime and another monitoring a conventional UHF channel. But, again, only one dongle per system there.

mndmr, 2018-07-11:
I had the same issue with my AirSpy, I later found out it was the USB issues and had to get a new USB card. Try running "hackrf_transfer -s 10000000 -f 855000000 -r /dev/null" and make sure you can transfer the full rate.

bctrainers, 2018-07-11:
Curious, what should the ideal values be? Mine is sitting at 19.9MiB/Sec to 20.2MiB/Sec.

mndmr, 2018-07-11:
Looks fine for the 10MHz sample rate. You could try and add to your config.json 
"controlWarnRate": -1,

above your frequencyFormat line and see if you get a decode rate. A count of 100+ is ideal (at least for p25 never used smartnet). If its low try moving the antenna to a new location.

This is what you would see (example):
[date/time] (error)   [shortName]         Control Channel Message Decode Rate: 40.3333/sec, count:  121

bctrainers, 2018-07-11:
Went ahead and bumped it up to 16MHz and 20MHz of coverage on my HackRF One. 

At 20MHz, ushing 40MB/sec nicely. Odd thing is with the systems I monitor, both are near-exact linear to my antenna. The omni antenna has issues while the yagi for the 806-896mhz range picks it up almost too strongly. Fun stuff. :)

billatq, 2018-11-03:
@beaugunderson: Stumbled across this while looking to see if anyone had put together a config for KCERS already. This system is a Motorola Type II SmartNet. If you just replace the digital recorders with analog recorders, the 44 byte files go away and everything is happy.

FarvaTechnology, 2018-11-27:
I am currently testing a theory on the 44 byte issue - It got me too. :(

I believe the issue comes about when you have multiple SDR cards and you overlap your frequency boundaries.    I say this because when it comes time to record, the code checks the frequency that the traffic is about to happen on - Then it attempts to find a card for that frequency.

If you have two cards overlapping freq - Then the object being returned to the caller is 2 objects instead of one.   I believe this results in the corruption.

I have re-configured my freq spread and so far so good...   I will update if I have success.

closed 2020-08-04 by robotastic