TrunkRecorder/trunk-recorder #165: Squelch per system

amlethojalen, 2018-01-10:
I'm looking at adding a squelch at a system level. Any objections, or better suggestions?

I'm monitoring two conventional channels that need a slightly different squelch, so i was going to add them as two different systems each with their own squelch.

My idea was that it can be optionally defined at the source level, and optionally overridden at the system level.

Regards

Amleth

robotastic, 2018-01-10:
I think I may have looked into this... If I remember right it is not easy to dynamically change the squelch levels on a recorder, it can only be done at creation time.

Actually... nevermind, it looks like it is totally doable: https://gnuradio.org/doc/doxygen/classgr_1_1analog_1_1pwr__squelch__cc.html

Yea this could be a good change!

amlethojalen, 2018-01-10:
I was looking at implementing this specifically for conventional channels, unless i'm not reading it correctly, each frequency in a conventional system has its own recorder, so all i was going to do when creating the recorder is use the system squelch, if there was one, otherwise default to the source squelch.

For non conventional calls i guess the squelch would be set when the recorder starts.

ginomiglio, 2020-09-27:
This should be addressed by #394, right?

closed 2021-04-28 by robotastic