TrunkRecorder/trunk-recorder #166: LimeSDR support

kswanson71, 2018-01-13:
Anyone running a LimeSDR successfully?  Hitting a wall where trunk-recorder is unable to setup a system..

[2018-01-13 00:51:06.646981] (info)   Using Config file: config-cobb-lime.json

[2018-01-13 00:51:06.647432] (info)   
-------------------------------------
SYSTEMS
-------------------------------------
0
[2018-01-13 00:51:06.647464] (info)   

System Number: 1
-------------------------------------
1
[2018-01-13 00:51:06.647480] (info)   Short Name: CobbP25
[2018-01-13 00:51:06.647491] (info)   System Type: p25
[2018-01-13 00:51:06.647507] (info)   Control Channels: 
[2018-01-13 00:51:06.647544] (info)   8.57212e+08 
[2018-01-13 00:51:06.647572] (info)   API Key: 
[2018-01-13 00:51:06.647586] (info)   Upload Script: 
[2018-01-13 00:51:06.647600] (info)   Call Log: true
[2018-01-13 00:51:06.647616] (info)   Audio Archive: true
[2018-01-13 00:51:06.647629] (info)   Loading Talkgroups...
[2018-01-13 00:51:06.648596] (info)   Read 219 talkgroups.
[2018-01-13 00:51:06.648614] (info)   Talkgroups File: CobbTrunkList.csv
[2018-01-13 00:51:06.648627] (info)   Record Unkown Talkgroups: true
[2018-01-13 00:51:06.648643] (info)   Talkgroup Display Format: Id
[2018-01-13 00:51:06.648661] (info)   
[2018-01-13 00:51:06.648675] (info)   

-------------------------------------
SOURCES
-------------------------------------

[2018-01-13 00:51:06.648688] (info)   Frequency format: 0
[2018-01-13 00:51:06.648719] (info)   Driver: osmosdr
[2018-01-13 00:51:06.648731] (info)   Center: 8.540000e+07
[2018-01-13 00:51:06.648759] (info)   Rate: 5000000
[2018-01-13 00:51:06.648777] (info)   Error: 0
[2018-01-13 00:51:06.648789] (info)   PPM Error: 0
[2018-01-13 00:51:06.648805] (info)   Gain: 0
[2018-01-13 00:51:06.648817] (info)   IF Gain: 0
[2018-01-13 00:51:06.648829] (info)   BB Gain: 0
[2018-01-13 00:51:06.648841] (info)   LNA Gain: 0
[2018-01-13 00:51:06.648854] (info)   MIX Gain: 0
[2018-01-13 00:51:06.648866] (info)   Squelch: 0
[2018-01-13 00:51:06.648877] (info)   Idle Silence: false
[2018-01-13 00:51:06.648889] (info)   Digital Recorders: 4
[2018-01-13 00:51:06.648902] (info)   Debug Recorders: 0
[2018-01-13 00:51:06.648916] (info)   Analog Recorders: 0
[2018-01-13 00:51:06.648929] (info)   Modulation: qpsk
[2018-01-13 00:51:06.648942] (info)   Source Device: soapy=0,driver=lime
gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp uhd plutosdr miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya 
[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 9062000C4381A'
[INFO] Reference clock 30.720 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] Init LMS7002M(0)
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Disable
CGEN: Freq=80 MHz, VCO=2.56 GHz, INT=82, FRAC=349525, DIV_OUTCH_CGEN=15
CGEN: Freq=61.44 MHz, VCO=2.4576 GHz, INT=79, FRAC=0, DIV_OUTCH_CGEN=19
CGEN: Freq=61.44 MHz, VCO=2.4576 GHz, INT=79, FRAC=0, DIV_OUTCH_CGEN=19
M=156, N=3, Fvco=1040.000 MHz
16: AF AA 2A
19: AB 5A 55
22: FF FF FF
phase: min 31.2; max 197.3; selected 114.2)
M=156, N=3, Fvco=1040.000 MHz
M=156, N=3, Fvco=1040.000 MHz
16: AA 52 D5
phase: min 5.2; max 176.5; selected 90.9)
M=156, N=3, Fvco=1040.000 MHz
M=156, N=3, Fvco=1040.000 MHz
16: A1 AA 2A
16: AA DA 5D
19: 00 F0 FF
phase: min 31.2; max 197.3; selected 114.2)
M=156, N=3, Fvco=1040.000 MHz
M=156, N=3, Fvco=1040.000 MHz
16: 57 DC 3F
16: AA 58 75
16: AA 5A D5
phase: min 15.6; max 171.3; selected 93.5)
M=156, N=3, Fvco=1040.000 MHz
MCU programming : 16384/16384
MCU Programming finished, 895 ms
CGEN: Freq=61.44 MHz, VCO=2.4576 GHz, INT=79, FRAC=0, DIV_OUTCH_CGEN=19
CGEN: Freq=61.44 MHz, VCO=2.4576 GHz, INT=79, FRAC=0, DIV_OUTCH_CGEN=19
M=156, N=3, Fvco=1040.000 MHz
phase: min 5.2; max 360.0; selected 182.6)
M=156, N=3, Fvco=1040.000 MHz
M=156, N=3, Fvco=1040.000 MHz
phase: min 5.2; max 360.0; selected 182.6)
M=156, N=3, Fvco=1040.000 MHz
M=156, N=3, Fvco=1040.000 MHz
phase: min 5.2; max 360.0; selected 182.6)
M=156, N=3, Fvco=1040.000 MHz
M=156, N=3, Fvco=1040.000 MHz
phase: min 5.2; max 360.0; selected 182.6)
M=156, N=3, Fvco=1040.000 MHz
[2018-01-13 00:51:10.236759] (info)   SOURCE TYPE OSMOSDR (osmosdr)
[2018-01-13 00:51:10.236796] (info)   Setting sample rate to: 5000000
CGEN: Freq=640 MHz, VCO=2.56 GHz, INT=82, FRAC=349525, DIV_OUTCH_CGEN=1
M=105, N=1, Fvco=1050.000 MHz
phase: min 11.1; max 189.4; selected 100.3)
M=105, N=1, Fvco=1050.000 MHz
M=105, N=1, Fvco=1050.000 MHz
16: AA 5A D5
phase: min 5.6; max 172.7; selected 89.1)
M=105, N=1, Fvco=1050.000 MHz
[2018-01-13 00:51:10.572533] (info)   Actual sample rate: 5000000
[2018-01-13 00:51:10.572630] (info)   Tunning to 8.540000e+07
[2018-01-13 00:51:10.591379] (info)   Gain Stage: TIA supported values: 0 1 2 3 4 5 6 7 8 9 10 11 12 
[2018-01-13 00:51:10.591499] (info)   Gain Stage: LNA supported values: 0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 
[2018-01-13 00:51:10.591603] (info)   Gain Stage: PGA supported values: -12 -11 -10 -9 -8 -7 -6 -5 -4 -3 -2 -1 0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 
[2018-01-13 00:51:10.591625] (info)   Max Freqency: 8.790000e+07
[2018-01-13 00:51:10.591658] (info)   Min Freqency: 8.290000e+07
[2018-01-13 00:51:10.591697] (info)   Gain set to: 9
[2018-01-13 00:51:10.605150] (info)   	 P25 Recorder Initial Rate: 500000 Resampled Rate: 62500 Initial Decimation: 10 Decimation: 8 System Rate: 60000 ARB Rate: 0.96
[2018-01-13 00:51:10.605183] (info)   P25 Recorder Taps - initial: 757 channel: 1515 ARB: 757 Total: 3029
Project 25 IMBE Encoder/Decoder Fixed-Point implementation
Developed by Pavel Yazev E-mail: pyazev@gmail.com
Version 1.0 (c) Copyright 2009
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; see the file ``LICENSE'' for details.
[2018-01-13 00:51:10.617245] (info)   	 P25 Recorder Initial Rate: 500000 Resampled Rate: 62500 Initial Decimation: 10 Decimation: 8 System Rate: 60000 ARB Rate: 0.96
[2018-01-13 00:51:10.617270] (info)   P25 Recorder Taps - initial: 757 channel: 1515 ARB: 757 Total: 3029
[2018-01-13 00:51:10.628119] (info)   	 P25 Recorder Initial Rate: 500000 Resampled Rate: 62500 Initial Decimation: 10 Decimation: 8 System Rate: 60000 ARB Rate: 0.96
[2018-01-13 00:51:10.628161] (info)   P25 Recorder Taps - initial: 757 channel: 1515 ARB: 757 Total: 3029
[2018-01-13 00:51:10.639138] (info)   	 P25 Recorder Initial Rate: 500000 Resampled Rate: 62500 Initial Decimation: 10 Decimation: 8 System Rate: 60000 ARB Rate: 0.96
[2018-01-13 00:51:10.639163] (info)   P25 Recorder Taps - initial: 757 channel: 1515 ARB: 757 Total: 3029
[2018-01-13 00:51:10.643178] (info)   
-------------------------------------

[2018-01-13 00:51:10.643194] (info)   

-------------------------------------
INSTANCE
-------------------------------------

[2018-01-13 00:51:10.643243] (info)   Capture Directory: /home/kevin/trunk-record
[2018-01-13 00:51:10.643255] (info)   Upload Server: 
[2018-01-13 00:51:10.643276] (info)   Status Server: 
[2018-01-13 00:51:10.643287] (info)   Instance Key: 
[2018-01-13 00:51:10.643298] (info)   Instance Id: 
[2018-01-13 00:51:10.643310] (info)   Default Mode: digital
[2018-01-13 00:51:10.643323] (info)   Call Timeout (seconds): 3
[2018-01-13 00:51:10.643336] (info)   Log to File: false
[2018-01-13 00:51:10.643349] (info)   Control channel warning rate: 10
[2018-01-13 00:51:10.643369] (info)   [CobbP25]	Started with Control Channel: 8.572125e+08
[2018-01-13 00:51:10.643392] (error)   Unable to setup a System to record, exiting...

libusb: warning [libusb_exit] application left some devices open

robotastic, 2018-01-13:
I have one I have been meaning to try. If no one else has tried, I can fire it up.

kswanson71, 2018-01-15:
After digging a little deeper, I was able to adjust the gain on the LimeSDR but I believe that the issue may still be in the device and antenna parameters given that the Lime supports two TX and two RX channels but also has a high, low and wide setting for each RX channel.

Not clear on how to pass these parameters properly but it would explain why it can't setup anything... it can't hear anything. :-)

grutz, 2018-02-10:
The LimeSDR does work with SoapyOsmo. IIRC you can pass device-specific settings not supported by trunk-recorder through the "device" line but I didn't have to do any of that. Use the low freq RX antenna (LNAL).

jareklupinski, 2019-01-21:
I'm using LimeSDR through SoapySDRServer and this branch, pulling in 5/5 :) https://github.com/robotastic/trunk-recorder/pull/216

leee, 2021-05-15:
@robotastic I think this issue can be closed since you merged that PR last year, although there's a comment in that PR that you may wish to review.

robotastic, 2021-05-16:
Agreed - The LimeSDR code may have also changed significantly since then too. I will try to confirm things work with my LimeSDR.

closed 2021-05-16 by robotastic