TrunkRecorder/trunk-recorder #168: Add / Subtract Channels by Status Socket

Dygear, 2018-01-17:
I’d like to be able to add and subtract channels, talk groups, frequencies on a live system by using the status socket interface. As the HackRF listens to 20MHz of bandwidth simultaneously I can pretty much cover the entire public safety spectrum from one location. With that being said as departments come onto the system I’d love to be able to provide them an interface to add their frequencies and have it automatically start recording for them.

robotastic, 2018-01-19:
I was looking at adding in support to mess around some of the source tuning, like gain and tuning error. Are you looking to add conventional systems or trunked systems? Either way it would probably make more sense to simply restart when adding new systems. With Gnuradio, there is no easy way to alter the graph of things without stopping and starting.

robotastic, 2020-08-04:
I am working on adding in Control and Status over MQTT that might help solve some of this... still has a long way to go though.

closed 2020-08-04 by robotastic