TrunkRecorder/trunk-recorder #172: Private Call wildcard

vabiro, 2018-01-25:
Hi,
Is there a way to use a wildcard to receive private calls on a system in the same way that Uniden scanners can use ".0"?

Right now I have to add ~400 TGIDs (harvested from DSD+ logs) to the csv files, and this seems a bit inefficient, as well as prone to errors/omissions.

Is there something in the messaging headers that trunk-recorder could watch for that indicates that it is a private message?

rosecitytransit, 2018-01-27:
I believe that if you set `recordUnknown` to true it will record everything,

vabiro, 2018-01-27:
@rosecitytransit That would work, but it would also capture a lot of traffic that is of no interest. This is a very busy system, and would grab traffic from dozens of unencrypted TGIDs

closed 2020-08-04 by robotastic