TrunkRecorder/trunk-recorder #173: Transition from RTLSDR "bundle" to a HackRF One - Decode Rate 1/sec

bctrainers, 2018-01-31:
Hey, 

My HackRF One came in the other day, got it hooked up this evening with everything configured with what i believe is correct. 

Gave it a good spin on my windows machine with SDR#, didn't notice notable drift - even after six hours of running. 

However, with the latest trunk-recorder revision, this is the result....

[image]

```
root@box006:/home/radio/trunk-build# hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: [redacted]56b35f
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x006e4756
```

The config.json file...

```json
{
    "sources": [{
        "center": 855100000,
        "rate": 16000000,
        "error": 0,
        "ppm": 0,
        "gain": 48,
        "ifGain": 24,
        "bbGain": 28,
        "debugRecorders": 0,
        "digitalRecorders": 16,
        "digitalLevels": 8,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "hackrf=56b35f"
    }],
    "systems": [{
        "control_channels": [853775000],
        "sysId": "688",
        "sysSite": "401",
        "type": "p25",
        "shortName": "MJoCo",
        "recordUnknown": "true",
        "callLog": "true",
        "audioArchive": "true",
        "autoRetune": 1,
        "talkgroupsFile": "../trunk-build-thm/kc_marrs_joco_tgs.csv"
    }],
    "captureDir": "/home/radio/trunk-build/recs/",
    "callTimeout": 2,
    "logFile": 1
}
```

Some of the config.json variables may have residual from the tree-house-man trunk-recorder fork. 

The text output from running ./recorder...

```
root@box006:/home/radio/trunk-build# ./recorder
linux; GNU C++ version 6.2.0 20161027; Boost_106200; UHD_003.009.005-0-unknown

[2018-01-30 23:22:42.609293] (info)   Using Config file: ./config.json

[2018-01-30 23:22:42.609676] (info)
[... 111 lines cut ...]
[2018-01-30 23:22:42.718030] (info)   Upload Server:
[2018-01-30 23:22:42.718041] (info)   Status Server:
[2018-01-30 23:22:42.718051] (info)   Instance Key:
[2018-01-30 23:22:42.718062] (info)   Instance Id:
[2018-01-30 23:22:42.718089] (info)   Default Mode: digital
[2018-01-30 23:22:42.718117] (info)   Call Timeout (seconds): 2
[2018-01-30 23:22:42.718129] (info)   Log to File: true
[2018-01-30 23:22:42.718138] (info)   Control channel warning rate: 10
[2018-01-30 23:22:42.718162] (info)   Control channel retune limit: 0
[2018-01-30 23:22:42.718171] (info)   Frequency format: 0
[2018-01-30 23:22:42.718195] (info)   Status as String: false
[2018-01-30 23:22:42.718322] (info)   [MJoCo]   Started with Control Channel: 8.537750e+08
[2018-01-30 23:22:42.722189] (info)      P25 Trunking - SysNum: 0
[2018-01-30 23:22:42.722230] (info)      P25 Recorder Initial Rate: 484848 Resampled Rate: 48484.8 Initial Decimation: 33 Decimation: 10 System Rate: 48000 ARB Rate: 0.99
[2018-01-30 23:22:42.722318] (info)   P25 Recorder Taps - initial: 2425 channel: 1101 ARB: 735 Total: 4261
[2018-01-30 23:22:45.012223] (error)   [MJoCo]  There is only one control channel defined
[2018-01-30 23:22:45.012391] (error)   [MJoCo]   Control Channel Message Decode Rate: 0.666667/sec, count:  2
[2018-01-30 23:22:48.015626] (error)   [MJoCo]  There is only one control channel defined
[2018-01-30 23:22:48.016044] (error)   [MJoCo]   Control Channel Message Decode Rate: 1/sec, count:  3
[2018-01-30 23:22:51.018783] (error)   [MJoCo]  There is only one control channel defined
[2018-01-30 23:22:51.018959] (error)   [MJoCo]   Control Channel Message Decode Rate: 0.666667/sec, count:  2
^C
 Signal caught!
[2018-01-30 23:22:51.817919] (info)   stopping flow graph
```

bctrainers, 2018-01-31:
I'm an idiot... didn't apply the error correct right. Should've been +12000 and not -12000.

Can be marked as resolved. :)

Time to see how much of a workhorse this HackRF One really is!

bctrainers, 2018-01-31:
There appears to be a bug with the driver "unlatching" when trunk-recorder is CTRL C'd or when its SIGTERM'd. There's previous bug reports regarding this as well.

EricTendian, 2018-02-01:
@bctrainers try compiling and using this executable to reset the HackRF, I've had success with it: https://gist.github.com/x2q/5124616

Usage: `usbreset /dev/bus/usb/099/099`

You can find what path your HackRF is at (the argument to usbreset) with a command like this: `lsusb -d 1d50:6089 | perl -n -e '/Bus ([0-9]*) Device ([0-9]*)/ && print "/dev/bus/usb/$1/$2\n"'`

Dygear, 2018-07-12:
I too am having this problem. Seems that the HackRF is actually at fault from what I can tell. If you watch it in GQRX you can see the device actually freeze after a certain amount of time. Looks like the clock can't handle the long term heat that the HackRF makes.

bctrainers, 2018-07-12:
@Dygear what i did on my HackRF One was open up the case cover and apply a few heatsinks on all of the chips. It's kept the excessive heat generation at bay, and the drift isn't too bad after 6 hours of run time.

[image]

dlaw, 2019-10-30:
Just want to note my experience here in case any others have the same issue:
I had trunk-recorder working great with my HackRF. The next day, exactly the same configuration was no longer working (it failed with the same "Control Channel Message Decode" errors as the original poster had). Since other posters had alluded to driver unlatching issues, I spent a while messing with usbreset and restarting and replugging everything. None of that helped, and it turns out that I did *not* have any driver unlatching issues. Instead, my HackRF had cooled down and the required frequency correction had shifted by a few hundred Hz.

In summary, HackRF users take note:
1. A stock HackRF does *not* have a temperature compensated oscillator, so its frequency reference will drift  with temperature.
2. trunk-recorder's control channel decoder more sensitive to frequency error than you might expect. A few hundred Hz difference is enough to cause decoding to fail.
3. If you receive "Control Channel Message Decode" errors, the first thing you should suspect is an error with your frequency configurations.

Cheers,
David

EricTendian, 2019-10-30:
^ I've had the temperature issue as well, especially now that seasons are changing. I've been running a fork with https://github.com/robotastic/trunk-recorder/pull/271 included which has generally solved the issue, since I no longer need to worry about making slight changes to account for drift. (also see this issue https://github.com/robotastic/trunk-recorder/issues/217)

riverar, 2020-02-18:
Am hitting this daily too. Going to install the tiny temperature controlled oscillator (tiny tcxo) add-on and see what happens.

leee, 2021-05-15:
@robotastic I think this issue can be closed. There are two conversations here, a smaller one on HackRFs and recycling USB ports (which probably isn't within scope), and a more generalized one on SDRs with drifting timers.

If there's interest, there's `kalibrate-rtl` and there was already work done on autotune bashing, and the process could be improved with peak detection or strongest bin tuning, but this means having to perform constant tuning/calibration on top of control channel search.

This is also just as easy of a problem to fix by hardware, or manual software compensation, which we already have.

riverar, 2021-05-15:
Agreed. And apologies for not retirning with results. The txco was a solid fix.

robotastic, 2021-05-16:
Agreed - Also adding that using a fan like this: https://www.amazon.com/AC-Infinity-MULTIFAN-Receiver-Playstation/dp/B00G05A2MU can be helpful no matter what SDR you use to help keep the temperatures stable.

closed 2021-05-16 by robotastic