TrunkRecorder/trunk-recorder #174: New to Trunk Recorder

viveksvgupta, 2018-01-31:
HI Everyone,
I am new to trunk recorder and I am having difficulty writing a config file and I am getting too many errors when I am trying to run it for my for my hackrf while trying to listen to P25 channels. Can you suggest me any materials that would help me understand the format of json files and on how to execute it for HackRF one without errors ?

Thank You

bctrainers, 2018-01-31:
Need a bit more information on this.

* What system are you attempting to record?
* What does the full startup sequence show?

viveksvgupta, 2018-01-31:
Thank you for your reply.
I am trying to listen to P25 channels, namely Police dispatch and other unencrypted police channels. I have a .csv file in my directory with all the talkgroups listed in it. 
I am unaware of the startup sequence. Can you please explain ?

bctrainers, 2018-02-01:
Funny enough, i posted an issue the other day and it turned out to being my own stupid getting the best of me (minor mis-config): https://github.com/robotastic/trunk-recorder/issues/173

In there, you can see the three core content items. hackrf_info, config.json, and the output from ./recorder that we're going to need.

viveksvgupta, 2018-02-02:
Thank you for your reply. But, I am now getting an error while executing the Config.json file. The error is Control Channel Message decode. I guess the error is mainly because I'm not selecting the center frequency and the control channels correctly. I also am unaware on how to set center frequency, the Error value and the control channel. I'm trying to listen to P25 channels in Tucson. If you can help me with this issue, it would be a great help, cause I couldn't find much material on this.  Thank You.

bctrainers, 2018-02-03:
Center frequency is the "down the middle" frequency for the hackRF. If you're monitoring 10MHz of bandwidth/spectrum between 850Mhz to 860Mhz, you're going to need to set the center to 855MHz. The config file does not up or down-convert MHz to Ghz or Hz, only in Hz. For a center frequency of 855.000MHz, you're going to set 855000000.

Control channel is the same. If the control channel is 853.775MHz, you're going to write in 853775000 in the Control frequency field. Do don't always need to add in the alternative CC's, but are of use if the main CC were to go offline for whatever reason. 

You may also need to set a drift correction value on either PPM or Error. Error is in Hz. My HackRF One has a general drift between 11000 Hz to 13000 Hz.

viveksvgupta, 2018-02-03:
Thank you so much for taking the time to reply. I will take all your suggestions and will give it a try. Will let you know soon if it worked or not. Again, Thank you so much. 
:)

bctrainers, 2018-02-03:
On the HackRF One, you may have 0/sec decodes or 0.6 or some realllly low value that does not jive well once you've ran ./recorder. You'll need to reboot your box for it to function once more, or use the usbreset function, as referenced by issue #173

viveksvgupta, 2018-02-06:
Hi,
I have included One primary, and three secondary control channels in my config file. I have also set the center frequency as you suggested me to, taking the center value over the frequency range I want to listen to. But still I'm getting the same error, Control Channel Message Decode Rate =1/sec. For me the command hackrf_info, doesn't return the serial number for my Hack RF one. So I'm using "hackrf" as the device name. Another thing I noticed is that the config file doesn't return me with an error when I try to run it without connecting my Hack RF One. I feel my Hack RF One is not being recognized in the system. Also, I've not set any error or PPM value, since I do not know how to find and set these values correctly. Kindly, help me with this problem. I have included my config file contents below. Thank You.
{
    "sources": [{
        "center": 852150000,
        "rate": 8000000,
        "error": 0,
        "ppm": 0,
        "gain": 48,
        "ifGain": 24,
        "bbGain": 28,
        "debugRecorders": 0,
        "digitalRecorders": 16,
        "digitalLevels": 8,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "hackrf"
    }],
    "systems": [{
        "control_channels": [853900000,853712500,853625000,853375000], 
        "sysId": "688",
        "sysSite": "401",
        "type": "p25",
        "shortName": "New",
        "recordUnknown": "true",
        "callLog": "true",
        "audioArchive": "true",
        "autoRetune": 1,
        "talkgroupsFile": "talkgroups.csv"
    }],
    "captureDir": "/home/radio/recordings",
    "callTimeout": 2,
    "logFile": 1
}

bctrainers, 2018-02-12:
>  But still I'm getting the same error, Control Channel Message Decode Rate =1/sec. 

You will need to re-plug the HAckRF into the machine, or use the usbreset application.

> For me the command hackrf_info, doesn't return the serial number for my Hack RF one. So I'm using "hackrf" as the device name. 

You may need to forcefully upgrade the firmware: https://github.com/mossmann/hackrf/wiki/Updating-Firmware then reboot the machine.

> Another thing I noticed is that the config file doesn't return me with an error when I try to run it without connecting my Hack RF One. I feel my Hack RF One is not being recognized in the system. 

This is probably the case due to the serial ID not being inputted. All you need on the serial is the last six alphanumeric values of the entire string. For me, mine is 56b35f. You're will be different than this.

>  I've not set any error or PPM value, since I do not know how to find and set these values correctly. 

As a general note, you need to set one or the other on the error/ppm setting. Never set both. Granted, trunk-recorder attempts to override and ignore it. 

The easiest way to calculate drifting of the SDR on the HackRF One is to use SDR# on windows or gqrx on a linux GUI. Set the frequency desired on SDR# to the primary control channel. You'll probably observe that the CC is a little "off" from the center of the set frequency. Essentially, you take the center frequency reported on SDR#. then subtract it from the proper CC frequency. It's easiest to do this in hertz. For example, 853.775MHz turns into 853775000Hz. If my HackRF One was reporting the CC on 853787000 when it should be on 853775000, that would translate to my HackRF One having +12000Hz of drift from the proper CC frequency.

closed 2020-07-31 by robotastic