TrunkRecorder/trunk-recorder #175: VHF SmartNet II System Not Getting Source IDs

dotsam, 2018-02-08:
First off, thanks for this software, it nice to be able to pair it with a $20 RTL-SDR and be capturing all talkgroups.

My local smartnet system is using a custom bandplan in the VHF space, and a few things don't seem to quite be working correctly. Mainly, the radio source IDs are rarely collected, and when they are, the same ID seems to get attached to future calls on a different talkgroup that shouldn't be from the same radio.

I see talk of some possible issues with VHF systems in the readme and in #136 (But that may only apply to P25) so I thought I'd open an issue and offer to test. I'm also wondering if this may be related to #160 (I'm going to try reverting that change and seeing what I get when I have a moment).

Here's my config:
```json
{
  "sources": [
    {
      "center": 142492500.0,
      "rate": 2048000.0,
      "error": -9000.0,
      "squelch": -50,
      "gain": 33,
      "digitalRecorders": 4,
      "modulation": "fsk4",
      "analogRecorders": 2,
      "driver": "osmosdr",
      "device": "rtl=0"
    }
  ],
  "systems": [
    {
      "control_channels": [
        142140000,
        142860000
      ],
      "type": "smartnet",
      "bandplan": "400_custom",
      "bandplanBase": 139635000.0,
      "bandplanHigh": 145320000.0,
      "bandplanSpacing": 15000.0,
      "bandplanOffset": 380,
      "talkgroupsFile": "crest.csv",
      "talkgroupDisplayFormat": "tag_id",
      "shortName": "crest",
      "uploadScript": "encode-upload.sh",
      "audioArchive": false
    }
  ],
  "frequencyFormat": "mhz",
  "callTimeout": 5,
  "logFile": true,
  "statusAsString": true
}
```

as well as some log messages about non-grants:

```
[2018-02-08 10:58:21.597958]: Non-Grant with source 0x2972 2972 on TG 0x41b0 16816
[2018-02-08 11:30:07.907800]: Non-Grant with source 0x2972 2972 on TG 0x42f0 17136
[2018-02-08 11:50:41.428452]: Non-Grant with source 0x3996 3996 on TG 0xadd0 44496
[2018-02-08 12:01:27.219367]: Non-Grant with source 0x8092 8092 on TG 0x47f8 18424
[2018-02-08 12:06:00.084740]: Non-Grant with source 0x3996 3996 on TG 0x41b0 16816
[2018-02-08 12:34:16.111681]: Non-Grant with source 0x3996 3996 on TG 0x41b0 16816
[2018-02-08 13:02:11.216090]: Non-Grant with source 0x9116 9116 on TG 0x41b0 16816
[2018-02-08 13:15:17.850411]: Non-Grant with source 0x5020 5020 on TG 0xa670 42608
[2018-02-08 14:18:33.575884]: Non-Grant with source 0x2972 2972 on TG 0x41b0 16816
[2018-02-08 14:18:43.682555]: Non-Grant with source 0x9116 9116 on TG 0x41b0 16816
[2018-02-08 14:21:05.763337]: Non-Grant with source 0x9116 9116 on TG 0x41b0 16816
[2018-02-08 14:22:26.845636]: Non-Grant with source 0x5020 5020 on TG 0x41b0 16816
[2018-02-08 14:25:38.255149]: Non-Grant with source 0x2972 2972 on TG 0x47f8 18424
[2018-02-08 14:38:45.536578]: Non-Grant with source 0x9116 9116 on TG 0x42f0 17136
```

Let me know if there's any other info that might help.

mrtrev, 2018-02-08:
I'm also running this on CREST and have noticed the same thing for probably a year or so now.  It used to work some of the time but now it rarely works.

dotsam, 2019-05-16:
Alright, after reading enough Smartnet OSW resources to make my head spin, I believe I have this working on my fork, in this commit https://github.com/dotsam/trunk-recorder/commit/4bda395457144caf7a3ae4f76173845ba73c30c4

They key is that on UHF/VHF Smartnet systems, the group call grant OSW has a command value of the input frequency of the requesting radio, rather than the 0x308 extended OSW code seen on 800Mhz systems. I only have the one Smartnet (SmartZone) system to test on, so I'm not 100% certain I haven't broken things anywhere else, but from all of my reading, everything looks sound.

In the process, I've stripped out anything not related to call updates/grants from the Smartnet parser, but I'm hoping to add back in unit affiliation and info on patched TGs at some point too.

If someone else wants to test this out and let me know if it works, I can break this off to another branch for a nice merge back in to master.

leee, 2020-07-31:
Oops, I didn't realize this issue existed when I wrote #383. @robotastic this is now a fixed issue (@dotsam gets first due credit!)

leee, 2020-08-04:
@robotastic: the primary issue is now covered with the Type II handler rewrite, and #388 covers emergency and encryption call handling and so all the features in the fork are now covered in master. This issue can be closed resolved.

robotastic, 2020-08-04:
Excellant!

closed 2020-08-04 by robotastic