TrunkRecorder/trunk-recorder #178: Keeping single recordings for retunes across devices

kkotowick, 2018-02-21:
I'm using 4 RTL-SDRs to capture a full P25 trunked system. Quite often, the system will re-tune a conversation from a frequency in one SDR to a frequency in a different one. When that happens, I get this message:

[2018-02-20 20:36:41.307941] (info)   [Trunked] TG: 4A5 Freq: 8.567875e+08      Update Retuning - New Freq: 8.511250e+08        Elapsed: 8s     Since update: 2s
[2018-02-20 20:36:41.308045] (info)      - Retune failed, New Freq out of range for Source: rtl=ems_delta,buflen=65536
[2018-02-20 20:36:41.308084] (info)      - Starting a new recording using a new source

I notice that it ends the original recording and starts a new one when this happens. My question is, is there any way to prevent it from creating a new recording and to instead simply keep appending the conversation to the ongoing recording?

johnrockie, 2018-05-19:
Were you able to get this fixed by any chance? Also having the same issue here.

robotastic, 2020-08-04:
Unfortunately, there isn't a great way around this. A `recorder` has to be assigned to an SDR. When the conversation moves to a new freq covered by a different SDR, it needs to stop the recording and start a new one.

closed 2020-08-04 by robotastic