TrunkRecorder/trunk-recorder #182: New to Trunk-Recorder, First-Time Questions

ayaresr, 2018-03-21:
Hi all,
As the title says, I'm new to Trunk-Recorder.  I'm having a bit of difficulty getting things going here.

I started out trying to get a couple of analog conventional frequencies to record.  I'm to the point where Trunk-Recorder will run, but I think I'm missing something.  It may be an issue with my squelch setting, but I'm not sure.  Depending on the value I use, I either get a bunch of 1kb files or a bunch of files which continually grow in size.  This seems to me to indicate the squelch either being open or closed, but I can't get any of these files to playback successfully, and I never see any json files added to the audio directory.

I've tested this with two devices.  The first being one of the RTL-SDR branded receivers with the TXCO, streamed across my network from a Raspberry Pi running rtl_tcp.  The second device is one of the cheaper generic RTL-SDR 820 sticks.

After tinkering on and off for days trying to get the conventional frequencies recording, I decided to try to instead get a P25 trunked digital system going.  I've successfully decoded this system many times using OP25 on the networked SDR I'm running.  That device doesn't require a correction, however when running OP25 I do have to configure a DC offset value or I get endless tuning errors.  Trunk-Recorder is giving me control channel decode errors when running on this P25 system.  I can't find any documentation for Trunk-Recorder relating to DC offset, but I took a guess and added dcOffset to the Source section of my config file to see if it would rectify the control channel errors.  I didn't receive any complaints from Trunk-Recorder regarding the config file, but I still can't overcome the control channel decode error.

Here's the config I'm using for the P25 system:

{
    "sources": [{
        "center": 774681250,
        "rate": 2048000,
        "modulation": "qpsk",
        "error": 0,
        "gain": 49,
        "dcOffset": 50000,
        "debugRecorders": 0,
        "digitalRecorders": 10,
        "analogRecorders": 0,
        "driver": "osmosdr",
        "device": "rtl_tcp:10.50.0.204"
    }],
    "systems": [{
        "type": "p25",
        "control_channels": [774681250],
        "shortName": "GLOUCO"
    }],
    "defaultMode": "digital",
    "frequencyFormat": "mhz",
    "captureDir": "/home/ayaresr/trunk-build/Audio"
}

Any insight would be appreciated.  I guess at this point since I've posted the config for the P25 system we can start there.  That said, if anyone has ideas on the conventional analog system I'm all ears.  I'll get the config file posted for that here shortly as well.

-Ryan

rosecitytransit, 2018-04-01:
I'm not sure if Trunk Recorder can use rtl_tcp, but if it's leaving up the radio interfacing to GNURadio/gr-osmosdr I could be wrong.

Try using the directly attached one and setting the `error` parameter.  If the 5000 doesn't work, run SDRSharp or GQRX or possibly `rtl_test -p`.  If using one of the first two, use the waterfall to manually line up the tuner to the control channel and note what frequency the program says it has tuned to.  The correct `error` setting will be the difference between that frequency and the actual one.  If using `rtl_test -p`, set the `ppm` parameter instead of `error`.

ayaresr, 2018-05-02:
Sorry for the really late reply.  Things got very busy around here and I had to put this project on the back burner for a while.  When I get the chance I'll give these suggestions a try and report back.  I do really appreciate the help.

As a side note, I was just going on the assumption that since I wasn't seeing any error directly related to my config file calling rtl_tcp that it must be supported.  I suppose there could possibly just not be any error check for that though.  Anyway, like I said, I'll report back when I get a little time to work at this again.

Thanks again,
Ryan

closed 2020-07-31 by robotastic