TrunkRecorder/trunk-recorder #184: Limit on number of sources?

kg6uyz, 2018-03-27:
Is there a limit on the number of sources one can define, or a number of recorders across a number of sources?  I have run into this issue multiple times trying to add a 5th source, whether it be trunked sources or analog sources, i have 4 sources currently, a VHF source for one of my p25 trunking systems, two VHF sources for two frequency ranges of analog channels and a low band source, i attempted to add a 5th source for a UHF frequency range but when i add the 5th source i get two errors "failed to submit transfer 3! and rtlsdr_read_async returned with  -5".  When this happens my p25 trunking system wont decode.  Any help would be greatly appreciated.

[image]

robotastic, 2018-03-27:
Check out the wiki. There is a string you have to add about buflen that should fix this.

kg6uyz, 2018-03-27:
That did the trick, iv read the wiki so many times i overlooked it.

closed 2018-03-28 by kg6uyz