TrunkRecorder/trunk-recorder #186: RPI recording capabilities

kg6uyz, 2018-03-29:
What is the recording capabilities of running trunk-recorder on a RPI? How many trunk systems and channels can it handle?

gallaway, 2018-07-06:
This is with 3 rtl usb dongles + raspberry pi 3. Looks like i have to use a laptop after all.
[image]

This is during a not so busy time on the radio

gallaway, 2018-07-07:
@kg6uyz i was running the standard configuration. But even with less recorder "maybe" one each I just dont think the RPi has the means for the task. 2 radio's with 2 recorders.... maybe but I would be worried about recordings being incomplete.

Dygear, 2018-07-12:
Have you tried running `volk_profile` on your pi and then trying it? You _might_ see some performance improvement as it will know the correct Vectored Kernal to use for your hardware.

closed 2020-07-31 by robotastic