TrunkRecorder/trunk-recorder #187: Analog VHF Low Band audio distorted

kg6uyz, 2018-03-29:
I am monitoring a couple analog VHF low band channels (42MHz) and have noticed the audio is distorted, some times the dispatcher is understandable but if they start speaking louder it distorts and the units in the field are always distorted, the low band channels are still 25khz wide band instead of 12.5khz narrowband like everything else now adays.  I am running a standard cheap RTL-SDR dongle with error correction based on GQRX that runs on my laptop.  I also noticed i have the same issue in GQRX with the audio, i can change the filter widths or the voice bandwidth and nothing seems to fix the distortion.  I rebooted to windows on that laptop and ran sdr-console, with filter changes i can get the audio to sound like normal radio traffic with no distortion.  I also noticed the PPM error correction is different between GQRX and SDR-Console, on GQRX i set it to 61 and on SDR-Console its -70.  Anybody else run into an issue like this?

robotastic, 2018-04-01:
Hmm... GQRX and Trunk Recorder are pretty similar on the backend. I would try to mess around with GQRX and see if there are setting that work well. I can try to make a recorder that works for low bands.

kg6uyz, 2018-04-01:
I played around with GQRX some more, the settings below work with no distortion;  

Input Controls: 
- LNA Gain: 40 dB

Reciever Options:
- Filter Width: Wide
- Filter Shape: Normal
- Mode: Narrow FM
 - - Max dev: 5KHz
 - - Tau: 530us
- AGC: off

robotastic, 2018-04-03:
Cool - I will go rummage through the GQRX code and see if I can make sense of it.

Dygear, 2018-07-12:
An LNA Gain of 40dBm seems super high to me. What antenna and SDR are you using?

I'm receiving audio in the VHF (150 - 150 MHz) range, and I don't have any distortion with a gain of 24dBm. As your getting analog channels, I'm just wondering if your clipping audio is due to you drowning the receiver's rf front end.

kg6uyz, 2018-07-12:
I receive 150 to 870mhz with no issues, i am running a 150 to 870mhz antenna, my low band (42mhz) dongle is a generic RTL2832u+R820t dongle.  It processes and gives me clean analog audio on sdr-console and gqrx with the same dongle and antenna setup.  Trunking-recorder gives me garbled audio.

jknepher, 2019-11-10:
Hi,  I noted this same problem and was able to resolve it.  Not sure of the right way to post back a patch, but have it ready to go if you are interested.
To fix this, one needs to make the deviation configurable, as the Low VHF still has a wide deviation.   In the process of making deviation variable, one then also has to make the audio resampler dynamic.  At that point, I also made the output sample rate variable.  After those happen, the ARB can sometimes be optimized out...

EricTendian, 2020-03-10:
@jknepher I would be interested in seeing how you did it, I'm running into a similar issue with one of my systems. Mind forking the repo and adding your patch?

kg6uyz, 2020-03-10:
I would like to try it out as well.

Jeff.

kg6uyz, 2020-05-20:
@jknepher Any chance of getting the fix you found for this submitted?

photounit, 2020-05-26:
I'm interested in this as well....

robotastic, 2020-07-31:
@jknepher Could you share the change you made? I am happy to include it in. Just feel free to email me the file and I will figure out what changed.

jknepher, 2020-08-03:
Diff attached here.  It has been working very well for me.  
[DevDiff.txt](https://github.com/robotastic/trunk-recorder/files/5018398/DevDiff.txt)

It adds a max_deviation parameter that can be increased for the wideband channels.  Default is 4000, high deviation channels should set this to 8000.

I also increase the wave sampling rate, which improves readability significantly.  I only found reliable copy when both changes were made.

There is also a case where the ARB can be optimized away, so there is a change for that as well.

For standard width channels, I took wave sample from 8000 to 16000, changed max_deviation from 4000 to 8000.   Note that changing the deviation on narrow width channels will make them significantly worse, which is why it needs to be configurable.

leee, 2020-08-03:
I haven't really looked at the diff closely, but thank you for contributing!

Some thoughts I wish to contribute:

Deviation could better be described by what the LMR industry uses, e.g.
- instead of a `max_dev` of `4000` (or in this case, `5000`), describe the channel as `2.5 kHz deviation (12.5 kHz channel bandwidth)` (`11K2` occupied bandwidth). Same for:
- instead of `8000`, use `4.0 kHz deviation / 20 kHz channel bandwidth` (`16K0` occupied bandwidth), and
- instead of `10000`, use `5.0 kHz deviation / 25 kHz channel bandwidth` (`20K0` occupied bandwidth).

800 SmartNet analog recorders should automatically be `16K0F3E`.

While the "easy" fix for now is to make analog channel deviation/bandwidth a source setting (`analogMaxDeviation`), it's definitely possible to have both narrowband and wideband channels in the same band that a source could cover. So this may be another reason to consider implementing unique-recorder settings.

jknepher, 2020-08-03:
Totally agree with you Leee.  My diff was to fix with minimal change, and I just run another instance with another receiver.  If enhancing to support multiple channel deviation in the same source, I suggest also moving analogLevels with it -- since they are closely related, and have to be tuned together.  Similarly, it would be nice also to be able to configure the filter width independently from the demodulation deviation if one were to put that level of effort into such an endeavor -- my diff changes the width percent from 0.8 to 1.0, which helps the wide channels while it doesn't seem to hurt the narrow channels for me, but I expect that might not be the right solution generally.

robotastic, 2020-09-19:
@jknepher I am moving recorder configurations to be at the system level instead of the source. It seems like a good time to make the analog recorder more configurable. What are the settings you would want to have in the config_file? Is it just the max_dev setting? We could also just have it be a Wide/Narrow setting, which might be more approachable. Would people want to set some different values besides the those 2 defaults?

I am not as familiar with tuning analog and I don't have any local systems to practice on.

Should the sample_rate for wav files be adjustable in analog recorders? It seems like a good addition or I could just fix it at a higher sample rate, unless people would want to adjust this.

jknepher, 2020-09-21:
Hi @robotastic!  Thank you for this project.

Max_dev ties closely with the analog gain setting as well, so they should move together.  While the quad_gain computed and then passed to quadrature_demod should normalize those, I have found that it ends up causing distortion if I tried to make them about the same.

If you decide to use a wide/narrow, there should actually be three settings as @leee pointed out for 11k2, 16k0, and 20k0.  Personally, I prefer just passing the value.

For higher sample rate, I have found it helps readability for everything; it also makes later compression not harm the readability as much.

robotastic, 2020-09-22:
Thanks @jknepher  - I will create system level settings for Analog Levels, Max Dev, and Filter Width. I will leave the as standalone values so people can tune as needed.

robotastic, 2020-09-24:
@jknepher I made these changes in this commit: https://github.com/robotastic/trunk-recorder/commit/e85195b564e55422a331e7df1ebd99e18609ad6e

I am a little confused on how to translate some of the LMR terms into what Gnuradio is looking. If I am understanding right, Max Deviation refers to the HZ a signal would be from the center of the channel. For the gnuradio settings, I think this is the same as cutoff_freq: https://www.gnuradio.org/doc/doxygen/classgr_1_1filter_1_1firdes.html#a57c2a9eb8dc61d3740d9eb0e02ebb880

There is also a setting for transition_width. I think this lets you configure how tight the channel windowing is. Is this a setting you would want to use too? 

I added in a filter width setting, but I am not sure if it is something that is needed. It seems like max_dev covers that.

robotastic, 2020-09-24:
PS - I didn't change the initial configuration. The analog recorder gets reconfigured each time a new recording is started: https://github.com/robotastic/trunk-recorder/commit/e85195b564e55422a331e7df1ebd99e18609ad6e#diff-dc0b5ca752890c9fa641cc29f5afae8cR309

jknepher, 2020-09-28:
@robotastic 

Yes, max deviation is how far the carrier moves from the set frequency in either direction, but you need to also account for the sidebands.  The width of the IF passband is important for FM, as the passband needs to include not only the full deviation, but also the higher energy sidebands.  Yes, it should be directly related to the deviation, but wider to include the first couple sidebands.  This is generally calculated as 2x(deviation + maximum modulated freq).  For the modulations leee pointed out earlier, consider:

25kHz channel width, 20kHz bandwidth occupied, 5kHz deviation, and 5kHz voice audio bandwidth.  2x(5 + 5)=20
20kHz channel width, 16kHz bandwidth occupied, 4kHz deviation, and 4kHz voice audio bandwidth.  2x(4 + 4)=16, etc.

On my hardware radios, the FM IF filters are stated as 15, 10, and 7 kHz, but the slopes (transition band) are wide enough to allow significant energy through at 20, 16, 11.2kHz to let the sidebands through.  The transition bands here should be similar -- wide enough to let the sidebands through, but sharp enough to not let much of the next channel in.

The way things are computed now, the max_dev is driving two parameters -- the filter width and the demodulation gain.  As the deviation is set to be wider, that effectively reduces the gain at the demodulator and widens the filter.  By having the later audio gain option, one can basically change the filtering indirectly, although it isn't as good as having all of the controls up front.

When listening, if the signal gets noisy/static at the highest volume parts of the signal, that is the indicator that the passband needs to be wider.  If the signal is getting clipped/distorted at the highest volume parts, that is in indication that the deviation is set too low (although it may also turn into noise/static/silence depending on the demodulator implementation).  If the passband is way too wide, the risk is capture by an adjacent carrier.

In case you are curious, AM and SSB IF filtering has a completely different effect on the output audio -- the transition bands there change the frequency response of the received audio.  Wider passbands and gentler/wider transitions allow more high frequency voice response through, at the expense of higher overall noise and potentially lots of hiss.  Lots of configuration in this case is desirable; and you can directly hear the changes as you modify the filters, just like adjusting a stereo's EQ.

closed 2024-12-15 by robotastic