TrunkRecorder/trunk-recorder #192: New to trunk recorder, need help to get started

johnrockie, 2018-05-17:
Hello guys,
This may sound like the most basic things but I have not been able to figure it out for the last 6 hours. I am trying to setup up this [system](https://www.radioreference.com/apps/db/?sid=7722). This is how my config.json looks:

> {
    "sources": [{
        "center": 853500000,
        "rate": 2048000,
        "error": 1100,
        "gain": 39,
        "debugRecorders": 0,
        "digitalRecorders": 4,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=2"
    }, {
        "center": 851000000,
        "rate": 2048000,
        "error": 1400,
        "gain": 39,
        "digitalRecorders": 4,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=00000001"
    }, {
        "center": 854000000,
        "rate": 2048000,
        "error": 1400,
        "gain": 39,
        "digitalRecorders": 4,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=5"
}],
    "systems": [{
        "control_channels": [853600000],
        "type": "p25",
        "talkgroupsFile": "pwc.csv",
        "shortName": "pwcva",
	"apiKey": "xxxxxxxxxxxxxxxxxxxxxx"
    }],
  "uploadServer": "https://api.openmhz.com"
}
This is the starting log with the error I get
[terminal.log](https://github.com/robotastic/trunk-recorder/files/2011425/terminal.log)

Any help would be appreciated. Thanks!

bduhan, 2018-05-17:
Add the alternate control channel frequencies to `control_channels`.

johnrockie, 2018-05-17:
@bduhan Thanks for your reply. I am getting something different now:

> [2018-05-17 06:46:42.005495] (error)   [pwcva] Retuning to Control Channel: 8.538750e+08
[2018-05-17 06:46:42.005572] (info)   	 - System Source - Min Freq: 8.524760e+08 Max Freq: 8.545240e+08
[2018-05-17 06:46:42.005837] (error)   [pwcva]	 Control Channel Message Decode Rate: 1/sec, count:  3
[2018-05-17 06:46:45.011782] (error)   [pwcva] Retuning to Control Channel: 8.536000e+08
[2018-05-17 06:46:45.011948] (info)   	 - System Source - Min Freq: 8.524760e+08 Max Freq: 8.545240e+08

johnrockie, 2018-05-23:
Fixed. Needed to retune sources.

closed 2018-05-23 by johnrockie