TrunkRecorder/trunk-recorder #193: Raspberry Pi Stretch + 2 dongles = eigen_unaligned_array_assert_workaround_gcc47

AppliedEllipsis, 2018-06-19:
I just built everything for a Raspberry Pi Stretch install on a 3 b+.
When using 2 rtl-sdr's I occasionally get this:

```
[2018-06-19 00:23:38.222403] (info)   [namehere]   Decoding System ID REMOVED WACN: REMOVED NAC: REMOVED 
recorder: /home/pi/Downloads/trunk-recorder/lib/lfsr/Eigen/src/Core/DenseStorage.h:109: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = int; int Size = 1936; int MatrixOrArrayOptions = 0]: Assertion `(internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (15)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
Aborted
```

Any advice?

Also, you should update the Raspberry Pi install instructions to use a separate build directory like the generic build instructions. (I've tried both btw)

I have a feeling this might have to do with some lib conflicts?

```
[100%] Linking CXX executable recorder
/usr/bin/ld: warning: libboost_date_time.so.1.61.0, needed by /usr/lib/gcc/arm-linux-gnueabihf/6/../../../arm-linux-gnueabihf/libuhd.so, may conflict with libboost_date_time.
so.1.62.0
/usr/bin/ld: warning: libboost_date_time.so.1.61.0, needed by /usr/lib/gcc/arm-linux-gnueabihf/6/../../../arm-linux-gnueabihf/libuhd.so, may conflict with libboost_date_time.
so.1.62.0
/usr/bin/ld: warning: libboost_filesystem.so.1.61.0, needed by /usr/lib/gcc/arm-linux-gnueabihf/6/../../../arm-linux-gnueabihf/libuhd.so, may conflict with libboost_filesyste
m.so.1.62.0
/usr/bin/ld: warning: libboost_filesystem.so.1.61.0, needed by /usr/lib/gcc/arm-linux-gnueabihf/6/../../../arm-linux-gnueabihf/libuhd.so, may conflict with libboost_filesyste
m.so.1.62.0
/usr/bin/ld: warning: libboost_program_options.so.1.61.0, needed by /usr/lib/gcc/arm-linux-gnueabihf/6/../../../arm-linux-gnueabihf/libuhd.so, may conflict with libboost_prog
ram_options.so.1.62.0
/usr/bin/ld: warning: libboost_program_options.so.1.61.0, needed by /usr/lib/gcc/arm-linux-gnueabihf/6/../../../arm-linux-gnueabihf/libuhd.so, may conflict with libboost_prog
ram_options.so.1.62.0
/usr/bin/ld: warning: libboost_regex.so.1.61.0, needed by /usr/lib/gcc/arm-linux-gnueabihf/6/../../../arm-linux-gnueabihf/libuhd.so, may conflict with libboost_regex.so.1.62.
0
/usr/bin/ld: warning: libboost_regex.so.1.61.0, needed by /usr/lib/gcc/arm-linux-gnueabihf/6/../../../arm-linux-gnueabihf/libuhd.so, may conflict with libboost_regex.so.1.62.
0
/usr/bin/ld: warning: libboost_system.so.1.61.0, needed by /usr/lib/gcc/arm-linux-gnueabihf/6/../../../arm-linux-gnueabihf/libuhd.so, may conflict with libboost_system.so.1.6
2.0
/usr/bin/ld: warning: libboost_system.so.1.61.0, needed by /usr/lib/gcc/arm-linux-gnueabihf/6/../../../arm-linux-gnueabihf/libuhd.so, may conflict with libboost_system.so.1.6
2.0
/usr/bin/ld: warning: libboost_thread.so.1.61.0, needed by /usr/lib/gcc/arm-linux-gnueabihf/6/../../../arm-linux-gnueabihf/libuhd.so, may conflict with libboost_thread.so.1.6
2.0
/usr/bin/ld: warning: libboost_thread.so.1.61.0, needed by /usr/lib/gcc/arm-linux-gnueabihf/6/../../../arm-linux-gnueabihf/libuhd.so, may conflict with libboost_thread.so.1.6
2.0
/usr/bin/ld: warning: libboost_chrono.so.1.61.0, needed by /usr/lib/gcc/arm-linux-gnueabihf/6/../../../arm-linux-gnueabihf/libuhd.so, may conflict with libboost_chrono.so.1.6
2.0
/usr/bin/ld: warning: libboost_chrono.so.1.61.0, needed by /usr/lib/gcc/arm-linux-gnueabihf/6/../../../arm-linux-gnueabihf/libuhd.so, may conflict with libboost_chrono.so.1.6
2.0
/usr/bin/ld: warning: libboost_atomic.so.1.61.0, needed by /usr/lib/gcc/arm-linux-gnueabihf/6/../../../arm-linux-gnueabihf/libuhd.so, may conflict with libboost_atomic.so.1.6
2.0
/usr/bin/ld: warning: libboost_atomic.so.1.61.0, needed by /usr/lib/gcc/arm-linux-gnueabihf/6/../../../arm-linux-gnueabihf/libuhd.so, may conflict with libboost_atomic.so.1.6
2.0
/usr/bin/ld: warning: libboost_system.so.1.58.0, needed by //usr/lib/arm-linux-gnueabihf/libgnuradio-iqbalance.so.0, may conflict with libboost_system.so.1.62.0
/usr/bin/ld: warning: libboost_system.so.1.58.0, needed by //usr/lib/arm-linux-gnueabihf/libgnuradio-iqbalance.so.0, may conflict with libboost_system.so.1.62.0
/usr/bin/ld: warning: libboost_filesystem.so.1.58.0, needed by //usr/lib/arm-linux-gnueabihf/libgnuradio-fcdproplus.so.3.7.10, may conflict with libboost_filesystem.so.1.62.0
/usr/bin/ld: warning: libboost_filesystem.so.1.58.0, needed by //usr/lib/arm-linux-gnueabihf/libgnuradio-fcdproplus.so.3.7.10, may conflict with libboost_filesystem.so.1.62.0
[100%] Built target recorder
```

gallaway, 2018-07-03:
I am having the same issue I think. This is where mine crashes.

[2018-07-03 03:41:05.868537] (info)   [dcfems]	TG: 11037	Freq: 0.000000e+00	Not Recording: no source covering Freq
[2018-07-03 03:41:06.172791] (info)   [dcfems]	Decoding System ID 445 WACN: BEE00 NAC: 443
recorder: /home/linaro/trunk-recorder/lib/lfsr/Eigen/src/Core/DenseStorage.h:109: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = int; int Size = 1936; int MatrixOrArrayOptions = 0]: Assertion `(internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (15)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.

Thread 1 "recorder" received signal SIGABRT, Aborted.
0xb5d2a6f6 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
(gdb) bt
#0  0xb5d2a6f6 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
#1  0xb5d383cc in raise () from /lib/arm-linux-gnueabihf/libc.so.6
#2  0xb5d390ba in abort () from /lib/arm-linux-gnueabihf/libc.so.6
#3  0xb5d33636 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)

Dygear, 2018-10-13:
I've never seen this error, but I don't actually use the Pi's for digital P25 decoding at all. How would a separate build directory like the generic build instructions help? You said you tried both and they didn't work out for you.

heathfx, 2018-12-14:
I also have this exact same problem on my raspi, sometimes it takes over 10 tries to get trunk-recorder to run.

jimbo1969, 2019-03-02:
I've had the same error (eigen_unaligned_array) pop up randomly, in the first 30 seconds of each attempt on my rpi with multiple RTL dongles.  Also, more often than not when it starts, it fails to decode the control channel after "rtlsdr_read_async returned with -5".  After I then stop the recorder process with [Ctrl+z], I try "rtl_eeprom -d 0" to see if the rtlsdr is live and accessible.  It detects all of the rtlsdr's, but I get a "usb_claim_interface error -6 [...] Failed to open rtlsdr device #0."

I don't know if that is related or not, but it is definitely inconsistent also.

closed 2021-04-28 by robotastic