TrunkRecorder/trunk-recorder #194: whitelist?

JustApu, 2018-07-07:
I've been using op25 to monitor a P25 system for a while but just found and started using trunk-recorder. Am I missing something or is the only way to record only specific talkgroups, instead of every talkgroup in the system, to use a combination of 

``        "talkgroupsFile": "talkgroups.csv",`` listing only a few talkgroups of interest and 
``        "recordUnknown": "false",``

in the config file? There's no way to keep the full list of all the talkgroups but only whitelist a few of them for recording?

bctrainers, 2018-07-07:
Currently, that would be correct.

I'd like to have a per-system RecordOnly comma separated value listing to do as you describe however. 👍

raygalasso, 2018-07-07:
In the talk group csv, set the priority to 99 for the talk groups you don’t want.

JustApu, 2018-07-07:
Unless I did that wrong, priority 99 just makes it a low priority channel. If there is an available recorder, the channel will still get recorded and take up disk space.

vabiro, 2018-07-07:
You can also set the RecordUnknown flag in the system, which will only record the TGIDs in the csv

"recordUnknown - record talkgroups if they are not listed in the Talkgroups File. The options are true and false (without quotes). The default is true."

JustApu, 2018-07-07:
@vabiro - that's what I'm using already (see initial post). I was inquiring if I missed a another way.

There are known talkgroups I do want to record (a few) and known talkgroups I don't want to record (most talkgroups on this system). If I could configure **trunk-recorder** to only record the groups I am interested in, but still know about the others, then the logs and ``recordUnknown`` parameter becomes valuable.

It could be useful to see info about and record truly unknown talkgroups while ignoring talkgroups you know and have no interest in.

vabiro, 2018-07-07:
@JustApu I did see the original post, but I was trying to point out that you had the setting false in quotation marks. That may be causing it to not work as expected.

If that isn't it, I had a similar problem when I first compiled trunk-recorder. For some reason, it was ignoring some of the settings in the configuration, including recordUnknown. It was behaving exactly like you describe. I did a make clean and recompiled, and it worked fine after that.

JustApu, 2018-07-07:
It works with the quotes.

The "problem" is simply that these are not unknown talkgroups. They are ignored talkgroups and I was looking to have a whitelist of talkgroups that should be recorded (or a blacklist of those to be ignored).

Rylvir, 2018-07-07:
So you want to define only the talkgroups you want to monitor in the 
ChanList.csv or talkgroups file. Then set the do not record unknown 
talkgroups, that will effectively whitelist the talkgroups identified in 
your csv file.

raygalasso, 2018-07-08:
If you set it to 99 you’d need 99 free recorders for it to record.  Just set the number of recorders less than 99 and it will never record.

vabiro, 2018-07-08:
@JustApu You can also only list the TGIDs that you want to record, in effect a whitelist. 

I have mine set-up exactly as you seem to want: recordUnknown set to false, and only the IDs I want to record in the csv.

closed 2020-07-31 by robotastic