TrunkRecorder/trunk-recorder #197: Enhancement: Implement a Mechanism to End Recording If A User-Defined Length is Exceeded

RachelOlivero, 2018-07-26:
### The Problem
Certain trunked systems are configured so that when a talk group is in emergency mode the assigned repeater will be held active and a new channel grant is not issued when a unit keys up. That is, all activity continues on one frequency until the emergency status is released. This can make for lengthy recordings, and make near real-time monitoring impossible.

### Proposed Solution
Allow the user to specify a maximum recording time in seconds. Once exceeded, begin looking for a good place to stop recording and start another file. A "logical break" could be inserted when a Termination Data Unit (TDU) is detected on the voice channel. If I understand correctly, a TDU follows the comletion of every voice message (See page 53 of http://www.dvsinc.com/papers/p25_training_guide.pdf)

### Assumptions
Trunk Recorder is already monitoring the low speed data signal on the voice channel. This is assumed because time indicators of units beginning transmissions are provided in the recording's json file.

### Anticipated changes
1. Recording code would need to be updated to check for recording length and whether it specifies the user-defined parameter; code added to terminate the recording and start a new file in the same way as happens when the call timeout is exceeded.
2. A parameter added to the system record in config.json, maximumRecordingTime, to allow a user to set a desired maximum length of any transmission file.

RachelOlivero, 2019-02-01:
@dkettmann I have started working on a solution to this. Hopefully I'll have it done shortly.

jareklupinski, 2019-03-14:
https://github.com/jareklupinski/trunk-recorder/tree/max_duration i threw this together quickly just to solve the problem for myself, not very flexible as its a global timeout but works great for me :)

dkettmann, 2019-03-18:
Caveat: I know just enough to be dangerous, as will probably be apparent in the below.

Yesterday morning I started looking to use/test this new code.  Not knowing git in general, nor how to overwrite my existing Robotastic files and re-compile from your tree(?), I made a blank/scratch directory, and did a git clone like so: https://github.com/jareklupinski/trunk-recorder/

This pulled down your fork with the minDur but didn't have the latest maxDur.  I then modified the couple of lines in config.cc, config.h, and main.cc to match the diff on your maxDuration tree.  I complied this, replaced the stock "recorder" binary with this new one, added maxDuration to my config.json, and started it up.

I am monitoring a P25 Phase 1 system with digital voice.  Later on in the day there was a major event and it behaved like nothing had changed.  Although I set the value to 240, it recorded one channel for 15 minutes before finishing it.  
Now that I read this a little more, I see this was only for conventional channels, my bad.  I guess that explains why it had no effect on my system?  Thanks for the effort.  I did  get a bit of a joy from seeing the modifications to the part of the code that seemed to me a good candidate a few months ago.  Look forward to if you ever push a fix for my situation (if I'm reading this right).

-Dan

jareklupinski, 2019-03-25:
Yup you are correct, the code as you downloaded only did max-length clipping for analog systems.

I just added another check to clip digital (P25 etc) systems, let me know if it works :)

One way to pull in and test the code would be:
```
[Open Terminal]
git clone https://github.com/jareklupinski/trunk-recorder/
cd trunk-recorder
git checkout max_duration
git pull
mkdir build
cd build
cmake ../
make
./recorder
```

dkettmann, 2019-03-25:
Will implement tonight and report back.  Thanks!

dkettmann, 2019-04-02:
I don't think it is working, I just checked it and saw a summary like this in the buffer when I had maxDuration set to 245.

<snip>
[2019-04-01 22:00:49.011867] (info)   Currently Active Calls: 1
[2019-04-01 22:00:49.011978] (info)   TG: 9105 Freq: 8.524625e+08 Elapsed: 477 State: 1
[2019-04-01 22:00:49.012082] (info)     [ 2 ] State: 3
[2019-04-01 22:00:49.012126] (info)   Recorders:
[2019-04-01 22:00:49.012163] (info)   [ rtl=0 ]
[2019-04-01 22:00:49.012199] (info)   [ 0 ] State: 2
[2019-04-01 22:00:49.012238] (info)   [ 1 ] State: 2
[2019-04-01 22:00:49.012279] (info)   [ 2 ] State: 3
[2019-04-01 22:00:49.012319] (info)   [ 3 ] State: 2
<snip>
(restarted to verify I was running right binary)
-------------------------------------
INSTANCE
------------------------------------
[2019-04-01 22:17:05.394410] (info)   Capture Directory: /home/dan/trunk-player/audio_files
[2019-04-01 22:17:05.394430] (info)   Upload Server:
[2019-04-01 22:17:05.394446] (info)   Status Server:
[2019-04-01 22:17:05.394469] (info)   Instance Key:
[2019-04-01 22:17:05.394485] (info)   Instance Id:
[2019-04-01 22:17:05.394501] (info)   Default Mode: digital
[2019-04-01 22:17:05.394526] (info)   Call Timeout (seconds): 3
[2019-04-01 22:17:05.394552] (info)   Log to File: false
[2019-04-01 22:17:05.394620] (info)   Control channel warning rate: 10
[2019-04-01 22:17:05.394642] (info)   Control channel retune limit: 0
[2019-04-01 22:17:05.394678] (info)   Maximum Call Duration (seconds): 245

Sorry for the delay in responding, time has been a little tight and the first attempts I had added maxDuration to config.json under System. Then I finally saw the start-up output enough times I realized it was under the Instance section (different than minDuration in your readme which I consulted).  Once I put the setting of 245 in that part of config I saw it reflected in startup.  I haven't spent a ton of time monitoring my system but it seems that its not clipping.  Happy to test or help if I can.

dkettmann, 2019-04-02:
....actually I just saw an entry come out that took up about 10 minutes of real-time, but the resulting silence-removed time was 4 minutes and 5 seconds.  Can't be coincidence can it?  Should I adjust downward on the setting, with the thinking that it is calculating how much non-silence time and not real-time? I'll continue to see how it behaves.

jareklupinski, 2019-04-02:
Interesting! I haven't had any luck with my local stations transmitting anything longer than a minute or so; I have my maxDuration set to 60 seconds and haven't seen anything like that but I'll keep an eye out.

dkettmann, 2019-04-02:
[image]
Ended up with this, seems pretty certain it was doing its calculation based on compressed time.  I changed the setting to 120 seconds to verify and make the clipping more agressive.

PR #480 (2021-05-21): Add maximum call duration to split long calls
Resolves #197, resolves #357.

---

I'd like to reopen for consideration #357, which @EricTendian submitted on behalf of @jareklupinski. #357 directly resolves #197, although more work can be done surrounding the topic.

There are many cases where a conventional or trunked resource stays in use for extremely long periods of times, making the receipt of that one transmission delayed for the duration of that period.

#197 is an example of dispatch consoles being able to mark a trunked resource in emergency, reserving and holding voice talkpaths for i [... cut]
commit 75b8796a (2021-05-21): Add maximum call duration to split long calls

Resolves #197, resolves #357.
closed 2021-05-21 by robotastic