TrunkRecorder/trunk-recorder #202: itemsize mismatch when using system type conventionalP25

mdozier9, 2018-10-09:
When setting up a "conventionalP25" system, I keep getting the error:

(info)   P25 Recorder Taps - initial: 363 channel: 1455 ARB: 727 Total: 2545
terminate called after throwing an instance of 'std::invalid_argument'
  what():  itemsize mismatch: multiply_const_ff64:0 using 4, nonstop_wavfile_sink26:0 using 2
Aborted

The same trunk-recorder instance works great for conventional and P25, as well as mixed systems, but as soon as I add the conventionalP25 system it will abort.  Here's my json config:
{
    "sources": [{
        "center": 155000000,
        "rate": 2400000,
        "error": 0,
        "gain": 28,
        "digitalRecorders": 2,
        "analogRecorders": 2,
        "squelch": -50,
        "driver": "osmosdr",
        "device": "rtl=00000018,buflen=65536"
     }],
    "systems": [{
        "channels": [155617500],
        "type": "conventionalP25",
        "controlWarnRate": 3,
        "shortName": "test",
        "audioArchive": "false",
        "callLog": "false"
    }],
        "captureDir": "/home/mdozier/trunk-recorder/media/",
        "audioArchive": "false"
}

Does anyone have any idea how to get past this error?  Can someone with a working conventionalP25 setup please post your json file?  Tnx, Mike

Cole0209, 2018-10-21:
I am also having this issue however it is only when I try to run my systems on my Raspberry Pi. On my x64 based PC it runs flawlessly.

mdozier9, 2018-10-21:
I'm experiencing this on an Intel Core 2 Duo 3.3Ghz running Ubuntu 18.04.

pcunning, 2018-10-22:
I'm having this issue on a Intel Atom and a fresh build of trunk-recorder. Older build worked great with the same config.

```[2018-10-22 10:45:56.510148] (info)   	 P25 Recorder Initial Rate: 600000 Resampled Rate: 30000 Initial Decimation: 3 Decimation: 20 System Rate: 30000 ARB Rate: 1
[2018-10-22 10:45:56.510495] (info)   P25 Recorder Taps - initial: 273 channel: 1819 ARB: 727 Total: 2819
terminate called after throwing an instance of 'std::invalid_argument'
  what():  itemsize mismatch: multiply_const_ff17:0 using 4, nonstop_wavfile_sink5:0 using 2
Aborted```

Cole0209, 2018-10-22:
Ok so I went back to my PC to fire everything back up... It seems as though when I updated something in Ubuntu now I get this error no matter what I run trunk-recorder on. Like @pcunning said older builds worked this fresh install is having serious issues with something. I am thinking one of the dependencies got updated and something is not right...

pcunning, 2018-10-22:
Mine broke doing a dist-upgrade. Haven’t had the time to dig in more.

robotastic, 2018-10-23:
I probably messed something up.... the mis-match error comes up from GnuRadio. It means that the size of the value being passed in doesn't match what was expected.

robotastic, 2018-10-23:
Ok - give it a try now, it should be work fine

mdozier9, 2018-10-23:
I just pulled master and recompiled...no luck...getting same error:
[2018-10-23 10:19:39.038539] (info)   P25 Recorder Taps - initial: 311 channel: 1551 ARB: 731 Total: 2593
terminate called after throwing an instance of 'std::invalid_argument'
  what():  itemsize mismatch: multiply_const_ff11:0 using 4, nonstop_wavfile_sink2:0 using 2

Is there a dev branch I should pull?

Cole0209, 2018-10-23:
Looks like it was commited to master @mdozier9  https://github.com/robotastic/trunk-recorder/pull/204 I am recompiling it now as well.

Cole0209, 2018-10-23:
Yup still the same error @robotastic the issue seems to be from a package upgrade. My issue started after doing a dist-upgrade on Ubuntu 16.04. I upgraded to 18.04 (fresh install) and am still getting the issue however without conventionalP25 the issue is not present

robotastic, 2018-10-24:
I was dumb - I had made the commit but hadn't pushed it. Should be all good now... sorry for the confusion.

Cole0209, 2018-10-24:
Yup I just recompiled from master everything is back to normal 👍

mdozier9, 2018-10-24:
Working well for me with mixed systems on one server.  Thanks for the help!

mdozier9, 2019-03-15:
This issue can now be closed, as looks like its been successfully corrected and verified.  Thanks again!

closed 2019-03-15 by mdozier9