TrunkRecorder/trunk-recorder #203: Continuous Control Channel Loop

RossIV, 2018-10-21:
I seem to be having a similar issue to many others based on looking at past issue reports. I'm attempting to monitor the [Atlanta Public Safety P25](https://www.radioreference.com/apps/db/?sid=5834) system using RTL SDRs. Each time I start the [trunk recorder Docker image](https://github.com/robotastic/trunk-recorder-docker), I get continuous errors of decode rate being too low and changing to a different control channel. If I leave the image running for several minutes, once every now and again I may see a "Current Call Status" output message (included below) but it's rare.

I'm able to monitor the same system fine in OP25 using `./rx.py --args 'rtl=00000011' -N 'LNA:30' -S 2400000 -f 857.8875e6 -o 25000 -V -U`, but not with the `config.json` as included below. 

Any suggestions greatly appreciated. 
Thanks!
--Ross

**Continuous errors**
```
recorder_1  | [2018-10-21 03:30:02.879996] (info)   [ATL]	Started with Control Channel: 857.887512
recorder_1  | [2018-10-21 03:30:02.893223] (info)   	 P25 Trunking - SysNum: 0
recorder_1  | [2018-10-21 03:30:02.893345] (info)   	 P25 Recorder Initial Rate: 500000 Resampled Rate: 50000 Initial Decimation: 16 Decimation: 10 System Rate: 48000 ARB Rate: 0.96
recorder_1  | [2018-10-21 03:30:02.893799] (info)   P25 Recorder Taps - initial: 1213 channel: 1137 ARB: 757 Total: 3107
recorder_1  | [2018-10-21 03:30:05.021990] (error)   [ATL] Retuning to Control Channel: 857.787537
recorder_1  | [2018-10-21 03:30:05.022184] (info)   	 - System Source - Min Freq: 851.000000 Max Freq: 859.000000
recorder_1  | [2018-10-21 03:30:05.023561] (error)   [ATL]	 Control Channel Message Decode Rate: 0.333333/sec, count:  1
recorder_1  | [2018-10-21 03:30:08.019914] (error)   [ATL] Retuning to Control Channel: 857.487488
recorder_1  | [2018-10-21 03:30:08.020441] (info)   	 - System Source - Min Freq: 851.000000 Max Freq: 859.000000
recorder_1  | [2018-10-21 03:30:08.024748] (error)   [ATL]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2
recorder_1  | [2018-10-21 03:30:11.017265] (error)   [ATL] Retuning to Control Channel: 857.462524
recorder_1  | [2018-10-21 03:30:11.017788] (info)   	 - System Source - Min Freq: 851.000000 Max Freq: 859.000000
recorder_1  | [2018-10-21 03:30:11.020590] (error)   [ATL]	 Control Channel Message Decode Rate: 1/sec, count:  3
recorder_1  | [2018-10-21 03:30:14.015993] (error)   [ATL] Retuning to Control Channel: 857.887512
recorder_1  | [2018-10-21 03:30:14.016512] (info)   	 - System Source - Min Freq: 851.000000 Max Freq: 859.000000
recorder_1  | [2018-10-21 03:30:14.020879] (error)   [ATL]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2
recorder_1  | [2018-10-21 03:30:17.014651] (error)   [ATL] Retuning to Control Channel: 857.787537
recorder_1  | [2018-10-21 03:30:17.014948] (info)   	 - System Source - Min Freq: 851.000000 Max Freq: 859.000000
recorder_1  | [2018-10-21 03:30:17.016751] (error)   [ATL]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2
recorder_1  | [2018-10-21 03:30:20.018095] (error)   [ATL] Retuning to Control Channel: 857.487488
recorder_1  | [2018-10-21 03:30:20.018616] (info)   	 - System Source - Min Freq: 851.000000 Max Freq: 859.000000
recorder_1  | [2018-10-21 03:30:20.023301] (error)   [ATL]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2
recorder_1  | [2018-10-21 03:30:23.017889] (error)   [ATL] Retuning to Control Channel: 857.462524
recorder_1  | [2018-10-21 03:30:23.018515] (info)   	 - System Source - Min Freq: 851.000000 Max Freq: 859.000000
recorder_1  | [2018-10-21 03:30:23.019777] (error)   [ATL]	 Control Channel Message Decode Rate: 1/sec, count:  3
```

**Rare call status message**
```
recorder_1  | [2018-10-21 03:26:47.025998] (info)   Currently Active Calls: 0
recorder_1  | [2018-10-21 03:26:47.026155] (info)   Recorders:
recorder_1  | [2018-10-21 03:26:47.026305] (info)   [ rtl=00000011 ]
recorder_1  | [2018-10-21 03:26:47.026460] (info)   [ 0 ] State: 2
recorder_1  | [2018-10-21 03:26:47.026634] (info)   [ 1 ] State: 2
```

**config.json**
```
{
    "sources": [
        {
            "center": 855000000.0,
            "rate": 8000000.0,
            "squelch": 0,
            "error": 0,
            "gain": 30,
            "digitalRecorders": 2,
            "driver": "osmosdr",
            "device": "rtl=00000011",
            "modulation": "qpsk"
        }
    ],
    "systems": [
        {
            "control_channels": [857887500,857787500,857487500,857462500],
            "type": "p25",
            "talkgroupsFile": "TG-ATL1.csv",
            "shortName": "ATL",
            "recordUnknown": "false"
        }
    ],
    "frequencyFormat": "mhz"
}
```

rosecitytransit, 2018-12-11:
Just as an FYI, the call status messages is something the program displays on a set timer.  If you're still having the problem, the only thing I can think of is increasing the gain to 40 or 42 or maybe decreasing it.  It can be hard to tell what's going on since there's no visual or audio feedback.  I've never gotten it to work, but there is a debug recorder option that is supposed to record the raw spectrum for analyzing in GNURadio.

heathfx, 2018-12-14:
when I have problems with poor signal I use op25 by itself because of the signal plotter giving me an instantaneous visual indication of signal quality. Your rate of 8000000 is far too high for the rtl-sdr, I found that 2560000 is the absolute max that gives me decent results.

closed 2021-04-28 by robotastic