TrunkRecorder/trunk-recorder #206: ENHANCEMENT - EDACS (Much NEEDED FOR GLOBAL USEABILITY) !BOUNTY!

MaxwellDPS, 2018-12-05:
I am re starting the request for EDACS integration. 
We really neeed edacs to many systemes have it still.
seattle-metro, las vegas, denver, lincoln, reno, LA.... etc

The addition of Edacs needs to be a when not if!

I am offering a Bounty of $15 for edacs integration. I ask you all comment what you will pay for it

src: https://github.com/robotastic/trunk-recorder/issues/129

mostlychris, 2020-04-28:
@rosecitytransit , Assuming there is still no EDACS support, how would I tie unitrunker into trunking recorder? I am running unitrunker on Windows so I've got that part covered.

rosecitytransit, 2020-04-28:
If you install Trunking Recorder, it should see Unitrunker and I think just work.  You have to configure Trunk Recorder to tell it which audio outputs go with which recorders.

robotastic, 2020-07-31:
Let's track this under issue #129

closed 2020-07-31 by robotastic