TrunkRecorder/trunk-recorder #207: High CPU - Parent process

FarvaTechnology, 2019-01-01:
ISSUE
SUPER HIGH CPU consumption on parent process.
1) 200+% on quad core processor.  In other words, 50% of CPU.
2) Only happens on parent process.
3) With my config, there are about 300+ child processes.
4) Issue has been tracked down to: main.cc usleep(1000 * 10);

ENV:
Ubunti 18.05
Linux  4.15.0-43-generic #46-Ubuntu SMP Thu Dec 6 14:45:28 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
Dell T30 Quad Core Xeon E3-1225 (default bios)
7 SDR Radio cards.

CONFIG
7 SDR/RTL Cards w/ driver: osmosdr
Signal for all frequencies is excellent.
Only 6 digital recorders per instance.  So it should be 42 recorders 7 * 6 = 42.
Total talk group entries:  About 40-70 per agency.
4 agencies (towers) total - But only 2 geo areas.  AKA 2 counties

So here is the deal:
Whenever I fire up my p25... It SKY ROCKETS CPU use - Even when there is no radio traffic.

Performing an strace showed me that usleep was causing a nanosleep.   So I replaced usleep with: boost::this_thread::sleep( boost::posix_time::milliseconds(100);  - But didn't see any improvement.

stracing the processes clearing shows a TON of activity relates to sleep.

All child threads show 0.0 - 2.0 CPU use (on quad core).

QUESTION
Why is parent processes killing CPU?

How do we fix this?

Can this wait/sleep interval be increased?   What is the purpose of this wait?

robotastic, 2019-01-01:
First off - that is an impressive setup! well done!

Do you know if a recent change triggered this? I am running 16.04 with 5 rtl-sdrs and haven't had problems. 

Background:
The main process has a loop. Basically, it checks if there is a new trunk message that has come in. If not, it goes to sleep for a sec. There are probably lots of Sleep calls, but it shouldn't be a lot of CPU usage from it. It could show up as CPU time though.

To get a better sense of CPU usage, try:
top -H -p PID-of-RECORDER

FarvaTechnology, 2019-01-01:
During research... There may be kernel concerns and nanosleep.  Something
to do with high precision timers. :(

This is a NEW build... Moved from lubuntu on 2ghz laptop Celeron single
core TO ubuntu 18.x 3ghz quad core Xeon.

Stange part is... It's only parent process.

Try running htop and see what your parent pid is showing for CPU.

I am curious if I must downgrade my OS.

FarvaTechnology, 2019-01-02:
All,

I ran through several different variations of sleep.  usleep, nansleep, boost sleep methods and the like.

The below sleep statement has helped me recover nearly 30% of the CPU that was being consumed.    Because with other sleep/wait methods the CPU was running around 200+% on a quad core.

This helped me realize about 170-180%CPU on a quad.

      std::this_thread::sleep_for(std::chrono::nanoseconds(10));//10 milli = 10,000 micro

The above fix may NOT be helpful for single core processors - But in a multi core ENV - It has helped me.

I am still running around 170% CPU on the parent process.    But from what I have seen, it appears that approximatley 20% of the CPU happens for EACH bank of frequencies I have in my json config file (aka each SDR card).

Since I am running 7 SDR cards at once - This makes since that 20-25% CPU for each card or bank of recorders would add up to around 180%CPU.

I am curious though - I have read that the Xeon processor I am using has some speciality stuff within the BIOS for processing encryption and stuff like that.    It even has failsafe mechanisms to protect data for things like banking applications that wouldn't be found in a consumer grade computer processor.

I am VERY curious if turning off some of those settings in the BIOS will help me realize some realistic numbers for CPU consumption.

If anyone has any input on how to get this reduced - Please let me know.

CPU Type is Xeon E3-1225

FarvaTechnology, 2019-01-03:
UPDATE:
Spoke to Dell engineer today - They "officially" support Ubuntu 16.04 64 bit for the E3-1225 T30 server.   I broken down my old server, did a fresh install.. And excessive CPU usage still exists.

With that said - I don't believe this is a kernel issue.    But I am still not sure.

I will review some of the BIOS settings - Because I believe this processor has special settings for encoding, encryption cracking and banking data verification.    Where I believe any of these "special" processor features could be causing excessive CPU.

If anyone has input that may help - Please update ticket so I can explore those paths to success.

kcwebby, 2019-01-03:
> Where you able to try running ‘ps -H -p (PID of recorder)’? That will tell
> you what process is using up the CPU cycles. I think strace maybe
> misleading in this case. Could you try reducing the number of RTLSDRs? is
> there a dramatic change at a certain number or does it decline linearly?

I run 8 SDR's in all of my recorders (five) ..  I only have high usage on
one of them, and its the most hectic site that I have.  I don't think its
related to the RTL-SDRs.....  I'm following this because I do fight CPU
usage from time to time on that machine.

FarvaTechnology, 2019-01-04:
Here is everything I have so far - If anyone has any input - Please let me know:

**Overview**
I am seeing between 170-200% CPU use on the e3-1225 quad core processor.   I am not sure if this is normal.  

**Issue appears to be related to main.cc->monitor_messages() and usleep(10,000 microseconds).  There is a while loop there with a rapid loop with 10 millisecond waits when no message is present.**

I am not sure if this wait / sleep can be increased... A constant loop with 10 millisecond sleeps seams aggressive to me... But smarter folks than myself have created this code. :)

**My settings**
My recorder config has 7 active cards.   No errors, everything configured and working correctly.

No error rates from rtl_test.

Reception area is GREAT with roof 800mhz antenna - So there shouldn't be any issues with retuning failures or the like.

My config file has 7 sections for frequencies and monitors 3 p25 systems.   

From what I can tell - Firing up recorder shows CPU use of about 15-20% per instance of a "section" within the config.

So... If each section aka card config setting consumes about 20% CPU.   Then it's realistic that 7 cards X 20% would equal 140-200% CPU that I am seeing.

My total number of recorders per section is:
6 recorders per section
6 X 7 = 42 total recorders

**My system is**
Quad core E3-1225 Xeon Processor
8gb of memory

**Operating systems**
Tried 18.04 - Had issues.
Currently using 16.04 since that is the "official" supported OS by Dell for this server (T30) - Same issues.

**strace**
strace of spawned (child) recorders are showing different activity than the parent PID.   The parent PID is show TONS of sleeps - That is coming form main.cc usleep code located inside monitor_messages - while loop else block.

**Analog or Digital?**
All systems are p25.  Code related to this would be: p25_parser->parse_message(msg) - Near the else statement.

**What is being monitored**
Police/Fire and other traffic across two counties is being processed.   Total talk groups are: 229 spread out across 3 systems.
75 TGs - System 1
77 TGs - System 2
77 TGs - System 3

**Things tried:**
I have altered the main code - Tried using usleep, nanosleep as well as calls to boost's sleep methods.   There is an improvement of about 20% reduced CPU by using the boost sleep method.
boost::this_thread::sleep( boost::posix_time::milliseconds(10) );

Downgraded from ubuntu 18.x to 16.x to "officially" use the "certified" operating system that Dell recommends for the T30 server.   Same performance concerns.

==================
Below are details/facts of what I have
==================

**How is recorder started?**
Recorder is started via systemctl with the following config:
[Unit]
Description=p25 Recorder
StartLimitIntervalSec=0

[Service]
Type=simple
Restart=always
RestartSec=3
User=CENSORED
ExecStart=/home/CENSORED/recorder/recorder_start.sh

**What is inside recorder_start?**
[... 57 lines cut ...]
11823 root      20   0 8078604 145188  60344 S  3.6  1.8   6:03.06 fix_cc744
11917 root      20   0 8078604 145188  60344 S  3.6  1.8   6:08.47 fix_cc596
12074 root      20   0 8078604 145188  60344 S  3.6  1.8   0:50.39 fft_filter_cc47
11723 root      20   0 8078604 145188  60344 S  2.6  1.8   3:57.70 feedforward_106
11761 root      20   0 8078604 145188  60344 S  2.6  1.8   3:50.15 copy982
11776 root      20   0 8078604 145188  60344 S  2.6  1.8   3:56.62 copy958
11806 root      20   0 8078604 145188  60344 S  2.6  1.8   4:06.90 copy910
11994 root      20   0 8078604 145188  60344 S  2.6  1.8   3:30.34 copy614
12268 root      20   0 8078604 145188  60344 R  2.6  1.8   3:42.64 copy218
11724 root      20   0 8078604 145188  60344 S  2.3  1.8   2:57.89 gardner_cost106
11731 root      20   0 8078604 145188  60344 R  2.3  1.8   3:37.83 copy1030
11746 root      20   0 8078604 145188  60344 S  2.3  1.8   3:43.37 copy1006
11791 root      20   0 8078604 145188  60344 S  2.3  1.8   4:01.51 copy934
11824 root      20   0 8078604 145188  60344 S  2.3  1.8   3:07.68 copy882
11870 root      20   0 8078604 145188  60344 S  2.3  1.8   3:15.89 copy810
11885 root      20   0 8078604 145188  60344 S  2.3  1.8   3:20.10 copy786
11918 root      20   0 8078604 145188  60344 S  2.3  1.8   3:14.75 copy734
... Truncated for sharing.

**Difference between usleep and using boost or std::thread sleep**
Substantial difference in CPU use.   But still not what I would expect.

[image]

[image]

robotastic, 2019-01-04:
Thanks for running top... recorder should definitely not be chewing up that much CPU. On my quad I7, I never see the main thread consuming any noticeable CPU time. Let me do some googling... Maybe there is some weird compiler thing. It does seem like the switch to boost for sleep is worth it though, I will go test that out.

robotastic, 2019-01-04:
Another thing to note, is variable clock speed. With frequency scaling, my processor will idle down to 800MHz and my CPU utilization will jump up to 200% of 800% but that is for an idled processor. try the cpupower prorgram to see what the processor speed is

robotastic, 2019-01-04:
PS - 10 milliseconds maybe the wrong value. It could be fine to goto 100ms. I would try playing with the number.
Each P25 system should be generating ~40 messages per second, and the sleep is only called when there are no messages left on the queue to be processed.

robotastic, 2019-01-04:
I did find this: https://stackoverflow.com/questions/16726872/cpu-high-usage-of-the-usleep-on-cent-os-6-3 
I wonder if some different kernel flags get set for a Xeon.
There are also different CPU freq management plans, maybe one of those might help ramp the freq up to lower utilization.
Still... sleep should not be using that much CPU.

I will keep digging, just came across this too:
https://serverfault.com/questions/738097/high-cpu-caused-by-pthread-cond-wait-or-nanosleep

FarvaTechnology, 2019-01-04:
ALL - 1st of all - THANK YOU SO MUCH for following this ticket and sharing your configs!   It really means a great deal to me that we have amazing folks supporting each other!   We get faster support here than with most paid services that keep you on hold forever.

Some updates on my end:
1) I disabled ALL things BIOS related that could possibly impact things.   
   Disabled all virtualization
   Disabled all speed enhancement settings related to boosting, hyper, etc.
   Confirmed that all security features related to CPU were disabled.

2) I tried altering the milliseconds of wait by multiples of 4.  10/40/80 - I see a little change.    I am up to 80 milliseconds using std::this_thread::sleep_for(std::chrono::milliseconds(80)) and also trying the original usleep of 8000 X 10.

I will keep at this until I find the issue.    If anyone has ideas, theories, etc.   I am willing to try them.

Thanks again everyone!   YOU ROCK!

FarvaTechnology, 2019-01-04:
Here is a snippet of an strace with usleep(80,000)  aka 80,000 microseconds = 80 milliseconds

I am not familiar w/ usecs per call... But based on my research... It appears that this count is rather high.

Command: strace -T -c -p [PARENT RECORD PID]  - One that is running near 200%.

so... for 1/2 second... There are 32,659 events for 17 calls?    

% time     seconds  usecs/call     calls    errors syscall
------ ----------- ----------- --------- --------- ----------------
 99.00    0.555200       32659        17           nanosleep
  1.00    0.005600        5600         1           restart_syscall
------ ----------- ----------- --------- --------- ----------------
100.00    0.560800                    18           total

FarvaTechnology, 2019-01-04:
**FIX FIX FIX
This reduced my CPU from 200% to about 60%!!!!**

Ok guys and gals - I am not sure what other impact this will have - But I found an article where a guy did a bunch of timing tests using a 3ghz system.
Source: http://charette.no-ip.com:81/programming/2013-08-05_Sleeping/

He recommended this:
**sudo apt-get install cpufrequtils
sudo cpufreq-set -r -g performance**

**NOTE:  The above will be reset on reboot.   To init on boot you can:
vi /etc/init.d/cpufrequtils

If you don't want to set this on boot - But want to set it when you start recorder, you could always add the >> cpufreq-set -r -g performance <<  setting to a .sh file prior to starting recorder.   YOU MUST RUN AS ROOT though.

There are other settings like: ondemand, powersave, etc.   But performance is the ONLY setting that fixed my issue.

**Results:**
I executed the above and my CPU dropped DRAMATICALLY!!!!   And this is using the default usleep(1000 * 10) code that already exists within recorder source main.cc.

**Related articles:**
https://lenovopress.com/lp0826.pdf

https://www.linuxdays.cz/2017/video/Giovanni_Gherdovich-CPU_frequency_scaling.pdf

https://askubuntu.com/questions/410860/how-to-permanently-set-cpu-power-management-to-the-powersave-governor

**Other Notes**
Article https://lenovopress.com/lp0826.pdf spoke about "C-State" and although the T30 Dell server has C-State defined within the BIOS - Turning it on/off had no effect on the high cpu use I encountered.

THis is WAY COOL - Because I can now leverage more resources from my server to add more systems and regions for listening/recording!

**Summary**
Please research this guys before implementing on your system.   If this turns out to be the "fix" for the high CPU use on linux - This may be a nice addition to the installation instructions.

robotastic, 2019-01-04:
Awesome!!
That makes a lot of sense... your CPU was throttling back the CPU freq and as a result, the % CPU usage was up. Best to think of it as (CPU Freq / Usage) or something like that. I keep mine in "power-save" mode instead of Performance, to lower power usage and keep the fan off. 

Try running the same 6 systems in Power-Save mode. It should auto scale the Freq and handle things just as well. It just opts to keep the CPU freq a little lower and CPU utilization a little higher.

This all still doesn't explain all the nano_sleep syscalls. I am going to look into this more. 

One last thing to check, can you look at the number of control channel messages you are decoding for each system? It should be about 38 msgs per second. Even with good audio decode you can have low control channel decode.

To get this, add "controlWarnRate": -1 to the top level of config.json

If the Message rate is low, sleep will be called a lot. With 6 systems though, there should be tons of messages.

FarvaTechnology, 2019-01-04:
robo - Thanks again!  You guys rock!

I tried the powersave mode - And the CPU just shot back up to 200+% again. :(

According to Dell - The BIOS is supposed to handle a ton of this stuff.   And even though ubuntu is "certified" for this server (16.04 only) - It appears that some of the BIOS settings don't work inside Ubuntu for some reason.   One would think it wouldn't matter.

As far as the nanosleep and Linux - There are a TON of articles out there of people having issues with the newer Kernels and sleep in some multi-core ENVs.   Especially on GHZ systems.

One article I found said that the source code needs to test for the clock speed of the CPU and adjust it's coding to map the correct microseconds or milliseconds for sleep applications.    There was even some articles talking about testing between linux OS and Windows and how that differs.

ONLY ME would elect to use a Xeon processor!  hahaha

But hey!  At least we found a hack.

Thanks again to everyone!   Your help is amazing.

Lets keep this ticket open a few more days to see how everything runs... And if this appears to be a solution, we can close it out if that is good with you guys.

FarvaTechnology, 2019-01-07:
**STATUS UPDATE**

All... The addition of the cpufreq tools fixed my problem!   I have successfully ran several days with AMAZING server efficiency!!!!

Before installing cpufreq tools - I had 200+ % CPU (4 core) and I was only able to monitor two p25 systems.   Even then my average server load was nearly 6-8 (way above the 4.0 cutoff).

After adding cpufreq tools - I was able to add several more p25 systems.   And I have consistently been running at 100%CPU or so (out of 400%) and sever load is 1.96 on the 15 minute average scale!

With this increased performance... I could probably squeeze in 2-3 more p25 systems and stay under my 4.0 server average threshold.

The only note is... YOU MUST RUN CPUFREQ TOOL AFTER trunk-recorder is up and running.

I would consider this ticket closed.

bctrainers, 2019-01-08:
Interesting findings there. For me, I use a HackRF One to monitor a singular P25 system. So there's no array of RTLSDR's going on here.  It is easily able to cover the entire swath of spectrum that the P25 system is configured to use - and then some. I too, have the moderate to high CPU usage, only with 10 recorders configured and ready from the single HackRF. Anything less with the CPU cores, I end up seeing OOOOOOOOOOO's all over the place in the console. Decode rates are quite bouncy.... from 24 to 40/sec.

For this VM, i've thrown all 12 cores available at this particular installation. As you can see below, it's having a joyride with all the cores. 

VM: 
[image]
VM Host: 
[image]

Now to find something like cpufreq that works on a VMWare ESXi/Workstation system......

closed 2020-08-04 by robotastic