TrunkRecorder/trunk-recorder #208: P25 PHASE 2 IS NOT RECEIVING

MaxwellDPS, 2019-01-05:
Hey all,

I have a new P25 P2 system in my area and I can decodit with unitrunker and with op25( see attached) but not with trunk recorder... does trunk recorder's op25 need to be updated am I doing something wrong?

below is op25, config.json and TR output
[image]

```
{
    "sources": [{
        "center": 853000000,
        "rate": 2048000,
        "error": 2000,
        "gain": 39,
        "digitalRecorders": 4,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=0,buflen=65536"
    }, {
        "center": 855000000,
        "rate": 2048000,
        "error": 2000,
        "gain": 39,
        "digitalRecorders": 4,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=1,buflen=65536"
    }, {
        "center": 857000000,
        "rate": 2048000,
        "error": 2000,
        "gain": 39,
        "digitalRecorders": 4,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=2,buflen=65536"
    }, {
        "center": 859000000,
        "rate": 2048000,
        "error": 2000,
        "gain": 39,
        "digitalRecorders": 4,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=3,buflen=65536"
    }],
    "systems": [{
        "control_channels": [854687500],
        "type": "p25",
        "talkgroupsFile": "/opt/recorder/nesr_tr.csv",
        "shortName":"NESR",
        "uploadScript": "/opt/recorder/encode.sh"
    }],
"captureDir":"/home/radio/trunk-player/audio_files",
"logFile": true,
"statusServer":"http://127.0.0.1:3010/server"
}
```

```
max@RadioLime:/opt/recorder$ ./recorder
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

[2019-01-04 20:39:48.029330] (info)   Using Config file: ./config.json

[2019-01-04 20:39:48.030357] (info)   
-------------------------------------
SYSTEMS
-------------------------------------
0
[... 381 lines cut ...]
[2019-01-04 20:43:13.010212] (info)   Recorders: 
[2019-01-04 20:43:13.010259] (info)   [ rtl=0,buflen=65536 ]  
[2019-01-04 20:43:13.010307] (info)   [ 0 ] State: 2
[2019-01-04 20:43:13.010366] (info)   [ 1 ] State: 2
[2019-01-04 20:43:13.010418] (info)   [ 2 ] State: 2
[2019-01-04 20:43:13.010468] (info)   [ 3 ] State: 2
[2019-01-04 20:43:13.010519] (info)   [ rtl=1,buflen=65536 ]  
[2019-01-04 20:43:13.010583] (info)   [ 4 ] State: 2
[2019-01-04 20:43:13.010635] (info)   [ 5 ] State: 2
[2019-01-04 20:43:13.010685] (info)   [ 6 ] State: 2
[2019-01-04 20:43:13.010734] (info)   [ 7 ] State: 2
[2019-01-04 20:43:13.010784] (info)   [ rtl=2,buflen=65536 ]  
[2019-01-04 20:43:13.010831] (info)   [ 8 ] State: 2
[2019-01-04 20:43:13.010881] (info)   [ 9 ] State: 2
[2019-01-04 20:43:13.010930] (info)   [ 10 ] State: 2
[2019-01-04 20:43:13.010980] (info)   [ 11 ] State: 2
[2019-01-04 20:43:13.011048] (info)   [ rtl=3,buflen=65536 ]  
[2019-01-04 20:43:13.011100] (info)   [ 12 ] State: 2
[2019-01-04 20:43:13.011143] (info)   [ 13 ] State: 2
[2019-01-04 20:43:13.011184] (info)   [ 14 ] State: 2
[2019-01-04 20:43:13.011234] (info)   [ 15 ] State: 2
[2019-01-04 20:43:16.006507] (error)   [NESR]   There is only one control channel defined
[2019-01-04 20:43:16.006619] (error)   [NESR]    Control Channel Message Decode Rate: 1/sec, count:  3

```

UnderwaterHockey, 2019-01-05:
An error of 2000 for each rtl, what is that based on?  Try setting that to zero and see if it improves.

MaxwellDPS, 2019-01-05:
> An error of 2000 for each rtl, what is that based on? Try setting that to zero and see if it improves.

I have... its a ppm of -2 the freq needed is 854.687500 what they are at is 854.689500
854.687500- 854.689500 = 2000 hence the setting but yes I have tried on all sdrs and tried an error of 0.
however an error of -2 ppm is common for rtl-sdr.com v3 dongles

based on that I think it might be the version of OP25 as with OP25 I use the same offset

robotastic, 2019-01-05:
2000 does seem a little high for a v3. Maybe try something in the middle, like 1000. Also try -2000 & -1000. The gain is also pretty high, maybe try a little lower. 

I haven’t tried phase 2 in a while but I will see if I can get my local system working to confirm everything is working. 

📱

MaxwellDPS, 2019-01-05:
That did it! I feel so stupid I swear i tweaked those already :\ now I an seeing this after every transmission though? any thoughts?

 ```
[2019-01-05 16:04:13.004622] (error)   Call::get_source_list State is recording, but no recorder assigned!
[2019-01-05 16:04:13.004940] (error)   Call::get_source_count State is recording, but no recorder assigned!
```

FarvaTechnology, 2019-01-06:
I noticed your log said: only one control channel defined.

I have come to live this message.  Because it tells me that my config file is wrong.

I have noticed that you can specify 0,1,2,3 and such for your device value... And it will use the INDEX of your cards.  BUT... If you specify rtl=... You MUST use the serial number of your card.

I see in your config... That your RTL is a single digit... But your serials are 000101, 103 and the like.

If you can't find to a p25 tower... It's normally due to:
Device value: index VS serial issue.
Gain is too low.
Your control channel is incorrect.

Get those 3 magic solutions in check.. and you should realize success.

Also see:rtl_test. For an index/serial list of your cards.

I hope this helped.

Todd

vabiro, 2019-02-17:
@MaxwellDPS Todd had suggested multiple things, but you didn't mention which seemed to resolve the `Call::get_source_list` and `Call::get_source_count` messages.

Everything seems to work, but I am seeing an awful lot of these messages.

Thanks
Victor