TrunkRecorder/trunk-recorder #211: have Json server also send encrypted calls to status page

MaxwellDPS, 2019-01-07:
have Json server also send encrypted calls to status page I fixed the code for it to work the encrypted calls just need to be sent out as calls_active

FarvaTechnology, 2019-01-07:
**Use caution with encrypted data in the US.**

I maintain a TON of certifications via Homeland Security on Cyber related stuff and legal certs on cyber topics.  (I am NOT a lawyer!)

Based on my training - De-crypting is illegal in most instances.   For legal reasons, I make sure that my p25 recorder instances are all set to IGNORE.   Because I don't want ANY storage of ANYTHING related to "encrypted" data.

Not even a database entry that says "ENCRYPTED".

I am sure nobody would de-crypt stuff - But even storing portions of stuff that is related to encryption could be an invitation for an over achieving prosecutor who wants to make a name for themselves.

It would be MY recommendation to:
1) Ignore ALL encrypted data.

2) Don't store ANYTHING related to encryption.

3) Document your Talkgroup file against Radio Reference to make sure that NOTHING in your list is "attempting" to receive Encrypted traffic.   (Create a baseline to prove that's what Radio Ref said when you created your TG file).

4) If you have permission to de-crypt and you have the proper keys... PLEASE make sure you document the daylights out of it.

In today's modern age... Many cyber teams are looking to increase budgets.   They do this by racking up convictions.   So please use caution!

And if folks don't think the authorities are not watching this stuff... They are!   There was a recent case in Ohio (Starke County I believe) - Where folks were re-programming radios with encrypted keys and selling them to bad guys.   I don't know if SDR devices were involved.

**Just use caution with anything encrypted guys.**

kcwebby, 2019-01-07:
I can't say I disagree with you..  However, I would like to ensure that we
are accurately recording the volume of calls on the system. While the
status page does not record to the database, and E calls are generally
disregarded by everything, I would like to see some statistical
representation (or the ability to) of what those calls make up.

BTW, on my system, they aren't yet using system level encryption, so we
record (unintentionally) encrypted calls all the time.   The encryption is
done on the radio, and not at the system level, so the system doesn't know
we need to ignore those calls.   This happens alot of switched encryption,
where one or two radios are encrypted, and some are not.

gdt, 2019-01-07:
It seems that there are many legitimate reasons to want to understand what's happening on a channel, separately from any attempt to decrypt.  One would be usage assessment, for system or frequency planning.   Another might be a journalist comparing the volume of encrypted traffic to stated policies.   A third might be monitoring a ham radio DMR system, so that the operators would become aware of users improperly using encryption.  It seems overly paranoid to view situational awareness of spectrum use as evidence of intent to violate some statute.  (Someone having obtained keys and selling radios to criminals is entirely different.)

If there is a blog post, web page, etc. that has a reasoned analysis with citations and specific statutes and links to cases, that would be interesting, as the law doesn't have to make sense.

leee, 2021-05-15:
@MaxwellDPS since the conversation has continued on #462, this can be closed.

closed 2021-05-15 by MaxwellDPS