TrunkRecorder/trunk-recorder #213: compile against system op25

ZeroChaos-, 2019-01-10:
op25 is bundled with trunk-recorder, which is great for a lot of people.  But some of us want to build against a different op25 fork or just the op25 we already have installed on our system.  please add a build flag to allow building against the system version of op25 instead of building and using the bundled one.

robotastic, 2019-01-11:
This could be pretty tough. I have done a bit of forking on OP25 to add things for pulling out decode errors. It could also be a support nightmare if versions change or people have trouble linking to the Libs. Are there some specific changes you have made to OP25? I would be open to pull requests against the embedded OP25

ZeroChaos-, 2019-01-11:
I was more thinking like https://github.com/boatbod/op25

One of the biggest issues with op25 (and sdr software in general) is nearly everything feels like abandonware with everyone having their own fork of this and that and nothing even going upstream.  It's hard for users to know which version of what to use.

If it's hard to use a different op25 than bundled, then maybe you could add a flag to staticly link just your op25 so it doesn't depend on the solib and I can have my system version installed and not conflicting?

I'd rather a unified op25 repo that people actually care about, but I think that's a much larger request than you or I can fulfill :-)

robotastic, 2024-12-15:
we are probably stuck with a custom fork of OP25 because of we need to add some special handlers in for thing like Source ID

closed 2024-12-15 by robotastic