TrunkRecorder/trunk-recorder #218: Strange P25 P2 TDMA 0 second issue..

Cole0209, 2019-01-13:
One of my trunked p25 systems records 44byte files that are empty. I switched systems and it seems to work ok so it ... When it uploads the files to openmhz the files are unplayable and are 0 seconds long. The other systems work so I presume my dongles are ok (2 RTLSDRs). Any help is appreciated.

Here's a sample of the issue

`01-12 14:55:49.303750]: [berksfire]       TG: 8009        Freq: 7.70094e+08       HTTPS Upload Success - file size: 2345
[2019-01-12 14:55:51.393995]: [berksfire]       TG: 28101       Freq: 7.700938e+08      Update Retuning - New Freq: 7.706438e+08        Elapsed: 25s    Since update: 3s
[2019-01-12 14:55:52.002292]: [berksfire]       TG: 28202       Freq: 7.706438e+08      Ending Recorded Call - Last Update: 4s  Call Elapsed: 26
[2019-01-12 14:55:52.002748]:   - Stopping P25 Recorder Num [2] TG: 28202       Freq: 7.706438e+08      TDMA: true      Slot: 1
[2019-01-12 14:55:52.287299]: [berksfire]       TG: 28202       Freq: 7.70644e+08       HTTPS Upload Success - file size: 2374
[2019-01-12 14:55:55.898523]: [berksfire]       TG: 2015        Freq: 7.706438e+08      Not Recording: TG not in Talkgroup File
[2019-01-12 14:56:05.525272]: [berksfire]       TG: 4001        Freq: 7.706438e+08      Not Recording: TG not in Talkgroup File`

robotastic, 2019-01-18:
Try the latest version - I had a bug in there for a little around Phase 2.

Cole0209, 2019-01-18:
Just updated it. Issue seems to still be happening but only on one system. I switched back to the backup system after 9:34:28 https://openmhz.com/system/berksfire. The issue one is the 0 seconds from 9:27:22-9:34:28.

robotastic, 2019-01-18:
Welp - Looks like it is still broken. Doh! I will roll things back. I don’t have good enough systems to test against.

Cole0209, 2019-02-09:
Finally I figured it out! The RTL- SDRs I have been using for almost a year have been at 2.56samples. I figured I would try rtl_test and I saw that for some reason (probably because they have been in use almost 24/7 for 1 year) they no longer run at that high sample rate. Turned it down to 2.4samples and it runs perfectly.

closed 2019-08-14 by Cole0209