TrunkRecorder/trunk-recorder #222: How do I update?

johnrockie, 2019-02-05:
Hello,

No issues here but I am not sure how github works. I got a P25 system working about 6 months ago. I noticed that there have been some "commits" (i guess these are updates?). I have issues like 0 second files being uploaded and to my understanding there has been a fix for issue #209 in latest commit. Im a little noob so I would appreciate some help on how to update, I am running the system on a Ubuntu machine. I tried downloading the Zip and simply moving the files to the folder but that didn't seem to work. 

Would someone mind explaining the proper way to update? Thank you!

johnrockie, 2019-02-16:
Probably not the correct way of doing it but I simply deleted both trunk-build and trunk-recorder and rebuilt it using this method https://github.com/robotastic/trunk-recorder/wiki/Building-Trunk-Recorder

closed 2019-02-16 by johnrockie

bctrainers, 2019-02-28:
On the git directory where you used `git clone`, type `git pull`. 

Your re-building process should look something like this:

```
cd trunk-recorder/
git pull
cd ../trunk-build/
cmake -j ../trunk-recorder
make -j
```

Note the `-j`  flag tells the compiler to use ALL available CPU cores. When using `-j` with `make`, memory usage _will_ skyrocket. Compiling Trunk Recorder can easily consume more than 8GB RAM while the compiler is running. 

If you do not have the machine memory resources available, use `-j` with a CPU value that is half (or a quarter) of all your CPU cores. So for example, if you've 12 CPU cores, you would use `-j6` to not use all cores, but half of all cores.

The -j is optional, but can greatly speed up compile time if you've the resources.

@robotastic it might be worth while adding the above to the Building Trunk Recorder wiki page as a side-section for updating/syncing Trunk Recorder.