TrunkRecorder/trunk-recorder #225: cant find config.json file ! how to create it and run recorder ??

khanfar, 2019-02-21:


johnrockie, 2019-02-21:
Check out [this page](https://github.com/robotastic/trunk-recorder/wiki/Running-Trunk-Recorder) on creating the config.json and running the 'recorder' file. You can also check out some examples of the config file [here](https://github.com/robotastic/trunk-recorder/tree/master/examples). Good luck!

UnderwaterHockey, 2019-02-21:
In the "examples" folder there are a bunch of example files depending on what source you are using.  Find one that suits your needs, edit it accordingly, rename it config.json, and then move it to the main trunk recorder folder.

khanfar, 2019-02-21:
I found it , but wich folder should i chose for recording from op25 software ? Op25 run on 3 monitor frequency on range 866mhz , i use one normal rtl2832u dongle , what should i do now to make it record the op25 when i run it ? Im sorry im beginner. Underwater Hockey

khanfar, 2019-02-21:
In trunk-recorder folder i cant find ./recorder file ! I see it on trunk-build folder , what should i do ? Copy ./recorder from build folder to trunk-recorder folder ?

johnrockie, 2019-02-21:
> In trunk-recorder folder i cant find ./recorder file ! I see it on trunk-build folder , what should i do ? Copy ./recorder from build folder to trunk-recorder folder ?

Please take a look at my post. Read it carefully, "From your **build** directory **(e.g. trunk-build)** you can now run ./recorder"

khanfar, 2019-02-21:
Yes in trunk-build i see recorder and when i try ./recorder i got error
faild parsing config ./config.json canot open file ??? Look in picture,
should i need to put in this directory i run ./record , does need to put
like 1.json file ? And how to run command ?

khanfar, 2019-02-21:
Does i need to put in trunk-build folder config.json file  ?

johnrockie, 2019-02-21:
Yes, the config.json needs to be in the trunk-build folder. This is explained in the post aswell.....
https://github.com/robotastic/trunk-recorder/wiki/Running-Trunk-Recorder
> Trunk Recorder will look for a config.json file in the same directory as it is being run in.

khanfar, 2019-02-21:
I put config.json in same recorder folder now start ./recorder and i change the capture directory in /home/mwk/trunk-recorder , does that ok ? But now i got in terminal all error dcfems there is only one control channel all error and control channel message decode  , all terminal like this ! What mistske i have ? The op25 run in 3 monitor frequency,  whats should i do now ? All three monitor frequency on op25 on 866.800 , 866.650 , 867.800   what should i do now ????

khanfar, 2019-02-21:
After run ./recorder

khanfar, 2019-02-21:
What frequency should i add in ./config.json ? I run op25 on 3 monitor frequency,  what should i put in ./config.json ???

johnrockie, 2019-02-21:
@khanfar , I wish I had more time to help and I hope someone else does but it certainly does not seem like you've gone through the Wiki page install notes. Please take a look at this https://github.com/robotastic/trunk-recorder/wiki

Go through the whole install process and It should get you closer if not solve all of your issues and get the system running.

Good Luck!

khanfar, 2019-02-21:
ok im looking now , thanks for your help but i have miss understand , should i put center frequency is the same control frequency ? and add 3 center frequency in config.json , does this correct ??  or put center frequency for every sdr dongle i need to use . and put  "control_channels": [855462500], together ? please post example to record three x frequency from p25 . thanks in advance  and im soory about your time .

khanfar, 2019-02-21:
op25 now running , but main question DOES I NEED TO ADD ANOTHER DONGLE FOR RECORDING ?

johnrockie, 2019-02-21:
Can you post what trunk system you are trying to record? What is the radioreference link? Please also post your config.json of what you have so far.

khanfar, 2019-02-21:
p25 plase 2 c4ft running ok in op25 with 3 control frequencies .  i try to put in file 1 channel of 3 like this and put another dongle . should i need this dongle number 2 or no need for that ? look 

{
    "sources": [{
        "center": 857000000.0,
        "rate": 2000000.0,
        "error": 1400,
        "gain": 36,
        "digitalRecorders": 4,
        "driver": "usrp"
    }],
    "systems": [{
        "control_channels": [855462500],
        "type": "p25",
        "talkgroupsFile": "ChanList.csv"
   
    }],
    "captureDir": "/home/mwk/trunk-recorder/"
}

khanfar, 2019-02-21:
in my op25 c4fm need modulation options qpsk or fsk4 ??????

khanfar, 2019-02-21:
the op25 run on sdr#2 and this is my latest config.json file , i put sdr#1 in config for recording : 

{
    "sources": [{
        "center": 866800000,
        "rate": 2048000,
        "error": 1000,
        "gain": 39,
        "debugRecorders": 0,
        "digitalRecorders": 4,
        "modulation": "fsk4",
        "driver": "osmosdr",
        "device": "rtl=khanfar1"
    }],
    "systems": [{
        "control_channels": [866825000],
        "type": "p25",
        "talkgroupsFile": "ChanList.csv",
				"shortName": "dcfems"
    }],
    "captureDir": "/home/mwk/trunk-recorder/"
}

---------------------------------------------------------------

the result is : 

mwk@mwk-STCK1A32WFC:~/trunk-build$ ./recorder
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

[2019-02-22 00:37:44.426212] (info)   Using Config file: ./config.json

[2019-02-22 00:37:44.428262] (info)   
-------------------------------------
SYSTEMS
-------------------------------------
0
[2019-02-22 00:37:44.429284] (info)   

System Number: 1
-------------------------------------
1
[2019-02-22 00:37:44.430122] (info)   Short Name: dcfems
[2019-02-22 00:37:44.430379] (info)   System Type: p25
[2019-02-22 00:37:44.430626] (info)   Control Channels: 
[2019-02-22 00:37:44.430952] (info)   8.66825e+08 
[2019-02-22 00:37:44.431271] (info)   API Key: 
[2019-02-22 00:37:44.431528] (info)   Upload Script: 
[2019-02-22 00:37:44.431779] (info)   Call Log: true
[2019-02-22 00:37:44.432037] (info)   Audio Archive: true
[2019-02-22 00:37:44.432291] (info)   Loading Talkgroups...
[2019-02-22 00:37:44.441901] (info)   Read 93 talkgroups.
[2019-02-22 00:37:44.442244] (info)   Talkgroups File: ChanList.csv
[2019-02-22 00:37:44.442521] (info)   Record Unknown Talkgroups: true
[2019-02-22 00:37:44.442799] (info)   Talkgroup Display Format: Id
[2019-02-22 00:37:44.443114] (info)   Hide Encrypted Talkgroups: false
[2019-02-22 00:37:44.443381] (info)   Hide Unknown Talkgroups: false
[2019-02-22 00:37:44.443647] (info)   Minimum Call Duration (in seconds): 0
[2019-02-22 00:37:44.443928] (info)   
[2019-02-22 00:37:44.444188] (info)   

-------------------------------------
SOURCES
-------------------------------------

[2019-02-22 00:37:44.445402] (info)   Driver: osmosdr
[2019-02-22 00:37:44.445671] (info)   Center: 8.668000e+08
[2019-02-22 00:37:44.446022] (info)   Rate: 2048000
[2019-02-22 00:37:44.446337] (info)   Error: 1000
[2019-02-22 00:37:44.446618] (info)   PPM Error: 0
[... 68 lines cut ...]
[2019-02-22 00:37:44.896147] (info)   Control channel warning rate: 10
[2019-02-22 00:37:44.896232] (info)   Control channel retune limit: 0
[2019-02-22 00:37:44.896319] (info)   Frequency format: 0
[2019-02-22 00:37:44.896399] (info)   Status as String: false
[2019-02-22 00:37:44.896516] (info)   [dcfems]	Started with Control Channel: 8.668250e+08
[2019-02-22 00:37:44.915005] (info)   	 P25 Trunking - SysNum: 0
[2019-02-22 00:37:44.915220] (info)   	 P25 Recorder Initial Rate: 512000 Resampled Rate: 51200 Initial Decimation: 4 Decimation: 10 System Rate: 48000 ARB Rate: 0.9375
[2019-02-22 00:37:44.916264] (info)   P25 Recorder Taps - initial: 311 channel: 1163 ARB: 775 Total: 2249
[2019-02-22 00:37:47.008788] (error)   [dcfems]	There is only one control channel defined
[2019-02-22 00:37:47.008962] (error)   [dcfems]	 Control Channel Message Decode Rate: 0.333333/sec, count:  1
[2019-02-22 00:37:50.005547] (error)   [dcfems]	There is only one control channel defined
[2019-02-22 00:37:50.005688] (error)   [dcfems]	 Control Channel Message Decode Rate: 1/sec, count:  3
[2019-02-22 00:37:53.008797] (error)   [dcfems]	There is only one control channel defined
[2019-02-22 00:37:53.009080] (error)   [dcfems]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2
[2019-02-22 00:37:56.011870] (error)   [dcfems]	There is only one control channel defined
[2019-02-22 00:37:56.012017] (error)   [dcfems]	 Control Channel Message Decode Rate: 1/sec, count:  3
[2019-02-22 00:37:59.010095] (error)   [dcfems]	There is only one control channel defined
[2019-02-22 00:37:59.010304] (error)   [dcfems]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2
[2019-02-22 00:38:02.014294] (error)   [dcfems]	There is only one control channel defined
[2019-02-22 00:38:02.014611] (error)   [dcfems]	 Control Channel Message Decode Rate: 1/sec, count:  3
[2019-02-22 00:38:05.014356] (error)   [dcfems]	There is only one control channel defined
[2019-02-22 00:38:05.014585] (error)   [dcfems]	 Control Channel Message Decode Rate: 1/sec, count:  3
[2019-02-22 00:38:08.009857] (error)   [dcfems]	There is only one control channel defined
[2019-02-22 00:38:08.010179] (error)   [dcfems]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2
[2019-02-22 00:38:11.007654] (error)   [dcfems]	There is only one control channel defined

mdozier9, 2019-02-21:
The number of sources you need depends on the system you are monitoring, the bandwidth of the SDR device you are using, and the sample rate.  Can you post the RadioReference link to the site you are trying to monitor?  Its not just about control frequencies.  You have to be able to cover all possible frequencies for the particular system you are monitoring.

mdozier9, 2019-02-21:
Also, your RTL device setting of "device": "rtl=khanfar1" appears to be either wrong, or your RTL was not plugged in.  Typically you will specify the serial number that you programmed using rtl_eeprom.  You set your RTL serial number first using rtl_eeprom, then you specify that serial number in your source.  For example: "device": "rtl=00000007,buflen=65536", would use RTL device number 00000007.

khanfar, 2019-02-21:
i use eeprom and put my serial number = khanfar1 , so i need to put all alternative frequency ? not just control frequency ??

khanfar, 2019-02-21:
and my question two is : im now running op25 on sdr like number 1 , so i need sdr number 2 to run ./recorder ??? just or need another sdr for alternative frequency ??? i mean 1 sdr for op25 and 1 for recorder just ??? or need another one for alternative frequency ??? 1 for control frequency and one for alternative frequency and one for op25 ?????? correct me ?????

khanfar, 2019-02-21:
were should i need to put all possible frequensies ?? on "talkgroupsFile": "ChanList.csv",  ????

s3m1s0n1c, 2019-03-01:
🤕 
@khanfar You really need to read the Wiki as multiple people pointed out...
Its mostly explained there how it all works..

Maybe read [https://github.com/robotastic/trunk-recorder/blob/master/README.md](url) and you will see what you need..

Look at this example file and see how its setup.. [https://github.com/robotastic/trunk-recorder/blob/master/examples/config-dcfd-rtl.json](url)

closed 2020-07-31 by robotastic