TrunkRecorder/trunk-recorder #227: Uploading files to OpenMHZ results in server error

uid0, 2019-02-22:
I'm attempting to upload my traffic to OpenMHZ, but, am receiving a 500 in doing so:

```[2019-02-22 04:05:31.081686] (info)   Error opening file 
[2019-02-22 04:05:31.211603] (info)   HTTPS: Response returned with status code 500
[2019-02-22 04:05:31.211810] (error)   [OHP25]  TG: 55651       Freq: 8.53638e+08       HTTPS Upload Error - file size: 1490```

I do have ffmpeg built with m4a support.  I do have m4a files in my recorded files directory.

Any information to assist me in debugging this would be helpful!

s3m1s0n1c, 2019-02-27:
What happens if u type the following command in your audio dir
``ffmpeg -y -i **(Insert Wav File Here)**  -c:a libfdk_aac -b:a 32k -filter:a "volume=15db" -filter:a loudnorm -cutoff 18000 test.m4a``

Do any errors come up in ffmpeg?

s3m1s0n1c, 2019-02-27:
I was getting an error for Unknown encoder 'libfaac'
I did a ``sudo apt-get install libfaac-dev`` then recompiled ffmpeg as per link below
https://github.com/robotastic/trunk-recorder/wiki/Uploading-to-OpenMHz

I'm still getting a 500 Error tho..

I can see the m4a files in the directory but its not uploading.. No firewalls or anything..

s3m1s0n1c, 2019-03-03:
So the solution for this is a pretty simple one if your still getting a 500 error after installing libfaac..
The shortname has to be **lower case** in openmhz and in your config.

robotastic, 2019-03-04:
Yep - I have a fix coming out for this soon... but it got tangled up in a
bunch of other changes I am making. Should have it out this week.

s3m1s0n1c, 2019-03-04:
Thanks @robotastic

closed 2021-04-28 by robotastic