TrunkRecorder/trunk-recorder #228: Failed parsing Config using Docker

Pacmannion, 2019-02-24:
Trying to setup a 3 dongle setup using the Docker method, tried running it and got this error:

````recorder_1  | [2019-02-24 19:56:02.553193] (info)   Using Config file: /app/config/config.json
recorder_1  | 
recorder_1  | [2019-02-24 19:56:02.555798] (error)   Failed parsing Config: /app/config/config.json(42): expected ',' or ']'
recorder_1  | [2019-02-24 19:56:02.555835] (error)   Unable to setup a System to record, exiting...
recorder_1  | 
recorder_1  | linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown
recorder_1  | 
````
Here's my config.json:
````
{
    "sources": [{
        "center": 854912500.0,
        "rate": 8000000.0,
        "squelch": -50,
        "error": 0,
        "gain": -10,
        "digitalRecorders": 4,
        "driver": "osmosdr",
        "device": "rtl=0",
        "modulation": "qpsk"
    }, {
    "sources": [{
        "center": 857437513.0,
        "rate": 8000000.0,
        "squelch": -50,
        "error": 0,
        "gain": -10,
        "digitalRecorders": 4,
        "driver": "osmosdr",
        "device": "rtl=1",
        "modulation": "qpsk"
    }, {
    "sources": [{
        "center": 859312500.0,
        "rate": 8000000.0,
        "squelch": -50,
        "error": 0,
        "gain": -10,
        "digitalRecorders": 3,
        "driver": "osmosdr",
        "device": "rtl=2",
        "modulation": "qpsk"
    }],
    "systems": [{
        "control_channels": [857012500,857437500,858437500,859437500],
        "type": "p25",
        "talkgroupsFile": "ChanList.csv"
    }],
    "captureDir": "/app/media"
}
````

Pacmannion, 2019-02-24:
Well, I did just realize my sample rate was too high. It's been set to 2 now. Nevertheless, still getting the same error.

dreinhold, 2019-02-24:
You only want 1 sources key so your first one, the just },{ between each source setup
"sources": [{
        "center": 854912500.0,
        xxxxxxxxx
        xxxxxxxxx
   }, {
        "center": 854912500.0,
        xxxxxxxxxx
       xxxxxxxxxxx
 }]

Pacmannion, 2019-02-24:
Well, that worked. Now there's a new issue... getting errors after running the instance.

````
recorder_1  | 
recorder_1  | [2019-02-24 22:41:21.397079] (info)   Capture Directory: /app/media
recorder_1  | [2019-02-24 22:41:21.397101] (info)   Upload Server: 
recorder_1  | [2019-02-24 22:41:21.397118] (info)   Status Server: 
recorder_1  | [2019-02-24 22:41:21.397135] (info)   Instance Key: 
recorder_1  | [2019-02-24 22:41:21.397150] (info)   Instance Id: 
recorder_1  | [2019-02-24 22:41:21.397166] (info)   Default Mode: digital
recorder_1  | [2019-02-24 22:41:21.397182] (info)   Call Timeout (seconds): 3
recorder_1  | [2019-02-24 22:41:21.397198] (info)   Log to File: false
recorder_1  | [2019-02-24 22:41:21.397218] (info)   Control channel warning rate: 10
recorder_1  | [2019-02-24 22:41:21.397237] (info)   Control channel retune limit: 0
recorder_1  | [2019-02-24 22:41:21.397256] (info)   Frequency format: 0
recorder_1  | [2019-02-24 22:41:21.397273] (info)   Status as String: false
recorder_1  | [2019-02-24 22:41:21.397301] (info)   [sys_1]	Started with Control Channel: 8.570125e+08
recorder_1  | [2019-02-24 22:41:21.401632] (info)   	 P25 Trunking - SysNum: 0
recorder_1  | [2019-02-24 22:41:21.401694] (info)   	 P25 Recorder Initial Rate: 500000 Resampled Rate: 50000 Initial Decimation: 4 Decimation: 10 System Rate: 48000 ARB Rate: 0.96
recorder_1  | [2019-02-24 22:41:21.401812] (info)   P25 Recorder Taps - initial: 303 channel: 1137 ARB: 757 Total: 2197
recorder_1  | [2019-02-24 22:41:24.021620] (error)   [sys_1] Retuning to Control Channel: 8.574375e+08
recorder_1  | [2019-02-24 22:41:24.021758] (info)   	 - System Source - Min Freq: 8.564376e+08 Max Freq: 8.584376e+08
recorder_1  | [2019-02-24 22:41:24.022427] (error)   [sys_1]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2
recorder_1  | [2019-02-24 22:41:27.014877] (error)   [sys_1] Retuning to Control Channel: 8.584375e+08
recorder_1  | [2019-02-24 22:41:27.015035] (info)   	 - System Source - Min Freq: 8.564376e+08 Max Freq: 8.584376e+08
recorder_1  | [2019-02-24 22:41:27.015950] (error)   [sys_1]	 Control Channel Message Decode Rate: 1/sec, count:  3
recorder_1  | [2019-02-24 22:41:30.016081] (error)   [sys_1] Retuning to Control Channel: 8.594375e+08
recorder_1  | [2019-02-24 22:41:30.016261] (info)   	 - System Source - Min Freq: 8.564376e+08 Max Freq: 8.584376e+08
recorder_1  | [2019-02-24 22:41:30.016518] (error)   	 - Unable to retune System control channel, freq not covered by the Source used for the inital control channel freq.
````

Just keeps repeating until I stop the instance in Docker.

frameshift18, 2019-02-24:
gain of negative 10?

Pacmannion, 2019-02-26:
So I've been running it fine so far, it's been making recordings. I created the account on OpenMHz, and pasted it into the config file. Which is now throwing out errors again on line 37.

````
{
    "sources": [{
        "center": 854912500.0,
        "rate": 2000000.0,
        "squelch": 0,
        "error": 0,
        "gain": 40,
        "digitalRecorders": 4,
        "driver": "osmosdr",
        "device": "rtl=0",
        "modulation": "qpsk"
    }, {
        "center": 857437700.0,
        "rate": 2000000.0,
        "squelch": 0,
        "error": 0,
        "gain": 40,
        "digitalRecorders": 4,
        "driver": "osmosdr",
        "device": "rtl=1",
        "modulation": "qpsk"
    }, {
        "center": 859312500.0,
        "rate": 2000000.0,
        "squelch": 0,
        "error": 0,
        "gain": 40,
        "digitalRecorders": 3,
        "driver": "osmosdr",
        "device": "rtl=2",
        "modulation": "qpsk"
    }],
    "systems": [{
        "control_channels": [857012500,857437500,858437500,859437500],
        "type": "p25",
        "talkgroupsFile": "ChanList.csv"
	"recordUnknown" false
	{
			"shortName": "bosops",
			"apiKey": "xxxxxxxxxxx"
		}
	,
	"uploadServer": "https://api.openmhz.com"
    }],
    "captureDir": "/app/media"
}
````

robotastic, 2019-02-26:
It looks like you are missing a comma on line 36. When I get an error, I always toss things into a JSON linter: https://jsonlint.com

The Systems block should probably look like:

"systems": [{
        "control_channels": [857012500,857437500,858437500,859437500],
        "type": "p25",
        "talkgroupsFile": "ChanList.csv",
	"recordUnknown" false,
	"shortName": "bosops",
	"apiKey": "xxxxxxxxxxx"
	
	
    }],
"uploadServer": "https://api.openmhz.com",
"captureDir": "/app/media"
}

closed 2019-02-27 by Pacmannion