TrunkRecorder/trunk-recorder #232: Smartnet drops control channel after ~10 minutes

Cole0209, 2019-03-20:
My smartnet system keeps dropping the control channel after around 10 minutes of working perfectly. I also tried to run just the smartnet (msp) system alone but the issue still occurs... If I restart trunkrecorder the problem goes away for another 10 minutes.
https://pastebin.com/WwEkECHV

I don't know if this is relevant but when it is trunking properly i get these info messages spamming up my console.
`
[2019-03-20 19:46:09.181964] (info)   Patching Command: 340 Freq: 0.000000e+00 Talkgroup: 6528

[2019-03-20 19:46:09.506073] (info)   Patching Command: 340 Freq: -inf Talkgroup: 6528

[2019-03-20 19:46:10.033055] (info)   Patching Command: 340 Freq: 0.000000e+00 Talkgroup: 6528

[2019-03-20 19:46:10.165018] (info)   Patching Command: 340 Freq: 0.000000e+00 Talkgroup: 6528

[2019-03-20 19:46:10.559699] (info)   Patching Command: 340 Freq: -inf Talkgroup: 6528

[2019-03-20 19:46:10.691175] (info)   Patching Command: 340 Freq: 0.000000e+00 Talkgroup: 6528

[2019-03-20 19:46:11.148797] (info)   Patching Command: 340 Freq: -inf Talkgroup: 6528

[2019-03-20 19:46:11.341134] (info)   Patching Command: 340 Freq: 0.000000e+00 Talkgroup: 6528

[2019-03-20 19:46:11.412282] (info)   Patching Command: 340 Freq: 0.000000e+00 Talkgroup: 6528

[2019-03-20 19:46:11.999130] (info)   Patching Command: 340 Freq: 0.000000e+00 Talkgroup: 6528`

mdozier9, 2019-03-21:
I had this happen before with cheaper RTL dongles that would thermally drift over time.  If you aren't already using them, I would recommend you get a TCXO dongle for a few extra bucks.

Cole0209, 2019-04-10:
Issue resolved. For future reference I had to swap the ports the dongles were in something was up with them even though they are all USB 3.0

closed 2019-04-10 by Cole0209