TrunkRecorder/trunk-recorder #235: locale::facet::_S_create_c_locale name not valid

Jims-do, 2019-04-21:
I am having trouble when trying to run recorder, I get the following

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  locale::facet::_S_create_c_locale name not valid
Aborted

```
I have tried changing loacle settings, this was no help.

thank you - Jim

bctrainers, 2019-04-21:
Seems to be a general compiler issue with locales, see: https://github.com/potree/PotreeConverter/issues/281#issuecomment-335471817 and https://github.com/potree/PotreeConverter/issues/281#issuecomment-375997005

bctrainers, 2019-04-21:
It may also be that your machines locales package has an issue or doesn't exist, or does exist - but not configured (rare!).

Debian:
`apt-get install locales && dpkg-reconfigure locales`

Ubuntu:
I may be incorrect on this part. Assuming you're running as root or an elevated user;

`apt install locales;locale-gen en_us.UTF-8;update-locale LANG=en_US.UTF-8`

Jims-do, 2019-04-21:
tried suggestions above and others, i am still getting nowhere, fresh install of raspbian still no luck.

closed 2020-07-31 by robotastic