TrunkRecorder/trunk-recorder #238: Permission Denied using upload script

ohaydel, 2019-05-22:
I am configuring Trunk Recorder to upload to OpenMhz.  I have everything running fine, but I am getting an error message about the upload script:

[2019-05-20 23:22:58.972221] (info)   Running upload script: ./encode-upload.sh /home/ohaydel/trunk-recorder/Audio/LWIN/2019/5/20/6852-1558412566_7.70469e+08.wav &
sh: 1: ./encode-upload.sh: Permission denied

Here is my script:  I have it in the trunk-build folder, where I am running TR.  

#! /bin/bash
echo "Encoding: $1" 
filename="$1"
basename="${filename%.*}"
#mp3encoded="$basename.mp3"
mp4encoded="$basename.m4a"
json="$basename.json"
#eval "lame --preset voice" ${filename} 
eval "nice -n 19 /home/ohaydel/ffmpeg -i $filename  -c:a libfdk_aac -b:a 32k -cutoff 18000 $mp4encoded > /dev/null 2>&1"

echo "Upload: $encoded"  
eval "scp $json $mp4encoded ohaydel@gmail.com:~/smartnet-upload "

Here is my config:

{
    "sources": [{
        "center": 770275000,
        "rate": 2048000,
        "error": -43500,
        "gain": 42.1,
        "debugRecorders": 0,
        "digitalRecorders": 4,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=0"
}, {
"center":  774000000,
        "rate": 2048000,
        "error": -34700,
        "gain": 42.1,
        "debugRecorders": 0,
        "digitalRecorders": 4,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=1"

    }],

    "systems": [{
        "control_channels": [769431250],
        "type": "p25",
"talkgroupsFile": "LWIN_SR.csv",
"shortName": "LWIN",
"apiKey": "xxxxxxxx",
"uploadServer": "https://api.openmhz.com",
"talkgroupDisplayFormat": "id_tag",
"hideEncrypted": true,
"hideUnknownTalkgroups": true,
"minDuration": 2,
"uploadScript": "encode-upload.sh"
    }],
    "captureDir": "/home/ohaydel/trunk-recorder/Audio/",
    "frequencyFormat": "mhz"
}

See any issues?  Also I am only getting .wavs in my audio folder, no m4a's.  Ffmeg looks like it loaded correctly, and I ran FFMPEG from command line with no errors.

vabiro, 2019-05-22:
Is the SCP command exactly as it is in the script? If so, I think you're trying to SCP the files to gmail's server. I believe the syntax is <user>@<host> for SCP, so that would result in permission denied

mrtrev, 2019-05-22:
Your upload script needs to be executable.

`chmod +x encode-upload.sh`

ohaydel, 2019-05-22:
I had put my email address in there, so I need to leave it as example.com?

⁣Sent from BlueMail ​

vabiro, 2019-05-22:
@ohaydel If you're uploading to OpenMHZ, I don't think you need the SCP command. The application automatically converts the wav to m4a and uploads based on the settings you have in the .config file, so `eval "nice -n 19 /home/ohaydel/ffmpeg -i $filename -c:a libfdk_aac -b:a 32k -cutoff 18000 ` and `eval "scp $json $mp4encoded ohaydel@gmail.com:~/smartnet-upload "` can be commented out

ohaydel, 2019-05-23:
@vabiro , thank you, I made that change and no more errors.   @mrtrev I also made it executable.

Thanks Guys

ohaydel, 2019-05-23:
Well, I let it run overnight and still nothing on the OpenMhz site.  No errors on TR.  Does it normally take a while?

Thanks for the help.

⁣Sent from BlueMail ​

ohaydel, 2019-05-28:
Well, everything is running great on the Trunk Recorder side, but still no uploads.  I found that when I ran Trunk Recorder as sudo:  sudo ./recorder it was creating the audio files in a locked folder, and the application did not have permission to access the files.  When I run the program without sudo, the permissions problem disappeared.  Also, I am now getting the m4a files in my audio folder, but I HAVE to have the line uncommented for the file to convert in the upload script.  Here is my script now:

#! /bin/bash
echo "Encoding: $1" 
filename="$1"
basename="${filename%.*}"
#mp3encoded="$basename.mp3"
mp4encoded="$basename.m4a"
json="$basename.json"
#eval "lame --preset voice" ${filename} 
eval "nice -n 19 /home/ohaydel/bin/ffmpeg -i $filename  -c:a libfdk_aac -b:a 32k -cutoff 18000 $mp4encoded > /dev/null 2>&1"

echo "Upload: $mp4encoded"  
#eval "scp $json $mp4encoded username@example.com:~/smartnet-upload "

Here is what it looks like when the program is running:

[2019-05-27 20:14:57.006349] (info)   [lwin]	TG: 6845 (-)	Freq: 770.031250	Ending Recorded Call - Last Update: 4s	Call Elapsed: 18
[2019-05-27 20:14:57.006986] (info)   	- Stopping P25 Recorder Num [0]	TG: 6845 (-)	Freq: 770.031250 	TDMA: false	Slot: 0
[2019-05-27 20:14:57.007177] (info)   Running upload script: ./encode-upload.sh /home/ohaydel/trunk-recorder/Recordings/lwin/2019/5/27/6845-1559006079_7.69769e+08.wav &
Encoding: /home/ohaydel/trunk-recorder/Recordings/lwin/2019/5/27/6845-1559006079_7.69769e+08.wav
Upload: /home/ohaydel/trunk-recorder/Recordings/lwin/2019/5/27/6845-1559006079_7.69769e+08.m4a
[2019-05-27 20:15:04.002952] (info)   [lwin]	TG: 24011 (Opelousas PD D1)	Freq: 774.531250	Ending Recorded Call - Last Update: 4s	Call Elapsed: 9
[2019-05-27 20:15:04.003174] (info)   	- Stopping P25 Recorder Num [4]	TG: 24011 (Opelousas PD D1)	Freq: 774.531250 	TDMA: false	Slot: 0
[2019-05-27 20:15:04.003239] (info)   Running upload script: ./encode-upload.sh /home/ohaydel/trunk-recorder/Recordings/lwin/2019/5/27/24011-1559006095_7.74531e+08.wav &
Encoding: /home/ohaydel/trunk-recorder/Recordings/lwin/2019/5/27/24011-1559006095_7.74531e+08.wav
Upload: /home/ohaydel/trunk-recorder/Recordings/lwin/2019/5/27/24011-1559006095_7.74531e+08.m4a
[2019-05-27 20:15:18.913897] (info)   [lwin]	TG: 6849 (-)	Freq: 769.768738	TG not in Talkgroup File 
[2019-05-27 20:15:18.913960] (info)   	- Starting P25 Recorder Num [0]	TG: 6849 (-)	Freq: 769.768738 	TDMA: false	Slot: 0
[2019-05-27 20:15:18.914140] (info)   [lwin]	TG: 6849 (-)	Freq: 769.768738	Starting Recorder on Src: rtl=0
[2019-05-27 20:15:25.008584] (info)   [lwin]	TG: 6849 (-)	Freq: 769.768738	Ending Recorded Call - Last Update: 4s	Call Elapsed: 7
[2019-05-27 20:15:25.008950] (info)   	- Stopping P25 Recorder Num [0]	TG: 6849 (-)	Freq: 769.768738 	TDMA: false	Slot: 0
[2019-05-27 20:15:25.009112] (info)   Running upload script: ./encode-upload.sh /home/ohaydel/trunk-recorder/Recordings/lwin/2019/5/27/6849-1559006118_7.69769e+08.wav &
Encoding: /home/ohaydel/trunk-recorder/Recordings/lwin/2019/5/27/6849-1559006118_7.69769e+08.wav
Upload: /home/ohaydel/trunk-recorder/Recordings/lwin/2019/5/27/6849-1559006118_7.69769e+08.m4a

Any ideas of why it is still not uploading?  I am not getting any error messages anywhere.
Thanks again.
OH

vabiro, 2019-05-28:
@ohaydel A couple of thoughts that may narrow down the problem:
- I am assuming that you have created an account on OpenMHz and that the api key provided by them is in the config.json
- If you're only uploading to OpenMHz then you don't need to run the `"uploadScript": "encode-upload.sh"` All the processing into .m4a files and uploading to the site happens in trunk-recorder.
- When it is uploading, you should see something like `HTTPS Upload Success - file size: 36134` 
- in the logs and nothing about running the upload.sh, or the "encoding" and "uploading" of files.
- I just noticed that `"uploadServer": "https://api.openmhz.com",` is in the system section, along with the api_key value. It should be in the settings where `"frequencyFormat": "mhz"` is, but leave the api_key in the system section

I am not seeing anything like  `HTTPS Upload Success - file size: 36134` which would indicate that it is uploading to the OpenMHz site, so I would do the following:

- remove the upload.sh from config.json
- in config.json move `"uploadServer": "https://api.openmhz.com",` to the position under `frequencyFormat`, and don't forget to add the comma at the end of the `frequencyFormat` line and remove from the `"uploadServer"` line

Let's see if that works.

ohaydel, 2019-05-28:
@vabiro looks like that did the trick.  I made both changes, and now I see the HTTPS upload Success messages.  Upon looking at my stats page on OpenMhz, I see it is receiving data.  I don't see the system just yet, but I expect that will take a few hours to populate.  I will keep an eye on it.  I really appreciate your help!

closed 2020-07-31 by robotastic