TrunkRecorder/trunk-recorder #240: trouble with audio

chrkyl, 2019-05-30:
So i have installed trunk-recorder and I am seeing the various signals and data including talk groups and frequencies, and it is trying to record but it says everything is less than 0 seconds so it auto deletes.  The files before they delete are 0 bytes.  I am stumped.  Anyone hear of anything similar?  I am receiving audio on other systems like SDR Trunk so its probably in my settings.  Here is my config file:

{
    "sources": [{
        "center": 854000000.0,
        "rate": 2600000.0,
        "error": 0,
        "squelch": -50,
        "gain": 30,
        "antenna": "RX",
        "digitalRecorders": 2,
        "driver": "osmosdr",
        "device": "rtl=rtl1",
        "modulation": "qpsk"
    },{
        "center": 858000000.0,
        "rate": 2400000.0,
        "error": 0,
        "squelch": -50,
        "gain": 30,
        "antenna": "TX/RX",
        "digitalRecorders": 2,
        "driver": "osmosdr",
        "device": "rtl=rtl2",
        "modulation": "qpsk"
    }],
    "systems": [{
        "control_channels": [854487500],
        "type": "p25",
	"recordUnknown": "true",
        "talkgroupsFile": "ChanList.csv"
    }],
 "captureDir": "/home/ckyle/recordings/"
}

vabiro, 2019-05-31:
Hi,
What SDR are you using? That's a big coverage area for an RTL-SDR on the first device. Is it possible there isn't an SDR covering the voice channels.
Are you getting any control channel rate notifications?
I don't have a .0 at the end of the center frequency or rate.
The rate I have is 2048000.
I also have 0 for squelch, which I understand is appropriate for P25 systems.

See if those changes make a difference

chrkyl, 2019-06-02:
That did it.  Thanks!

closed 2019-06-02 by chrkyl