TrunkRecorder/trunk-recorder #241: OpenMHZ RaspberryPi no sound

jkole13261, 2019-06-05:
So i recently got my program to start recording to my local computer using 2 RTL_SDR dongles. and on OpenMHZ it is showing the talkgroups that i am recording uploading successfully however there is no actual sound attached. Does anybody have any insight to this? the system is on open MHZ as Monmouth County P25 and my config file is below... 

{
    "sources": [{
        "center": 770800000.0,
        "rate": 2400000.0,
        "error": 0,
        "gain": 16,
	"modulation":"qpsk",
        "antenna": "TX/RX",
        "digitalRecorders": 3,
        "driver": "osmosdr",
	"device":"rtl=0"
    },
	{
	"center": 773200000.0,
        "rate": 2400000.0,
        "error": 0,
        "gain": 16,
	"modulation":"qpsk",
        "antenna": "TX/RX",
        "digitalRecorders": 3,
        "driver": "osmosdr",
	"device":"rtl=1"
    }],
    "systems": [{
        "control_channels": [772456250,773431250,773956250,774206150],
        "type": "p25",
        "talkgroupsFile": "mcsotgs.csv",
        "shortName": "monconjp25",
	"hideEncrypted":"true",
	"talkgroupDisplayFormat":"id_tag",
        "apiKey": "**THE CORRECT API KEY**"
    }],
    "uploadServer": "https://api.openmhz.com",
    "statusAsString":"true"
}

Thanks for the help!

vabiro, 2019-06-05:
Hi,
How large are the wav files? Are they 44 bytes or 0?
Are you seeing any errors on the decode rates for the control channels? That may be helped by raising the gain level to something like 25. 16 looks a bit low, unless you're really close to the repeater.

jkole13261, 2019-06-05:
the wav files are various sizes from 2kB and up. I adjusted to 35 gain and still has the same issue. The radio traffic sounds as if it is being encrypted when uploaded to openMHZ even though i know for fact the talk groups I am monitoring are all transmitted in the clear.

As for the errors, i am not getting anything. I attached a sample of the output.
[image]

vabiro, 2019-06-09:
@jkole13261 Are the locally stored wav files OK, or are they distorted?
The files are uploaded to OpenMHz as M4a files, and I wonder if there is something funky in that process.

jkole13261, 2019-06-09:
both the m4a and wav files are all sounding distorted. Ive adjusted the gain and error for both antennas and seem to still have the same issue

iknoxx, 2019-07-05:
Could this issue be something to do with your SDR? Could you be experiencing a drift in freq?

jkole261, 2019-08-10:
> 
> 
> Could this issue be something to do with your SDR? Could you be experiencing a drift in freq?

So i used GQRX to check, and the main control channel is 772.456250 which seems very spot on even when you zoom in on the frequency. 

[image]
[image]

jkole13261, 2019-08-10:
> 
> 
> Could this issue be something to do with your SDR? Could you be experiencing a drift in freq?

@iknoxx 

So i used GQRX to check, and the main control channel is 772.456250 which seems very spot on even when you zoom in on the frequency. 
[image]
[image]

robotastic, 2019-09-12:
Checking in - did you end up getting this solved? Sometimes the Pi just doesn;t have enough juice.

jkole13261, 2019-09-14:
> Checking in - did you end up getting this solved? Sometimes the Pi just doesn;t have enough juice.

that was the issue. switched to a new computer up and going with flying colors

closed 2019-09-14 by jkole13261