TrunkRecorder/trunk-recorder #247: Unable to setup a system to record, exiting

faragher, 2019-07-01:
I am on day three of attempting to get this P25 trunk system running, and this is so close, but I'm not able to proceed any further.

I have provided the complete output from the run below, and my config,json. OP25 runs without issue. Installed via Git. RTL-SDR dongle has a serial number of "0". Ubuntu 18.04 on bare metal. ~/trunk-recorder exists and is owned by "faragher:faragher"

Let me know if anything else would be helpful

`linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

[2019-07-01 17:51:08.859813] (info)   Using Config file: ./config.json

[2019-07-01 17:51:08.860133] (info)
-------------------------------------
SYSTEMS
-------------------------------------
0
[2019-07-01 17:51:08.860162] (info)

System Number: 1
-------------------------------------
1
[2019-07-01 17:51:08.860173] (info)   Short Name: WaCo
[2019-07-01 17:51:08.860182] (info)   System Type: p25
[2019-07-01 17:51:08.860189] (info)   Control Channels:
[2019-07-01 17:51:08.860230] (info)   154.965
[2019-07-01 17:51:08.860265] (info)   API Key:
[2019-07-01 17:51:08.860274] (info)   Upload Script:
[2019-07-01 17:51:08.860281] (info)   Call Log: true
[2019-07-01 17:51:08.860292] (info)   Audio Archive: true
[2019-07-01 17:51:08.860301] (info)   Loading Talkgroups...
[2019-07-01 17:51:08.860315] (error)   Error Opening TG File: WaCo.csv

[2019-07-01 17:51:08.860325] (info)   Talkgroups File: WaCo.csv
[2019-07-01 17:51:08.860334] (info)   Record Unknown Talkgroups: true
[2019-07-01 17:51:08.860345] (info)   Talkgroup Display Format: Id
[2019-07-01 17:51:08.860359] (info)   Hide Encrypted Talkgroups: false
[2019-07-01 17:51:08.860368] (info)   Hide Unknown Talkgroups: false
[2019-07-01 17:51:08.860376] (info)   Minimum Call Duration (in seconds): 0
[2019-07-01 17:51:08.860387] (info)
[2019-07-01 17:51:08.860394] (info)

-------------------------------------
SOURCES
-------------------------------------

[2019-07-01 17:51:08.860421] (info)   Driver: osmosdr
[2019-07-01 17:51:08.860430] (info)   Center: 1.549650e+08
[2019-07-01 17:51:08.860454] (info)   Rate: 2048000
[2019-07-01 17:51:08.860472] (info)   Error: 1000
[2019-07-01 17:51:08.860487] (info)   PPM Error: 0
[2019-07-01 17:51:08.860498] (info)   Gain: 39
[2019-07-01 17:51:08.860512] (info)   IF Gain: 0
[2019-07-01 17:51:08.860523] (info)   BB Gain: 0
[2019-07-01 17:51:08.860533] (info)   LNA Gain: 0
[2019-07-01 17:51:08.860544] (info)   PGA Gain: 0
[2019-07-01 17:51:08.860554] (info)   TIA Gain: 0
[2019-07-01 17:51:08.860563] (info)   MIX Gain: 0
[2019-07-01 17:51:08.860572] (info)   VGA1 Gain: 0
[2019-07-01 17:51:08.860584] (info)   VGA2 Gain: 0
[2019-07-01 17:51:08.860594] (info)   Squelch: 0
[2019-07-01 17:51:08.860604] (info)   Idle Silence: false
[2019-07-01 17:51:08.860613] (info)   Digital Recorders: 8
[2019-07-01 17:51:08.860624] (info)   Debug Recorders: 0
[2019-07-01 17:51:08.860635] (info)   SigMF Recorders: 0
[2019-07-01 17:51:08.860644] (info)   Analog Recorders: 0
[2019-07-01 17:51:08.860654] (info)   Modulation: qpsk
[2019-07-01 17:51:08.860664] (info)   Source Device: rtl=0
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
Using device #0 Realtek RTL2838UHIDIR SN: 0
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[2019-07-01 17:51:09.533383] (info)   SOURCE TYPE OSMOSDR (osmosdr)
[2019-07-01 17:51:09.533426] (info)   Setting sample rate to: 2048000
[R82XX] PLL not locked!
[2019-07-01 17:51:09.653013] (info)   Actual sample rate: 2048000
[2019-07-01 17:51:09.653059] (info)   Tunning to 1.549660e+08
[2019-07-01 17:51:09.705203] (info)   Gain Stage: LNA supported values: 0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48 49.6
[2019-07-01 17:51:09.705234] (info)   Max Freqency: 1.559890e+08
[2019-07-01 17:51:09.705256] (info)   Min Freqency: 1.539410e+08
[2019-07-01 17:51:09.731388] (info)   LNA Gain set to: 0
[2019-07-01 17:51:09.757254] (info)   TIA Gain set to: 0
[2019-07-01 17:51:09.783386] (info)   PGA Gain set to: 0
[2019-07-01 17:51:09.809397] (info)   Gain set to: 38.6
[2019-07-01 17:51:09.809443] (info)   Setting antenna to [RX]
[2019-07-01 17:51:09.817699] (info)      P25 Recorder Initial Rate: 512000 Resampled Rate: 30117.6 Initial Decimation: 4 Decimation: 17 System Rate: 30000 ARB Rate: 0.996094
[2019-07-01 17:51:09.817732] (info)   P25 Recorder Taps - initial: 311 channel: 1551 ARB: 731 Total: 2593
Project 25 IMBE Encoder/Decoder Fixed-Point implementation
Developed by Pavel Yazev E-mail: pyazev@gmail.com
Version 1.0 (c) Copyright 2009
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; see the file ``LICENSE'' for details.
[2019-07-01 17:51:09.826663] (info)      P25 Recorder Initial Rate: 512000 Resampled Rate: 30117.6 Initial Decimation: 4 Decimation: 17 System Rate: 30000 ARB Rate: 0.996094
[2019-07-01 17:51:09.826692] (info)   P25 Recorder Taps - initial: 311 channel: 1551 ARB: 731 Total: 2593
[2019-07-01 17:51:09.834472] (info)      P25 Recorder Initial Rate: 512000 Resampled Rate: 30117.6 Initial Decimation: 4 Decimation: 17 System Rate: 30000 ARB Rate: 0.996094
[2019-07-01 17:51:09.834501] (info)   P25 Recorder Taps - initial: 311 channel: 1551 ARB: 731 Total: 2593
[2019-07-01 17:51:09.842314] (info)      P25 Recorder Initial Rate: 512000 Resampled Rate: 30117.6 Initial Decimation: 4 Decimation: 17 System Rate: 30000 ARB Rate: 0.996094
[2019-07-01 17:51:09.842343] (info)   P25 Recorder Taps - initial: 311 channel: 1551 ARB: 731 Total: 2593
[2019-07-01 17:51:09.850183] (info)      P25 Recorder Initial Rate: 512000 Resampled Rate: 30117.6 Initial Decimation: 4 Decimation: 17 System Rate: 30000 ARB Rate: 0.996094
[2019-07-01 17:51:09.850212] (info)   P25 Recorder Taps - initial: 311 channel: 1551 ARB: 731 Total: 2593
[2019-07-01 17:51:09.858095] (info)      P25 Recorder Initial Rate: 512000 Resampled Rate: 30117.6 Initial Decimation: 4 Decimation: 17 System Rate: 30000 ARB Rate: 0.996094
[2019-07-01 17:51:09.858125] (info)   P25 Recorder Taps - initial: 311 channel: 1551 ARB: 731 Total: 2593
[2019-07-01 17:51:09.866062] (info)      P25 Recorder Initial Rate: 512000 Resampled Rate: 30117.6 Initial Decimation: 4 Decimation: 17 System Rate: 30000 ARB Rate: 0.996094
[2019-07-01 17:51:09.866091] (info)   P25 Recorder Taps - initial: 311 channel: 1551 ARB: 731 Total: 2593
[2019-07-01 17:51:09.874058] (info)      P25 Recorder Initial Rate: 512000 Resampled Rate: 30117.6 Initial Decimation: 4 Decimation: 17 System Rate: 30000 ARB Rate: 0.996094
[2019-07-01 17:51:09.874087] (info)   P25 Recorder Taps - initial: 311 channel: 1551 ARB: 731 Total: 2593
[2019-07-01 17:51:09.877175] (info)
-------------------------------------

[2019-07-01 17:51:09.877196] (info)

-------------------------------------
INSTANCE
-------------------------------------

[2019-07-01 17:51:09.877220] (info)   Capture Directory: /home/faragher/trunk-recorder
[2019-07-01 17:51:09.877229] (info)   Upload Server:
[2019-07-01 17:51:09.877237] (info)   Status Server:
[2019-07-01 17:51:09.877257] (info)   Instance Key:
[2019-07-01 17:51:09.877274] (info)   Instance Id:
[2019-07-01 17:51:09.877290] (info)   Default Mode: digital
[2019-07-01 17:51:09.877304] (info)   Call Timeout (seconds): 3
[2019-07-01 17:51:09.877320] (info)   Log to File: false
[2019-07-01 17:51:09.877336] (info)   Control channel warning rate: 10
[2019-07-01 17:51:09.877355] (info)   Control channel retune limit: 0
[2019-07-01 17:51:09.877418] (info)   Frequency format: 0
[2019-07-01 17:51:09.877436] (info)   Status as String: false
[2019-07-01 17:51:09.877457] (info)   [WaCo]    Started with Control Channel: 1.549650e+02
[2019-07-01 17:51:09.877485] (error)   Unable to setup a System to record, exiting...
`

`{
    "sources": [{
        "center": 154965000,
        "rate": 2048000,
        "error": 0,
        "gain": 39,
        "debugRecorders": 0,
        "digitalRecorders": 8,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=0"
    }],
    "systems": [{
        "control_channels": [154.965000],
        "type": "p25",
        "talkgroupsFile": "WaCo.csv",
                                "shortName": "WaCo"
    }],
    "captureDir": "/home/faragher/trunk-recorder/"
}
`

gofaster, 2019-07-01:
Your control channels line should be
 "control_channels": [154965000],

faragher, 2019-07-01:
Wow. I can't believe I missed that. Works now. Thanks!

closed 2019-07-01 by faragher