TrunkRecorder/trunk-recorder #248: recorder stops recording

imagesafari, 2019-07-10:
I've got a weird issue whereI have a recorder instance recording several conventional P25 channels as well as a couple conventional analog channels.

If left running, over time, it will stop recording some of those channels. After a restart, recording resumes.

Any ideas?

kg6uyz, 2019-07-11:
I have a similar issue, on mine the p25 trunking recorder stops recording
after a long period of time not restarting tr.  My conventional recorders
dont get affected and continue to work.  I might have had one or two
occasions where my conventional recorders might have stopped recording.

robotastic, 2019-09-12:
I haven't seen this issue before. Could it be the system hitting some limit around open files or something like that? I haven't run into this issue myself, but I am not recording Conventional. One option could be to setup a script to automatically reboot every night.

closed 2019-09-12 by robotastic